Koenkk/zigbee2mqtt #20271: [Wrong device]: Avatto ME167 is seen as ME168 with wrong picture

marcelhoogantink, 2023-12-18:
### Link

https://nl.aliexpress.com/item/1005006191834430.html?spm=a2g0o.order_list.order_list_main.23.56cd79d2t3hWcv&gatewayAdapt=glo2nld

### Model

TRV06

### Description

AVATTO Tuya Zigbee 3.0 Thermostat Radiator Valve Actuator Smart TRV

### Vendor

Avatto

### Picture (link)

https://ae01.alicdn.com/kf/Sc783b2ec5b0b4371a4e9ca948937525dV/AVATTO-Tuya-Zigbee-3-0-Thermostat-Radiator-Valve-Actuator-Smart-TRV-Programmable-Temperature-Controller-with-Alexa.jpg_80x80.jpg_.webp

### Database entry

{"id":27,"type":"EndDevice","ieeeAddr":"0xa4c13891c22788a8","nwkAddr":48235,"manufId":4417,"manufName":"_TZE200_p3dbf6qs","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_p3dbf6qs","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1702894664475,"defaultSendRequestWhen":"immediate"}

2pirko, 2023-12-30:
I have the same problem

Klepth, 2024-01-10:
Hello, same problem i have a ME167 but zigbee2mqtt detect ME168 and  the TVR didn't work normally. Schedule dont work and the TVR always heated at 23°C

closed 2024-08-08 by github-actions[bot]

smithbill17, 2024-10-11:
Can anyone tell me how to physically identify my Avatto TRV as an ME167 or ME168 ?

In Zigbee2MQTT it's added as an ME168, but there are a number of user issues indicating that the setpoint temperature on their TRV keeps jumping to 153degrees & issues with 4 or 6 time schedules per day. I don't think I see those issues, but all three of my Avatto TRVs randomly open/close at odd times of the day (I leave them all in 'heat' mode & dont use the schedule).

So how do I know if my TRVs are actually ME168 & correctly identified in Zigbee2MQTT, or they are really ME167 and wrongly identified as ME168?

bakulapal, 2024-11-07:
> Can anyone tell me how to physically identify my Avatto TRV as an ME167 or ME168 ?
>
> In Zigbee2MQTT it's added as an ME168, but there are a number of user issues indicating that the setpoint [...]

You can figure it out ftom the "Zigbee Manufacturer" string in the about tab. Mine is the problematic  ME167 falsely recognized as ME168. It has the _TZE200_p3dbf6qs identifier. 
This fix seem to work for me: https://github.com/sychu/avatto_me167_TZE200_p3dbf6qs?tab=readme-ov-file#avatto_me167_tze200_p3dbf6qs

smithbill17, 2024-11-07:
So are you saying that if Zigbee2MQTT shows under the About tab, that the Device Manufacturer = _TZE200_p3dbf6qs, then that means it's an ME167 ? (even although the Model is identified as ME168)

smithbill17, 2024-11-07:
Okay, I added the ME167 external converter & it seems to have picked it up in my Zigbee2MQTT in Home Assistant.

Although I haven't actually noticed any difference with my TRV's as I never actually used the schedule for adjusting temperature setting (I generally just leave them set at 22degC & use my room Thermostat to turn the heating on/off).

I one difference I have noticed, is that the 'Temperature Calibration' value is now a maximum of 3degC.