Koenkk/zigbee2mqtt #20273: [New device support]: Immax Neo Smart Keypad 07505L

nobges, 2023-12-18:
### What happened?

Hello,

I bought a [Immax 07505L](https://www.zigbee2mqtt.io/devices/07505L.html) alarm keypad and I am trying to get it working. No issue adding the device to the network and getting it identified, however no actions are recognized when I press its buttons. I looked it up and found those posts:
[https://github.com/Koenkk/zigbee2mqtt/issues/17800](https://github.com/Koenkk/zigbee2mqtt/issues/17800)
[https://github.com/zigpy/zha-device-handlers/issues/1852](https://github.com/zigpy/zha-device-handlers/issues/1852)
[https://github.com/Koenkk/zigbee2mqtt/issues/11728](https://github.com/Koenkk/zigbee2mqtt/issues/11728)

Which all use an external converter with varying levels of success. I tried [this one](https://github.com/Koenkk/zigbee2mqtt/issues/11728#issuecomment-1125365768) (including the modifications suggested in the thread). But while the converter is properly linked to the device, it's still not working. 

This is what logs indicate: 
`Exception while calling fromZigbee converter: Cannot read properties of undefined (reading '0')}
debug 2023-12-17 21:53:12 TypeError: Cannot read properties of undefined (reading '0') at Object.convert (/app/data/extension/externally-loaded.js:36:32) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51) at EventEmitter.emit (node:events:525:35) at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22) at Controller.<anonymous> (/app/lib/zigbee.ts:108:27) at Controller.emit (node:events:513:28) at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:527:14) at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:738:18) at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70) at ZStackAdapter.emit (node:events:513:28)`

"`Object.convert (/app/data/extension/externally-loaded.js:36:32`" refers to this line in the converter : "`const dp = msg.data.dpValues[0].dp;`"

But from there I don't know how to troubleshoot it. Any idea?

Thanks a lot!

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20221226

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus TI CC2652P

besiktas97, 2024-01-02:
I am strugling with the same issue and didn't also tried the pr's without succes.

panostaja, 2024-02-07:
By default, the keyboard only sends action messages after the correct pin+action_button is pressed.
Default user pin = 1234, so pressing 1234(lock) should result as an action.
Refer here for keypad internal commands to change its behavior: https://manuals.plus/immax/07505l-neo-smart-keypad-manual#axzz8R2Z5dWOe

martin1997nm, 2024-06-03:
I had the same issue(1.35), but diving deeper into the debug log, i found out that not only the manufacturer code was not right, but the tuya datapoint were wrong too. There for i satrted thinking that the device got an updated FW in a newer production.
[image]
**Zigbee Manufacturer _TZE200_moycceze**
So i made my custom converter for this device.
[image]

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const toZigbeeConverters = {
    arm_delay_time: {
        key: ['arm_delay_time'],
        convertSet: async (entity, key, value, meta) => {
            const dp = 103; 
            await tuya.sendDataPointValue(entity, dp, value);
        },
        convertGet: async (entity, key, meta) => {
            const dp = 103; 
            await tuya.sendDataPointValue(entity, dp, 0);
        },
    },
    beep_sound_enabled: {
        key: ['beep_sound_enabled'],
        convertSet: async (entity, key, value, meta) => {
            const dp = 104; 
            await tuya.sendDataPointBool(entity, dp, value === 'ON');
        },
        convertGet: async (entity, key, meta) => {
            const dp = 104; 
            await tuya.sendDataPointValue(entity, dp, 0);
        },
    },
    quick_home_enabled: {
        key: ['quick_home_enabled'],
        convertSet: async (entity, key, value, meta) => {
            const dp = 105; 
            await tuya.sendDataPointBool(entity, dp, value === 'ON');
        },
        convertGet: async (entity, key, meta) => {
            const dp = 105; 
            await tuya.sendDataPointValue(entity, dp, 0);
        },
    },
    quick_disarm_enabled: {
        key: ['quick_disarm_enabled'],
        convertSet: async (entity, key, value, meta) => {
            const dp = 106; 
            await tuya.sendDataPointBool(entity, dp, value === 'ON');
        },
        convertGet: async (entity, key, meta) => {
            const dp = 106; 
            await tuya.sendDataPointValue(entity, dp, 0);
        },
    },
    quick_arm_enabled: {
        key: ['quick_arm_enabled'],
        convertSet: async (entity, key, value, meta) => {
            const dp = 107; 
            await tuya.sendDataPointBool(entity, dp, value === 'ON');
        },
        convertGet: async (entity, key, meta) => {
            const dp = 107; 
[... 186 lines cut ...]
    meta: {
        tuyaDatapoints: [
            [3, 'battery', tuya.valueConverter.raw],
            [24, 'tamper', tuya.valueConverter.raw],
            [26, 'action', tuya.valueConverter.static('disarm')],
            [27, 'action', tuya.valueConverter.static('arm_away')],
            [28, 'action', tuya.valueConverter.static('arm_home')],
            [29, 'action', tuya.valueConverter.static('sos')],
            [108, 'admin_code', tuya.valueConverter.raw],
            [109, 'user_code', tuya.valueConverter.raw],
            [103, 'arm_delay_time', tuya.valueConverter.raw],
            [104, 'beep_sound_enabled', tuya.valueConverter.trueFalse],
            [105, 'quick_home_enabled', tuya.valueConverter.trueFalse],
            [106, 'quick_disarm_enabled', tuya.valueConverter.trueFalse],
            [107, 'quick_arm_enabled', tuya.valueConverter.trueFalse],
            [111, 'arm_delay_beep_sound', tuya.valueConverter.trueFalse],
        ],
    },
};

module.exports = definition;

           
```
Have fun whit it!

NikolamViser, 2024-06-15:
@martin1997nm Thank you for the code. I have managed to get it working finally.
Just one thing, with the latest Z2M version I had to remove the line: const extend = require( 'zigbee-herdsman-converters/lib/extend'); 
Converter wouldn't load with it.

Koenkk, 2024-06-16:
Could you make a PR to add out of the box support for this device?

xmatr1x69, 2024-09-06:
Hi @martin1997nm 
in the action for the keypad when arming/disarming/any action there is no info  "action": "", I`ve got "user_id": "0".
Any idea what I can do?

closed 2025-01-09 by github-actions[bot]

gmartin42, 2025-04-21:
Tank you @martin1997nm with your previous externals converters I made a new converter with new version of z2m (working on 2.2.1 for me)

```
import {presets, access, text, numeric, binary} from "zigbee-herdsman-converters/lib/exposes";
import {fingerprint, valueConverter, tz, fz} from "zigbee-herdsman-converters/lib/tuya";

export default {
    fingerprint: fingerprint("TS0601", ["_TZE200_n9clpsht", "_TZE200_nyvavzbj", "_TZE200_moycceze"]),
    model: "07505L",
    vendor: "Immax",
    description: "Neo smart keypad",
    fromZigbee: [
	fz.datapoints,
    ],
    toZigbee: [
	tz.datapoints,
    ],
    exposes: [
        presets.action(["disarm", "arm_home", "arm_away", "sos"]),
        presets.battery(),
        presets.tamper(),
        text("admin_code", access.STATE_SET).withDescription("Admin code").withAccess(access.STATE),
        text("last_added_user_code", access.STATE_SET).withDescription("Last Added User code").withAccess(access.STATE),
        numeric("arm_delay_time", access.STATE_SET).withValueMin(0).withValueMax(180).withDescription("Arm Delay Time"),
        binary("beep_sound_enabled", access.STATE_SET, "ON", "OFF").withDescription("Beep Sound Enabled"),
        binary("quick_home_enabled", access.STATE_SET, "ON", "OFF").withDescription("Quick Home Enabled"),
        binary("quick_disarm_enabled", access.STATE_SET, "ON", "OFF").withDescription("Quick Disarm Enabled"),
        binary("quick_arm_enabled", access.STATE_SET, "ON", "OFF").withDescription("Quick Arm Enabled"),
        binary("arm_delay_beep_sound", access.STATE_SET, "ON", "OFF").withDescription("Arm Delay Beep Sound"),
        text("user_id", access.STATE).withDescription("User ID"),
    ],
    meta: {
        tuyaDatapoints: [
            [3, "battery", valueConverter.raw],
            [24, "tamper", valueConverter.trueFalse1],
            [26, "action", valueConverter.static("disarm")],
            [27, "action", valueConverter.static("arm_away")],
            [28, "action", valueConverter.static("arm_home")],
            [29, "action", valueConverter.static("sos")],
            [108, "admin_code", valueConverter.raw],
            [109, "last_added_user_code", valueConverter.raw],
            [103, "arm_delay_time", valueConverter.raw],
            [104, "beep_sound_enabled", valueConverter.onOff],
            [105, "quick_home_enabled", valueConverter.onOff],
            [106, "quick_disarm_enabled", valueConverter.onOff],
            [107, "quick_arm_enabled", valueConverter.onOff],
            [111, "arm_delay_beep_sound", valueConverter.onOff],
            [112, "user_id", valueConverter.raw],
        ],
    },
};
```

Next step is to find how  I can do a proper PR