Koenkk/zigbee2mqtt #20276: Led stripts (CCT) turn off quickly after turning on, while z2m still shows them as on

borgqueenx, 2023-12-18:
### What happened?

I am now using my THIRD cct led controller, that also has rgb capability's.
While my first one never had this problem while being connected to smartthings.
On z2m with home assistant, calling the light_on service OR simply a device on, does only sometimes work. a lot of times, it turns on with a fade effect and then turns off again.

### What did you expect to happen?

Lights staying on after turning them on.

### How to reproduce it (minimal and precise)

Any led strip controller from amazon, try to put light_on service calls in automations or simply device on automations. You'l notice it flashes to turn on, then instantly turns off(without a fade out effect)

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

 20230507

### Adapter

Zigstar UZG-01

julianfs, 2023-12-18:
I have the problem with MOES GU10 bulbs.

I'm not 100% sure, but I think it has something to do with using a transition setting & Z2M seeing a brightness of 1 & turning the device off. 

Also, are you using HA helper groups at all? I've seen occasional unusual behaviour here also.

borgqueenx, 2023-12-18:
> I have the problem with MOES GU10 bulbs.
> 
> I'm not 100% sure, but I think it has something to do with using a transition setting & Z2M seeing a brightness of 1 & turning the device off.
> 
> Also, are you using HA helper groups at all? I've seen occasional unusual behaviour here also.

i doubt it, since using the device setting to turn it on (device> turn on) does the same. it does not allow a transition effect.

hallenberg, 2024-01-01:
I see the same issue with my Hue bulbs. I traced it back to commit ddcab827ad086949b87d5eaff8d261d8f2d529bf which updated zigbee-herdsman-converters to 15.14.1

Koenkk, 2024-01-01:
@hallenberg does zigbee-herdsman-converters 15.14.0 work?

hallenberg, 2024-01-01:
No, if I update zigbee-herdsman-converters to 15.14.0 it doesn't work either, but 15.13.1 works.

Koenkk, 2024-01-01:
Could you provide the herdsman debug log of this issue when it works with 15.13.1 and breaks with 15.14.0?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

hallenberg, 2024-01-01:
Sure, here you go. I only included the output from the set color command now, if that's sufficient.
[broken.txt](https://github.com/Koenkk/zigbee2mqtt/files/13804290/broken.txt)
[working.txt](https://github.com/Koenkk/zigbee2mqtt/files/13804291/working.txt)

borgqueenx, 2024-01-01:
> I see the same issue with my Hue bulbs. I traced it back to commit [ddcab82](https://github.com/Koenkk/zigbee2mqtt/commit/ddcab827ad086949b87d5eaff8d261d8f2d529bf) which updated zigbee-herdsman-converters to 15.14.1

that's a whole different product then led strips though. hope the same bug fix could solve the issue.

Koenkk, 2024-01-02:
Fixed it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

borgqueenx, 2024-01-02:
Awesome, thanks a lot! how long does it take to arrive on home assistant approximately?
I have more issues with transitioning effects and colors, will this fix also address that or is it perhaps limits of led controllers?
i notice that the led controllers always do a transitioning effect, even if i tell it not to. And sometimes, if i make a transitioning effect, it is ignored. i then add a delay before the next action and then it works....

also, when i start a light with a specific color, the device first turns on at the previous color in the previous brightness level (with a transition effect, even if not asked for), then turns towards the new set color, also with a transition effect (also again, even if not having selected a transition effect)

Koenkk, 2024-01-02:
Edge in a few hours from now, for the release you have to wait till 1 February.

closed 2024-07-31 by github-actions[bot]