Koenkk/zigbee2mqtt #20286: No Power data returned by Xiaomi WS-USC03 ('TypeError: Cannot read properties of undefined (reading 'read')')

TraceWilson, 2023-12-18:
### What happened?

All attempts to update Power data for all Xioami WS-USC03 switches on the network return 

'TypeError: Cannot read properties of undefined (reading 'read')'

Enabled debugging of zigbee-hersman*, and see the following in the Docker log

```
Zigbee2MQTT:debug 2023-12-17 14:57:47: Received MQTT message on 'zigbee2mqtt/Garage Light Switch/get' with data '{"power":""}'
Zigbee2MQTT:debug 2023-12-17 14:57:47: Publishing get 'get' 'power' to 'Garage Light Switch'
Zigbee2MQTT:error 2023-12-17 14:57:47: Publish 'get' 'power' to 'Garage Light Switch' failed: 'TypeError: Cannot read properties of undefined (reading 'read')'
Zigbee2MQTT:debug 2023-12-17 14:57:47: TypeError: Cannot read properties of undefined (reading 'read')
    at Object.convertGet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:2279:28)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:275:37)
    at EventEmitter.emit (node:events:529:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:141:27)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:130:27)
    at WebSocket.emit (node:events:517:28)
    at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1192:20)
    at Receiver.emit (node:events:517:28)
    at Receiver.dataMessage (/app/node_modules/ws/lib/receiver.js:560:14)
```

### What did you expect to happen?

Power data from switch returned

### How to reproduce it (minimal and precise)

Query power data from switch

### Zigbee2MQTT version

 1.34.0 commit: aae7312

### Adapter firmware version

20210708

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_88389f3af112ec11976821c7bd930c07

Xtiger123X, 2024-05-23:
is there a known fix for this. I have the same problem with the Aqara EU H1 with Neutral.

jvrietschoten, 2024-08-01:
Same issue here except I’m using the deconz adapter (Conbee II)

jfgirouxhome, 2024-11-04:
Same issue with my WS-USC03

Koenkk, 2024-11-04:
Could you provide the debug log of this with z2m 1.41.0?

jfgirouxhome, 2024-11-05:
[2024-11-05 11:35:13] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x54ef441000c37b69', payload '{"action":"","device_temperature":null,"energy":null,"flip_indicator_light":"OFF","linkquality":75,"operation_mode":"decoupled","power":null,"power_outage_count":null,"power_outage_memory":true,"state":"ON","update":{"installed_version":31,"latest_version":31,"state":"idle"},"update_available":null,"voltage":null}'

Nooton92, 2024-12-27:
I have the same issue. No power, temperature, voltage, energy or actions are coming through. Decoupled mode disables the relay properly but is essentially a dead switch then since single and double presses are not registered at all.

Koenkk, 2025-02-20:
I will need the debug log of this, otherwise I cannot help.

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

TraceWilson, 2025-02-21:
I've got several logs, going back about a month, all about 10MB.  How many would you like, adn how would you like me to get them to you?

Koenkk, 2025-02-22:
@TraceWilson I need the logging when this error is logged: `TypeError: Cannot read properties of undefined (reading 'read')`

TraceWilson, 2025-02-22:
I was able to trigger the message...

```
[2025-02-22 14:01:15] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Garage Pedestrian Light Switch/get' with data '{"power":""}'
[2025-02-22 14:01:15] debug:    z2m: Publishing get 'get' 'power' to 'Garage Pedestrian Light Switch'
[2025-02-22 14:01:15] error:    z2m: Publish 'get' 'power' to 'Garage Pedestrian Light Switch' failed: 'TypeError: Cannot read properties of undefined (reading 'read')'
[2025-02-22 14:01:15] debug:    z2m: TypeError: Cannot read properties of undefined (reading 'read')
    at Object.convertGet (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/lumi.ts:3664:28)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:276:37)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:23)
    at EventEmitter.emit (node:events:536:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:137:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:157:27)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:182:27)
    at WebSocket.emit (node:events:524:28)
    at Receiver.receiverOnMessage (/app/node_modules/.pnpm/ws@8.18.0/node_modules/ws/lib/websocket.js:1220:20)
    at Receiver.emit (node:events:524:28)
```

Koenkk, 2025-02-22:
Could you provide the data/database.db entry of your device?

TraceWilson, 2025-02-23:
`{"id":15,"type":"Router","ieeeAddr":"0x54ef44100062d2f5","nwkAddr":9434,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.switch.b1naus01","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2,3,4,5,6,9,1794,2820],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.switch.b1naus01","manufacturerName":"LUMI","powerSource":1,"zclVersion":3,"appVersion":31,"stackVersion":2,"hwVersion":1,"dateCode":"12-28-2021"}},"genOnOff":{"attributes":{"onOff":0}},"aqaraOpple":{"attributes":{"240":0,"512":1,"513":1}},"haElectricalMeasurement":{"attributes":{"activePower":0}},"seMetering":{"attributes":{"status":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00257ca310","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00257ca310","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":31,"stackVersion":2,"hwVersion":1,"dateCode":"12-28-2021","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1740313719817}`

Koenkk, 2025-02-23:
Pushed a fix.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) tomorrow and in the next release which is every 1st of the month.

TraceWilson, 2025-03-06:
That's got a power value returning, although it's returning 0, even while the switch is on.  Also, none of the other power/energy properties are returning values either.  I'm not noticing any errors in the logs.  Below are the state properties returned while the light was on:

```
{
    "flip_indicator_light": "OFF",
    "last_seen": "2025-03-05T22:08:34-05:00",
    "linkquality": 84,
    "operation_mode": "control_relay",
    "power": 0,
    "power_outage_memory": true,
    "state": "ON",
    "update": {
        "installed_version": 31,
        "latest_version": 31,
        "state": "idle"
    },
    "device_temperature": null,
    "energy": null,
    "power_outage_count": null,
    "voltage": null
}
```

Nooton92, 2025-03-08:
> That's got a power value returning, although it's returning 0, even while the switch is on. Also, none of the other power/energy properties are returning values either. I'm not noticing any errors [...]

I've got the same. No errors in the log.

TraceWilson, 2025-05-08:
Any update on this one?  Seemed like we were making progress.  If there is anything I can do to help, please let me know.

closed 2025-09-19 by github-actions[bot]