Koenkk/zigbee2mqtt #20290: Tempiro Smart Fuse

gr8future, 2023-12-19:
### Link

https://tempiro.com/en/produkt/smart-fuse/

### Database entry

n/a

### Comments

Replace the standard plug holder with the Tempiro Smart Fuse. Tempiros controls your electric heating via the Smart Fuse, which has wireless radio communication.

The Smart Fuse contains a switch that can turn the power on and off in the fuse, which allows you to control electric heating, sauna, electric car charger or whatever you want.
It reads the energy consumption of each fuse so you always have full control of what is happening in your home.
Tempiro Smart Fuse uses Zigbee HA 1.2 radio protocol and is compatible with other Zigbee HA products.

I have tried to add this unsupported device for a while now but with no luck and for some reason im not able to open my database.db it says this is not a database when im trying with sqllite browser.

It got paired though after some tries, I have a Sonoff Zigbee Stick and I have tried to get the On/off to work to start with.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ZHA-SmartPlug 40'], 
    model: 'ZHA-SmartPlug 40', 
    vendor: 'Tem', 
    description: 'Smart Fuse', 
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    exposes: [e.switch()],
};

module.exports = definition;

I'm kind of new to this so all help will be appreciated.
```

gr8future, 2024-01-09:
With the latest Zigbee2MQTT update the ON/OFF function works :) but no power metering or battery report..

closed 2024-08-07 by github-actions[bot]

dankfred, 2024-11-14:
Hello, did somebody ever create a converter for this device? Thanks