Koenkk/zigbee2mqtt #20296: Channel change fails badly

AndKe, 2023-12-19:
### What happened?

I tried to change the ZigBee channel from 11 to 25 in WebUI:

The docker failed after the mandatory "restart"  with this error: 
```
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:27: - Channel List: configured=25, adapter=11
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:27: Please update configuration to prevent further issues.
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:27: If you wish to re-commission your network, please remove coordinator backup at /app/data/coordinator_backup.json.
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:27: Re-commissioning your network will require re-pairing of all devices!
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:27: Error while starting zigbee-herdsman
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:27: Failed to start zigbee
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:27: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:27: Exiting...
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:28: Error: startup failed - configuration-adapter mismatch - see logs above for more information
zigbee2mqtt  |     at ZnpAdapterManager.determineStrategy (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:229:35)
zigbee2mqtt  |     at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:63:26)
zigbee2mqtt  |     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
zigbee2mqtt  |     at Zigbee.start (/app/lib/zigbee.ts:60:27)
zigbee2mqtt  |     at Controller.start (/app/lib/controller.ts:101:27)
zigbee2mqtt  |     at start (/app/index.js:107:5)
zigbee2mqtt  | Using '/app/data' as data directory
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-19 16:42:38: Logging to console and directory: '/app/data/log/2023-12-19.16-42-38' filename: log.txt
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-19 16:42:38: Starting Zigbee2MQTT version 1.34.0 (commit #aae7312)
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-19 16:42:38: Starting zigbee-herdsman (0.25.0)
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:40: Configuration is not consistent with adapter state/backup!
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:40: - PAN ID: configured=6754, adapter=6754
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:40: - Extended PAN ID: configured=00124b0024c0f17d, adapter=00124b0022812e92
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:40: - Network Key: configured=XXXXXXXXXXXXXXXXXXXXXXXXXXXXX
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:40: - Channel List: configured=25, adapter=11
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:40: Please update configuration to prevent further issues.
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:40: If you wish to re-commission your network, please remove coordinator backup at /app/data/coordinator_backup.json.
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:40: Re-commissioning your network will require re-pairing of all devices!
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:40: Error while starting zigbee-herdsman
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:40: Failed to start zigbee
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:40: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:40: Exiting...
zigbee2mqtt  | Zigbee2MQTT:error 2023-12-19 16:42:41: Error: startup failed - configuration-adapter mismatch - see logs above for more information
zigbee2mqtt  |     at ZnpAdapterManager.determineStrategy (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:229:35)
zigbee2mqtt  |     at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:63:26)
zigbee2mqtt  |     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
zigbee2mqtt  |     at Zigbee.start (/app/lib/zigbee.ts:60:27)
zigbee2mqtt  |     at Controller.start (/app/lib/controller.ts:101:27)
zigbee2mqtt  |     at start (/app/index.js:107:5)
zigbee2mqtt  | Using '/app/data' as data directory

```

The docker and webUI fails after that, manual change of configuration.yaml was needed.

### What did you expect to happen?

channel change, painless

### How to reproduce it (minimal and precise)

change channel in WebUI 

### Zigbee2MQTT version

1.34.0 (commit #aae7312)

### Adapter firmware version

20230507

### Adapter

POE ethernet from HamGeek

### Debug log

see above.

chris-1243, 2023-12-19:
According to your log, it seems your adapter is not on channel 25 as configured in the web UI.

Any means to change it directly via a web interface (not Z2M) as it is a PoE device.

AndKe, 2023-12-19:
That's the thing, I see absolutely no way to change it using the ESP32's webpage:
[image]

Hence I would expect zigbee2mqtt to set this option, or at least not fail this badly if it is unable to affect the setting.

chris-1243, 2023-12-19:
At the top right, you have a tab (config) and an other one (tools). Have you tried there?

I do not use such adapter but as far as I know, you shiuld have access to some settings.

As a small fix, you may revert the change you did in configuration.yaml and set channel 11 again... not the best option but you may have a working network.

I do assume you followed the how to published on [https://www.zigbee2mqtt.io/guide/getting-started/#installation]()

Ip address and port are well configured in Z2M.

AndKe, 2023-12-19:
the IP is reserved in DHCP, it is configured and has been working fine for weeks. 
The only issue is with my attempt to change the channel to something with less traffic.

chris-1243, 2023-12-20:
Could you try to remove  /app/data/coordinator_backup.json

Before doing so, create a full copy of your /app/data directory.

(Stop your container and then make the backup)

Keep in mind, changing the network channel involve a re-commissioning of your devices

closed 2024-07-18 by github-actions[bot]