Koenkk/zigbee2mqtt #20298: Xiaomi SP-EUC01 - Firmware on 'xxxxxxxx' is newer than latest firmware online.

nlindblo, 2023-12-19:
### What happened?

Getting repeated error messages for all my Xiami wall plugs that current firmware is newer than latest. Current firmware in OTA is showing as "12-06-2021"

### What did you expect to happen?

No error message to occur and firmware being in line with vendors latest firmware

### How to reproduce it (minimal and precise)

Plugs paired as they came out of the box, nothing specific required to reproduce 

### Zigbee2MQTT version

1.34.0 commit: 56589dcc

### Adapter firmware version

20210708

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P

### Debug log

Dec 19 16:09:41 zigbee1.supernovagate.internal npm[26401]: Zigbee2MQTT:warn  2023-12-19 16:09:41: Firmware on '0x54ef4410007ffc1e' is newer than latest firmware online.

nlindblo, 2023-12-19:
Here are some further details on the versions installed vs available

{
    "auto_off": false,
    "button_lock": "OFF",
    "device_temperature": 29,
    "energy": 0.3,
    "linkquality": 84,
    "power": 4,
    "power_outage_memory": true,
    "state": "ON",
    "update": {
        "installed_version": 41,
        "latest_version": 32,
        "state": "idle"
    },
    "consumer_connected": null,
    "current": null,
    "led_disabled_night": false,
    "overload_protection": null,
    "update_available": null,
    "voltage": null
}

Akitash, 2024-07-09:
This issue still exist

closed 2025-01-09 by github-actions[bot]