Koenkk/zigbee2mqtt #20299: shows 200% battery

lukasjakobi, 2023-12-19:
### What happened?

After updating the IKEA TRADFRI ON/OFF Switch (https://www.zigbee2mqtt.io/devices/E1743.html#ikea-e1743) to version     24.4.6 and later swapping the battery, it started showing 200% of battery and continued to the current state of 180%

### What did you expect to happen?

I expected the battery to be capped at a maximum of 100%

### How to reproduce it (minimal and precise)

Update the switch to 24.4.6, Swap battery

### Zigbee2MQTT version

1.29.2 commit: bb3e8f6

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

[image]

lukasjakobi, 2023-12-19:
i just updated to [1.34.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.34.0) and the issue still exists

antti365, 2023-12-27:
@Koenkk I confirm there is such a problem, 1.34.0 commit: 56589dc. I also encountered it myself.

Koenkk, 2023-12-28:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-28 by Koenkk

jvuotila, 2024-09-10:
Still same problem with 1.40.1.