Koenkk/zigbee2mqtt #20303: [New device support]: Avatto TRV10 (Tuya based Thermostatic radiator valve)

MicSchi, 2023-12-19:
### Link

https://avatto-smart.com/product/smart-thermostat-radiator-valve/

### Database entry

{"id":17,"type":"EndDevice","ieeeAddr":"0xa4c138de77bf3426","nwkAddr":17795,"manufId":4417,"manufName":"_TZE200_rtrmfadk","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0006\u0000\u0000\u0000e\u0007\u0000\u0000\u0000\u0012","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_rtrmfadk","powerSource":3,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1703016958320,"defaultSendRequestWhen":"immediate"}

### Comments

Hello!

I followed the new device and referenced Tuya guide.
It is not completely new, but a modification of "TS0601_thermostat_1" in Tuya.ts
Basically the 6 devices I have were working but some features were not ok. 

I look in the debug logs for datapoints and as result:
I had to remove some, like "boost function" and e.g. the "position" is a different data point.
Preset contains the 4 modes the devices can operate in: auto - using weekly schedule, manual - using the current_heating_setpoint, off - value closed, on - value open
system_mode only auto, heat and off are applicable

What is missing is the synchronisation between preset and system_mode. I tried without system_mode, but then UI for thermostat was greyed out.
Also was not sure how to get the picture uploaded, I'm not familiar with git. 

[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const hy_set_time_request = {
    cluster: 'manuSpecificTuya',
    type: ['commandMcuSyncTime'],
    convert: async (model, msg, publish, options, meta) => {
        const OneJanuary2000 = new Date('January 01, 2000 00:00:00 UTC+00:00').getTime();
        const currentTime = new Date().getTime();
        const utcTime = Math.round((currentTime - OneJanuary2000) / 1000);
        const localTime = Math.round(currentTime / 1000) - (new Date()).getTimezoneOffset() * 60;
        const endpoint = msg.endpoint;
        const payload = {
            payloadSize: 8,
            payload: [
                ...tuya.convertDecimalValueTo4ByteHexArray(utcTime),
                ...tuya.convertDecimalValueTo4ByteHexArray(localTime),
            ],
        };
        await endpoint.command('manuSpecificTuya', 'mcuSyncTime', payload, {});
    },
};

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_rtrmfadk'},
    ],
    model: 'TS0601_thermostat_1',
    vendor: 'Tuya',
    description: 'Thermostatic radiator valve',
    whiteLabel: [
        tuya.whitelabel('AVATTO', 'TRV10', 'Thermostatic radiator valve with schedule', ['_TZE200_rtrmfadk']),
    ],
    onEvent: tuya.onEventSetLocalTime,
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.battery(), e.child_lock(), e.max_temperature(), e.min_temperature(),
        e.position(), e.window_detection(),
        e.binary('window', ea.STATE, 'CLOSED', 'OPEN').withDescription('Window status closed or open '),
        e.climate()
            .withLocalTemperature(ea.STATE).withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withLocalTemperatureCalibration(-30, 30, 0.1, ea.STATE_SET)
            .withPreset(['auto', 'manual', 'off', 'on'],
                'MANUAL MODE ☝ - In this mode, the device executes manual temperature setting. ' +
            'When the set temperature is lower than the "minimum temperature", the valve is closed (forced closed). ' +
            'AUTO MODE ⏱ - In this mode, the device executes a preset week programming temperature time and temperature. ' +
            'ON - In this mode, the thermostat stays open ' +
            'OFF - In this mode, the thermostat stays closed')
            .withSystemMode(['auto', 'heat', 'off'], ea.STATE)
            .withRunningState(['idle', 'heat'], ea.STATE),
        ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
    ],
    meta: {
        tuyaDatapoints: [
            [1,'system_mode', tuya.valueConverterBasic.lookup({'auto': tuya.enum(0), 'heat': tuya.enum(1), 'off': tuya.enum(2)})],
            [1, 'preset', tuya.valueConverterBasic.lookup({'auto': tuya.enum(0), 'manual': tuya.enum(1), 'off': tuya.enum(2), 'on': tuya.enum(3)})],
            [2,'current_heating_setpoint',tuya.valueConverter.divideBy10],  // devided by 10 is °C
            [3,'local_temperature',tuya.valueConverter.divideBy10], // devided by 10 is °C
            [6, 'running_state', tuya.valueConverterBasic.lookup({'heat': 1, 'idle': 0})],
            [8, 'window_detection', tuya.valueConverter.onOff],
            [12, 'child_lock', tuya.valueConverter.lockUnlock],
            [15, 'min_temperature', tuya.valueConverter.divideBy10],
            [16, 'max_temperature', tuya.valueConverter.divideBy10],
            [17, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
            [18, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
            [19, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
            [20, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [21, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [22, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [23, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
            [101, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration1],
            [108,'position',tuya.valueConverter.divideBy10],
        ],
    },
};

module.exports = definition;
```

MicSchi, 2023-12-28:
Hello again,

meanwhile I bought a gateway for the Smart Life App from Toya to be able to see all the data points available, which resulted in this output:

> {
> "1":"Mode",
> "2":"Target temperature",
> "3":"Current temperature",
> "6":"Working status",
> "7":"Window status",
> "8":"Open window",
> "12":"Child lock",
> "13":"Battery",
> "14":"Fault alarm",
> "15":"Min. limit temperature",
> "16":"Max. limit temperature",
> "17":"Heating schedule",
> "18":"Week program Tuesday",
> "19":"Week program Wednesday",
> "20":"Week program Thursday",
> "21":"Week program Friday",
> "22":"Week program Saturday",
> "23":"Week program Sunday",
> "101":"Room sensor calibration",
> "108":"Valve",
> "109":"Modell",
> "110":"Motor thrust",
> "111":"Display brightness",
> "112":"Software version",
> "113":"Screen orientation",
> "114":"System mode",
> "115":"Switch deviation (energy-saving mode only)",
> "116":"Motordaten"
> }

from the info found in the debug are of Tuya I updated the external converter defintion to that: 

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const hy_set_time_request = {
    cluster: 'manuSpecificTuya',
    type: ['commandMcuSyncTime'],
    convert: async (model, msg, publish, options, meta) => {
        const OneJanuary2000 = new Date('January 01, 2000 00:00:00 UTC+00:00').getTime();
        const currentTime = new Date().getTime();
        const utcTime = Math.round((currentTime - OneJanuary2000) / 1000);
        const localTime = Math.round(currentTime / 1000) - (new Date()).getTimezoneOffset() * 60;
        const endpoint = msg.endpoint;
        const payload = {
            payloadSize: 8,
            payload: [
                ...tuya.convertDecimalValueTo4ByteHexArray(utcTime),
                ...tuya.convertDecimalValueTo4ByteHexArray(localTime),
            ],
        };
        await endpoint.command('manuSpecificTuya', 'mcuSyncTime', payload, {});
    },
};

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
[... 65 lines cut ...]
            [111, 'display_brightness',tuya.valueConverterBasic.lookup({'high': 0, 'mid': 1, 'low': 2})],
            [112, 'sw_version',tuya.valueConverter.raw], // integer 0..65535
            [113, 'screen_orientation',tuya.valueConverterBasic.lookup({'up': 0, 'down': 2})],
            [114, 'operation_mode',tuya.valueConverterBasic.lookup({'comfort_mode': 0, 'Eco_mode': 1})],
            [115, 'switch_deviation_eco',tuya.valueConverter.divideBy10],
            [116, 'motor_data',tuya.valueConverter.raw], // string length max 255
        ],
    },
};

module.exports = definition;

```

I had to comment out the expose for the datapoints because it is missing converters toZigbee. 

> 2023-12-28 19:10:30Received MQTT message on 'zigbee2mqtt/BZ_thermostat/set' with data  {"operation_mode":"comfort_mode"}'
> Error 2023-12-28 19:10:30No converter available for 'operation_mode' ("comfort_mode")

Likely it is not very complicated to solve this, if you know were to look. But being new to the whole home automation topic, I haven't found that place yet. Can someone assist me here, please?

PS: in the zigbee2mqtt log the datapoints seem to work for the TRV10:

> Debug 2023-12-28 20:30:06Received Zigbee message from 'WZ_thermostat', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":6}],"seq":8192}' from endpoint 1 with groupID null
> Info 2023-12-28 20:30:06MQTT publish: topic 'zigbee2mqtt/WZ_thermostat', payload '{"battery":100,"child_lock":"UNLOCK","current_heating_setpoint":16,"display_brightness":"high","fault":0,"last_seen":"2023-12-28T20:30:06+01:00","linkquality":255,"local_temperature":22.9,"local_temperature_calibration":-1,"max_temperature":30,"min_temperature":5,"model":"zt3l_60x\u0000","motor_data":"558, 2575, 0, 216","motor_thrust":"middle","operation_mode":"Eco_mode","position":100,"preset":"auto","running_state":"idle","schedule_friday":"05:50/21.0 07:30/16.0 13:00/21.0 21:30/16.0","schedule_monday":"05:50/21.0 07:30/16.0 13:00/21.0 21:30/16.0","schedule_saturday":"05:50/21.0 07:30/20.0 13:00/21.0 21:30/18.0","schedule_sunday":"05:50/21.0 07:30/20.0 13:00/21.0 21:30/16.0","schedule_thursday":"05:50/21.0 07:30/16.0 13:00/21.0 21:30/16.0","schedule_tuesday":"05:50/21.0 07:30/16.0 13:00/21.0 21:30/16.0","schedule_wednesday":"05:50/21.0 07:30/16.0 13:00/21.0 21:30/16.0","screen_orientation":"up","sw_version":1001,"switch_deviation_eco":0.5,"system_mode":"heat","window":"CLOSE","window_detection":"ON","window_state":"closed"}'

MicSchi, 2023-12-28:
It seems the device is already being worked on in latest sources https://github.com/Koenkk/zigbee-herdsman-converters/blob/c3555af2c37c8b1c142c23d8b287abc0f13bdb99/src/devices/tuya.ts#L3233C52-L3233C68
Maybe the additional infos provided above can be helpful.

Anashost, 2024-01-05:
@MicSchi  I've just got this same TRV a week ago, it was working somewhat with `v1.34.0-1` then stopped working in `v1.35.0-1` until i used this [Ext converter](https://gist.github.com/Koenkk/388387067ac18ba01369dfb898b72a2b) from this issue #20476.

But i still don't like how it works, like the `off` mode doesn't seem to turn off the TRV, instead i have to set it to a low temp like 5C to turn it off. The modes are also kinda strange in z2m.

Maybe It's a good idea to add it as a new device.

MicSchi, 2024-01-06:
Does these issues mean there are several different devices out their with fingerprint "_TZE200_rtrmfadk" but behaving differently? If so, how to distinguish between those to add a new device?

I'm using 1.35.0 of Zigbee2MQTT and my external converter since some days. My TRV10s are still working like I expect them to work. I use the hvac_mode to switch TRVs off if a window opens and switch back to heat when windows are closed again. And I'm changing the setpoint_temperature for day and night times. This works fine so far. (I'm not using auto because I don't want the internal weekly schedule of the TRV to interfere with automation in Home Assistant for now.)

What is not fully working for me are the advanced settings. So I had to use the toya hub and app to change these. I can read them with Zigbee2MQTT and also write without an error. But changing their value does not have an effect on the TRVs.

b3nJ0E, 2024-02-10:
I am struggling with the Arvatto TRV in better thermostat. Anyone got a this to work?
When I change the temp in BT the thermostat jumps to 35 degrees and than to 15 degrees. All within a second.

Fragmon, 2024-03-23:
+1 I  buy this thermostat too and it heats a lot. I use better thermostat but the controll dont work exactly. The integrated schedule on the thermostat cant be deleted in zigbee2mqtt UI.

mistraller, 2024-04-09:
> Does these issues mean there are several different devices out their with fingerprint "_TZE200_rtrmfadk" but behaving differently? If so, how to distinguish between those to add a new device?
>
I think you are right. I have a MOES TRV with the same fingerrprint. I use this quirk in ZHA and it works. Not everything but I can set and read temperatures, valve position etc. https://github.com/jacekk015/zha_quirks/blob/main/ts0601_trv_rtitek.py
I know that ZHA is not Z2M but it draw my attention that different suppliers seem to use same code.

closed 2024-11-07 by github-actions[bot]

radurambet, 2024-12-12:
It seems that there is another AVATTO TRV10 with a different fingerprint :  _TZE200_uf1qujxj

Maybe it helps to someone the firmware is kind of similar to other Tuya TRVs from the TS0601 family but I could make it work.

Using the clasical setpoints it doesnt work : Battery Level , Window Open , and error state. 
It DOES work : everything else including TRV position and comfort mode that is needed for continous adjustments of TRV position . If confort mode is not selected it works as on/off TRV.

So for me it is decent as it was unsuported and with this converter I made it work.

`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const thermostatSystemModeAndPreset = (toKey) => {
    return {
        from: (v) => {
            utils.assertNumber(v, 'system_mode');
            const presetLookup = {0: 'auto', 1: 'manual', 2: 'off', 3: 'on'};
            const systemModeLookup = {0: 'auto', 1: 'auto', 2: 'off', 3: 'heat'};
            return {preset: presetLookup[v], system_mode: systemModeLookup[v]};
        },
        to: (v) => {
            const lookup = toKey === 'preset' ? {'auto': 0,'manual': 1,'off': 2,'on': 3} : {'auto': 0,  'auto': 1,  'off': 2,  'heat': 3};
            return utils.getFromLookup(v, lookup);
        }
    }
}

const definition = {
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_uf1qujxj',
        },
    ],

    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_uf1qujxj',
    description: 'Thermostatic radiator valve.',
    onEvent: tuya.onEventSetLocalTime,
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.battery(), e.child_lock(), e.max_temperature(), e.min_temperature(),
        e.position(), e.window_detection(),
        e.binary('window', ea.STATE, 'OPEN', 'CLOSE').withDescription('Window status closed or open '),
        e.binary('alarm_switch', ea.STATE, 'ON', 'OFF').withDescription('Thermostat in error state'),
        e.climate()
            .withLocalTemperature(ea.STATE).withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withLocalTemperatureCalibration(-30, 30, 0.1, ea.STATE_SET)
            .withPreset(['auto', 'manual', 'off', 'on'],
                'MANUAL MODE ☝ - In this mode, the device executes manual temperature setting. ' +
                'When the set temperature is lower than the "minimum temperature", the valve is closed (forced closed). ' +
                'AUTO MODE ⏱ - In this mode, the device executes a preset week programming temperature time and temperature. ' +
                'ON - In this mode, the thermostat stays open ' +
                'OFF - In this mode, the thermostat stays closed')
[... 20 lines cut ...]
            [13, 'battery', tuya.valueConverter.raw],
            [14, 'alarm_switch', tuya.valueConverter.onOff],
            [15, 'min_temperature', tuya.valueConverter.divideBy10],
            [16, 'max_temperature', tuya.valueConverter.divideBy10],
            [17, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
            [18, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
            [19, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
            [20, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [21, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [22, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [23, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
            [101, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration1],
            [108, 'position', tuya.valueConverter.divideBy10],
            [111, 'display_brightness', tuya.valueConverterBasic.lookup({'high': tuya.enum(0), 'medium': tuya.enum(1), 'low': tuya.enum(2)})],
            [113, 'screen_orientation', tuya.valueConverterBasic.lookup({
                'up': tuya.enum(0), 'right': tuya.enum(1), 'down': tuya.enum(2), 'left': tuya.enum(3),
            })],
            [114, 'mode', tuya.valueConverterBasic.lookup({comfort: tuya. enum(0), eco: tuya.enum(1)})],
        ],
    },
};

module.exports = definition;

`