Koenkk/zigbee2mqtt #20306: Philips 5047130P9

chris-1243, 2023-12-19:
### What happened?

On the latest dev branch, I unfortunately get this error:

error 2023-12-19 23:00:22: Publish 'get' 'color_temp' to 'Enzo lit' failed: 'Error: Read 0x0017880108c6ceb8/11 lightingColorCtrl(["colorMode","currentX","currentY","enhancedCurrentHue","currentSaturation","colorTemperature"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
error 2023-12-19 23:01:17: Publish 'get' 'color_temp' to 'Enzo plafonnier' failed: 'Error: Read 0x0017880109aebafe/11 lightingColorCtrl(["colorMode","currentX","currentY","enhancedCurrentHue","currentSaturation","colorTemperature"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 33270 - 11 - 9 - 768 - 1 after 10000ms)'

### What did you expect to happen?

To have the same behavior as 1.33.2 or 1.34.0 :latest-dev (before 17.12.2023)

### Zigbee2MQTT version

1.34.0 latest-dev

### Adapter firmware version

20231112

### Adapter

Sonoff Dongle-P

Koenkk, 2023-12-20:
Can you try reading the attributes one by one via the z2m frontend -> dev console -> read from lightingcolorctrl cluster attributes `colorMode","currentX","currentY","enhancedCurrentHue","currentSaturation","colorTemperature`, which one fails?

chris-1243, 2023-12-20:
Below are the results:

"enhancedCurrentHue", "currentSaturation" are generating the errors

error 2023-12-20 21:23:36: Publish 'set' 'read' to 'Enzo lit' failed: 'Error: Read 0x0017880108c6ceb8/11 lightingColorCtrl(["enhancedCurrentHue"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

error 2023-12-20 21:25:15: Publish 'set' 'read' to 'Enzo lit' failed: 'Error: Read 0x0017880108c6ceb8/11 lightingColorCtrl(["currentSaturation"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

Koenkk, 2023-12-21:
Thanks, should be fixed now!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-21 by Koenkk

chris-1243, 2023-12-21:
It may not be the right place to ask but any major code change regarding Philips light ?

I have a bunch of them and different models. I need to provide you the errors and which attributes but I have been able to reproduce the same behaviour, same field with other lights.

The errors are not presents on version 1.33.2.

Koenkk, 2023-12-22:
Which other lights?

Koenkk, 2023-12-22:
I pushed a correct fix now (also for non Philips lights)

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

chris-1243, 2023-12-22:
It seems to be solved with the latest-dev.

Thanks for you support