Koenkk/zigbee2mqtt #20321: Hue Tap Dial converter is missing a cluster

phbasler, 2023-12-20:
### What happened?

The [Hue Tap Dial](https://www.zigbee2mqtt.io/devices/8719514440937_8719514440999.html) only reports the actions `dial_rotate_left_slow`, `dial_rotate_left_fast` and `dial_rotate_left_step`.

In ZHA the amount by which the dial was rotated is directly reported in the event trigger and enables smooth light dimming. This is sadly not the case here.

Rotating the dial with debug logging enabled yields the message

~~~
 2023-12-21 00:07:33Received Zigbee message from 'Wickeltisch Dial', type 'commandStepWithOnOff', cluster 'genLevelCtrl', data '{"stepmode":1,"stepsize":8,"transtime":4}' from endpoint 1 with groupID 0
debug 2023-12-21 00:07:33No converter available for '8719514440937/8719514440999' with cluster 'genLevelCtrl' and type 'commandStepWithOnOff' and data '{"stepmode":1,"stepsize":8,"transtime":4}'
~~~

Which seems like the `stepsize` is recognized by Z2M but no [converter](https://github.com/Koenkk/zigbee-herdsman-converters/blob/3de732a483e789ab4b81184895245a79d9657ced/src/devices/philips.ts#L2846-L2872) for the cluster.
This was already reported in #15761 but got dropped due to being stale.

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff ZBDongle-E

phbasler, 2023-12-20:
I found the type [commandStepWithOnOff](https://github.com/Koenkk/zigbee-herdsman-converters/blob/3de732a483e789ab4b81184895245a79d9657ced/src/converters/fromZigbee.ts#L1335) in the `fromZigbee.ts`. Is there a way to include it in the dial converter?

Koenkk, 2023-12-21:
Do you get any more `No converter available for` messages?

phbasler, 2023-12-21:
After enabling debug in the log option and turning the dial I get 

~~~
debug 2023-12-21 22:04:55Received Zigbee message from 'Wickeltisch Dial', type 'commandStepWithOnOff', cluster 'genLevelCtrl', data '{"stepmode":0,"stepsize":26,"transtime":4}' from endpoint 1 with groupID 0
debug 2023-12-21 22:04:55No converter available for '8719514440937/8719514440999' with cluster 'genLevelCtrl' and type 'commandStepWithOnOff' and data '{"stepmode":0,"stepsize":26,"transtime":4}'
debug 2023-12-21 22:04:55Received Zigbee message from 'Wickeltisch Dial', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":20,"time":60,"type":1,"unknown1":3145984,"unknown2":0}' from endpoint 1 with groupID 0
info 2023-12-21 22:04:55MQTT publish: topic 'zigbee2mqtt/Wickeltisch Dial', payload '{"action":"dial_rotate_right_slow","action_direction":"right","action_time":60,"action_type":"step","battery":100,"brightness":200,"linkquality":144,"update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null}'
info 2023-12-21 22:04:55MQTT publish: topic 'zigbee2mqtt/Wickeltisch Dial', payload '{"action":"","action_direction":null,"action_time":null,"action_type":null,"battery":100,"brightness":200,"linkquality":144,"update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null}'
info 2023-12-21 22:04:55MQTT publish: topic 'zigbee2mqtt/Wickeltisch Dial', payload '{"action":"","action_direction":null,"action_time":null,"action_type":null,"battery":100,"brightness":200,"linkquality":144,"update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null}'
info 2023-12-21 22:04:55MQTT publish: topic 'zigbee2mqtt/Wickeltisch Dial', payload '{"action":"","action_direction":null,"action_time":null,"action_type":null,"battery":100,"brightness":200,"linkquality":144,"update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null}'
info 2023-12-21 22:04:55MQTT publish: topic 'zigbee2mqtt/Wickeltisch Dial', payload '{"action":"","action_direction":null,"action_time":null,"action_type":null,"battery":100,"brightness":200,"linkquality":144,"update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null}'
info 2023-12-21 22:04:55MQTT publish: topic 'zigbee2mqtt/Wickeltisch Dial/action', payload 'dial_rotate_right_slow'
info 2023-12-21 22:04:55MQTT publish: topic 'zigbee2mqtt/Wickeltisch Dial/action', payload 'dial_rotate_right_slow'
info 2023-12-21 22:04:55MQTT publish: topic 'zigbee2mqtt/Wickeltisch Dial/action', payload 'dial_rotate_right_slow'
info 2023-12-21 22:04:55MQTT publish: topic 'zigbee2mqtt/Wickeltisch Dial/action', payload 'dial_rotate_right_slow'
~~~

So only this no converter message

Koenkk, 2023-12-22:
Added it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-22 by Koenkk

phbasler, 2023-12-22:
Thanks, I will test it out tomorrow :)

phbasler, 2023-12-23:
Works like a charm, thanks again