Koenkk/zigbee2mqtt #20325: Moes AM43-0.45/40-ES-EB (_TZE200_zah67ekd)

illuzn, 2023-12-21:
### What happened?

This is for Moes AM43-0.45/40-ES-EB (_TZE200_zah67ekd).

1. DPS 1 is used for multiple purposes.
   - setting DPS 1 to 0, 1 or 2 will open, stop or close the cover respectively.
   - reading DPS 1, it will report 1 when the motor is stopped but otherwise will not report.
  This should be combined with DPS 7 which is read only.
  DPS 7 is 0 for opening and 1 for closing.
  I unsuccessfully tried to attempt correcting this myself. Am I correct in saying that Zigbee2MQTT doesn't allow combining 2 DPS in the way described above. I think this is what was missed when originally writing this converter (and why DPS 7 is currently ignored).
2. (Less importantly) The device incorrectly reports as mains powered (it is battery powered). See here this extract: [Li-battery Charging: Powered by charger(adaptor is included) ensures long time continuous working with the large battery capacity.](https://moeshouse.com/products/zigbee-chain-motorized-blinds-motor). I can't for the life of me find the DPS where it reports battery life (or the ambient light using the sensor).

### What did you expect to happen?

The device to properly report opening/ stopped/ closing.

### Zigbee2MQTT version

 1.34.0 

### Adapter firmware version

20230507

### Adapter

Tubez USB Zigbee CC2652

Aepx, 2024-01-20:
To touch on this...
I have other blind motors and the state on them reports the actual current state of the motor.
So if I open the blinds, the state reports as open.
Close the blinds, reports as closed.
Stop the blinds, reports as stopped.
This motor seems to report as stopped even when they are open or closed.
I can see the logic in it, when I open the blinds, the motor runs to the open position then stops there. Hence the stop state, but the blinds are opened so the state should also be open, shouldn't it?
The way the other motors I have work is if the blinds go to the open position, the state is open.
If I open the blinds and stop them halfway, the state shows stop.
I believe that is the way they should work. 
Fully open, state Open.
Fully closed, state Close.
Stopped anywhere in between, state Stop to signify they are not "fully" either way.

illuzn, 2024-01-21:
When you say other blind motors do you mean this specific model or another model?

Aepx, 2024-01-21:
I mean another model.
I have 4 of this type, they all show the state as "stop" even when they're open.
The other model shows the state correctly (in my opinion) as "open" when they're open, "close" when closed.

illuzn, 2024-01-22:
I think the open/ closed state should be derived from position which makes more sense.

DPS1 tells us whether the motor is moving.
DPS2 tells u which direction the motor is moving (if it is moving at all).

I wish my java was upto snuff and I'd do myself but I can't get my custom quirk that attempts to implement this working.

bubez81, 2024-11-15:
there are news about the battery status?

bubez81, 2024-12-25:
nothing about the battery?

bubez81, 2025-02-24:
up

bubez81, 2025-04-26:
Not solved

Stuartyboy75, 2025-11-20:
> I have two of those as well and they are marked as powered as well even they are using batteries

I had two of these and showed Mains (single phase) when the solar/sensor is plugged in. I have now added another two which just say battery