Koenkk/zigbee2mqtt #20329: Aqara Radiator Thermostat E1 not executing commands

Alfy1080, 2023-12-21:
### What happened?

Hello.

Got a weird issue recently. Sometimes when I set the target temperature on my Aqara TRVs, the command goes through, i can see in Z2M and in Home assistant that the target temperature was set to the desired one, my automation turns on the central heating as expected but the TRV valve does not open at all even though it should. Restarting Z2M seems to not help in this situation and the only thing that made the TRVs actually open the valve was waking them up by pressing the physical button. When i press the button the TRV immediately opens the valve as appropriate. Not sure if this has something to do with Z2M or with the TRVs firmware though.

### What did you expect to happen?

When i set a target temperature that is above the ambient temperature i expect the valve to open immediately, without any physical interaction with the TRV.

### How to reproduce it (minimal and precise)

It happens randomly. for me it happened a few times these last 2 months. A friend of mine with the same setup had this issue several times this month. Not sure how to replicate it.

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20230507

### Adapter

Sonoff ZB Dongle-P (CC2652P)

### Debug log

[logs.zip](https://github.com/Koenkk/zigbee2mqtt/files/13739812/logs.zip)

Alfy1080, 2023-12-21:
Forgot to mention that the living_room_TRV was set to 30 degrees Celsius at 07:01 PM on 21 Dec and stayed like that (with the valve closed) until 11:14 PM on the same day when i've noticed and pressed the TRV's button to get it to work again. So maybe there is something relevant to the issue in the logs around that timeframe, but i couldn't find anything that might indicate an issue. 

Other potentially relevant information regarding my setup: 

* I use external temperature sensors with the TRVs and i publish the temperature via home assistant automation with the following action:

service: mqtt.publish
data:
  topic: zigbee2mqtt/Living_Room_TRV/set/sensor_temp
  payload: >-
    {{
    (states('sensor.living_room_ambient_sensor_temperature')|float(0)|round(1))
    }}

* I had times when the sensor mode was changing by itself from external to internal, so i've created a home assistant automation that changes the sensor mode back to external every 10 minutes if the sensor mode is not external like so:

service: mqtt.publish
data:
  topic: zigbee2mqtt/Living_Room_TRV/set
  payload: "{\"sensor\": \"external\"}"

* I use virtual thermostats to set the desired temperature and an automation that sets the TRVs to 30 degrees Celsius (so that the valve would open completely) when the target temperature of that TRV is higher than the ambient temperature. However the interaction with the TRV is a basic climate set target temperature so I don't see why this would cause any issues.

Alfy1080, 2024-01-02:
TRV got stuck again today. Pressing its display opened the valve as usual. Logs attached.
[log (1).txt](https://github.com/Koenkk/zigbee2mqtt/files/13813359/log.1.txt)

Alfy1080, 2024-01-30:
Had a few more occurrences of this issue since i last posted so i decided to try workarounds.
I ended up modifying my whole heating automations so that i no longer need to publish the external sensor temperature data to the trv. Now the trv has its own temperature and the sensor mode is set to internal and whenever there is no need for an increase in temperature, the trv is set to off so that the valve would not open if the ambient temperature detected by the trv is lower than its target temperature (the trv's temperature while not heating is always much lower than the external ambient sensor's temperature). 
Ever since i did that change i no longer had the issue of the trv valve getting stuck, so i assume that the issue may be a bug with the external sensor's data. Will report back if this happens again.

closed 2024-08-28 by github-actions[bot]

vistvanv, 2024-11-28:
> Had a few more occurrences of this issue since i last posted so i decided to try workarounds. I ended up modifying my whole heating automations so that i no longer need to publish the external [...]

@Alfy1080 could you share this automation?

Alfy1080, 2024-11-28:
Hi @vistvanv 

I will gladly share the automations with you, however, since i made these almost a year ago and blocked them out of my memory, i barely remember exactly what and how i changed. 

My current system is fairly complex taking into account room presence, window state, trv internal temperature at the time of opening and 1 hour after opening and so on. 
I will attach here all the automation templates that i shared with a friend for him to just replace the dummy entities with his own. Just so you know, my setup also requires several additional entities like timedate helpers, generic thermostats, better thermostat entities and boolean entities. 

[Generic Climate Control.zip](https://github.com/user-attachments/files/17945339/Generic.Climate.Control.zip)

In this archive you will find the following automations:

Generic_7_Room_TRV_Control - this automation allows setting up the temperature either via the better thermostat entity in the home assistant UI or via the physical trv. it will sync the target temps between these entities. Also whenever the better thermostat entity calls for heat, the physical TRV will be set to 30 degrees, this is for the room to heat up faster.

Generic_TRV_State_Control - this automation controls the hvac mode of the climate entities based on window contact sensors

Generic_TRV_Valve_State_Monitoring - this automation keeps track of the TRV valve status, valve opening time and
  initial temperature of the TRV when the valve was opened. The data recorded by this automation is what i use in the main thermostat's automation as failsafes to turn off the central heating in cases where the trv's temperature remains unchanged for too long while calling for heat.

Generic_Room_Presence_Tracking - this automation keeps track of when was a room populated last and it is used with the main thermostat control automation to prevent heating an empty room even if the thermostat calls for heat.

Generic_Main_Thermostat_Control - this is what controls the main thermostat of the boiler. it turns it on if needed and turns it off whenever all rooms target temps are satisfied or if there is a problem with the TRVs (trv internal temperature did not increase for one hour since the valve was supposedly opened)

You will also find an Entity List file which basically contains a list of all the entities that are needed for these automations to work. The names of the entities in this list are the same as the dummy entities in the generic automations so after you created your own entities you can just find and replace each entity in the automations at the same time (ctrl+f and replace all). All automations are currently tailored for 7 room setups but you can adjust that by removing/duplicating the actions according to your needs.

Also FYI, I eventually fixed my initial issue of the TRV getting stuck by replacing my Home Assistant SkyConnect zigbee stick with a Sonoff Dongle-P and re-pairing all my zigbee devices to the Sonoff. However one of my friends has this issue on Sonoff Dongle-P so your mileage may vary :))

I hope this helps. :)

vistvanv, 2024-11-28:
oh wow, thanks for all the detailed instructions

my setup is somewhat similar to yours besides the presence detection. 
The issue i am facing many times is when the generic thermostat is asking for heat for a room, and I want to sync the external temp to the TRV, it doesn't react, even if the trv was last seen seconds ago. so i either have to restart z2m or repair the TRV and then it works for a while, but i wasn't able to figure out why this happens

Alfy1080, 2024-11-28:
> oh wow, thanks for all the detailed instructions
>
> my setup is somewhat similar to yours besides the presence detection. The issue i am facing many times is when the generic thermostat is asking for [...]

This sounds like the same issue that i used to have while i used the SkyConnect as a zigbee coordinator. Only in my case nothing made the TRV work again besides pressing the physical button on it. Not even full Z2M restart did the trick. 

What zigbee coordinator are you using? Is it flashed with the latest firmware? Are you using the latest Z2M version? What is the current transmit power of your coordinator? Do you have any zigbee routers in your network?

All of my zigbee network issues were solved by either changing the zigbee stick, updating the firmware or z2m version, increasing the transmit power or adding extra zigbee routers/connecting the problematic zigbee client through a different zigbee router.

Alfy1080, 2024-11-28:
I think it's a safe bet that the issue is caused by some compatibility weirdness with the skyconnect since we basically had the same setup (skyconnect, good coverage, latest firmware and z2m) and the thing that magically solved my TRV issue was swapping the skyconnect for Sonoff.

For the coordinator upgrade it depends on what are your needs, for me the best option was the Sonoff Dongle-P because it is under the recommended hardware on z2m's page, it has an external antenna so the coverage is much better than the skyconnect, and everyone that i know who has a smart home said to get rid of the skyconnect and go Dongle-P :)
Also since you already have a skyconnect you don't have to worry about the Dongle-P not being compatible with Matter or Thread because the skyconnect already is, so you could use the Dongle-P as the main zigbee network for everything zigbee and the skyconnect for matter/thread if you need it. Personally i don't so i just gave it away.

I guess if you need the coordinator to sit in a different room than the one where your HA server is you could go with a network coordinator with the CC2652P chip, the experience should be pretty much the same.

See here the list of options: https://www.zigbee2mqtt.io/guide/adapters/zstack.html