Koenkk/zigbee2mqtt #20331: Hive TRV (UK7004240) max setpoint should be 32C not 35C

DavidGeorge528, 2023-12-21:
### What happened?

It appears that the max supported value for the Hive TRV (UK7004240) is 32C and not 35C. 35C is the current max you can set the setpoint value to. If you set the setpoint value above 32C nothing happens on the device and eventually, it syncs back to the previous value

Also, on the TRV itself, when turning the dial, the maximum it can be set to is 32C

### What did you expect to happen?

Setting the setpoint to any value within the supported range should update the TRV to that value

### How to reproduce it (minimal and precise)

Change the setpoint on the Hive TRV (UK7004240) to any value above 32C, observe the value does not change and will eventually sync back to the previous value

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20210708

### Adapter

Sonoff ZBDongle-P

Koenkk, 2023-12-22:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-22 by Koenkk

robertalexa, 2024-01-01:
@Koenkk just wanted to mention that the danfoss actually supports values up to 35. Hive however, who run their own flavour of the firmware only support 32. This used to be the case in z2m before all devices were consolidated as 1 source, in this case danfoss. Probably not a deal breaker for danfoss and popp trv users, as it is unrealistic that anyone sets temps so high, but will cause issues if they set the temps manually via the trv itself.

Koenkk, 2024-01-02:
> but will cause issues if they set the temps manually via the trv itself.

What issues will it cause?

robertalexa, 2024-01-02:
If the temperature is set to 35 from the TRV itself, but the range is up to 32, it will give an unsupported value, especially in HA. If done the other way around, you get the behaviour described by OP in this thread

I've always wondered if Hive TRV should have stayed separate from Danfoss Ally and Popp TRV in terms of the conververter. Hive TRV has different values, some attributes are not supported, OTA is not supported either.

Koenkk, 2024-01-02:
I see, fixed it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.