Koenkk/zigbee2mqtt #20333: Interview failed on pairing

gecko17, 2023-12-21:
### What happened?

I read multiple issues on github and searched everywhere for a solution, but I could not fix my issue.
My Popp TRV did not respond to commands like setpoint changes anymore. So I resetted the TRV and tried to re-pair it.

However, I am not able to pair now and always getting the error, that the Interview failed.
What is interesting is, that it pops up up to three times that the device wants to pair.

I am attaching the herdsman logs.

I tried pairing a lot of times, I tried restarting the controller, restarting hass, restarting z2m, changing batteries of the device, resetting the device. Nothing helped to fix the issue.

Any ideas how I can fix this?

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff ZBDongle-E

### Debug log

[herdsman.logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/13742994/herdsman.logs.txt)

Mikkooooo, 2023-12-21:
I see that you have the Sonoff dongle E too.
I have seen the same behaviour - somtetimes it takes several tries to pair the POPP TRV‘s.
But after pairing the network is stable until now.
I think it‘s caused in the Dongle E.
I don‘t know which is the recommended FW of the dongle, I‘ve also 6.10.3.0.
The latest are 7.xx   .

gecko17, 2024-01-08:
Can anyone help by investing the logs? Any ideas why the interview fails? I really tried everything except for completely resetting the coordinator.

Is there any solution except for resetting/flashing the coordinator and setting up my whole zigbee network completely from scratch? This would be a mess because I would need to re-pair some flush-mounted actors.

Additional question to @Mikkooooo: Where can I find the latest 7.xx firmware? Official docs still points to the 6.10.3 version https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/NCP.

gecko17, 2024-01-10:
@Koenkk I tried to do some debugging with the logs and found out the following information. Maybe you can help?

Active endpoints are requested in this line: https://github.com/Koenkk/zigbee-herdsman/blob/00f2eca9f73306fcda50895e2448169f7fad4cf1/src/controller/model/device.ts#L609C4-L609C4

And for me it is running in to the catch block at line 612.

So, I checked the activeEndpoints function at: https://github.com/Koenkk/zigbee-herdsman/blob/00f2eca9f73306fcda50895e2448169f7fad4cf1/src/adapter/ezsp/adapter/ezspAdapter.ts#L391

From there I moved to the zdoRequest function of the driver at: https://github.com/Koenkk/zigbee-herdsman/blob/00f2eca9f73306fcda50895e2448169f7fad4cf1/src/adapter/ezsp/driver/driver.ts#L561

I noticed in the logs that I got a message "ZDO Active_EP_req params: {"dstaddr":24853}", so line 565 has been executed.

However I do not get the log messages of lines 573 or 578. This means, that neither the response has been received successfully, nor a timeout has been reached and the "!res" condition has been met. So it seems like the request function (https://github.com/Koenkk/zigbee-herdsman/blob/00f2eca9f73306fcda50895e2448169f7fad4cf1/src/adapter/ezsp/driver/driver.ts#L445) throws an exception?

However, there is a surrounding try catch block in it and I also don't get the log message from the catch handler, so I am confused what exactly happens in the code. I am not seeing any of the logs that should be written after "ZDO Active_EP_req_params.

The only thing I get in the logs after the sendUnicast ist the messageSentHandler:
0x3f: {"_cls_":"messageSentHandler","_id_":63,"_isRequest_":false,"type":0,"indexOrDestination":24853,"apsFrame":{"profileId":0,"clusterId":5,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":40},"messageTag":48,"status":102,"message":{"type":"Buffer","data":[]}}

You can see the status is EMBER_DELIVERY_FAILED: The APS layer attempted to send or deliver a message, but it failed.

Can you help me with this information? Maybe a root cause could be, that it is not fetching the node descriptor and says it already got it? "Interview - skip node descriptor request for '0x0c4314fffe88953c', already got it".
The device was already paired, but I removed it and also tried to clear the database.

gecko17, 2024-01-12:
> > "Interview - skip node descriptor request for '0x0c4314fffe88953c', already got it".
> > The device was already paired, but I removed it and also tried to clear the database.
> 
> To clear this, force remove the device and restart z2m afterwards, then it will redo the `Node Descriptor Request`.

So you mean during the pairing attempt, when the device appears in the device list, click on force remove? Because currently the device is not shown, because pairing always fails during interview. I think I tried that in the past, but I will try again and investigate the logs this time.

gecko17, 2024-01-15:
> So you mean during the pairing attempt, when the device appears in the device list, click on force remove? Because currently the device is not shown, because pairing always fails during interview. I [...]

This is not working. I tried to pair, after the interview failed I force removed the device before it left the network. Then I tried to pair again, but it still says in the logs that it skips the node descriptor.

gecko17, 2024-01-15:
I would have been interested in what was causing the actual issue, but after upgrading to the new coordinator firmware 7.3.1 and also upgrading to zigbee2mqtt 1.35.1, the issue has been resolved. I could now pair the TRV without any issues.