Koenkk/zigbee2mqtt #20336: Phillips hue light 'announcing itself' and turning itself back on after command to turn off via a group

LukeDefeo, 2023-12-21:
### What happened?

I have a group of lights in my bedroom, one of them is a hue but called bedroom main. When this light is in a group and I turn off via the group all of the lights turn off and then a second later the bedroom main light turns on again. In the logs I can see the devide has announced itself and is reconnecting. Not sure what to make of this.

### What did you expect to happen?

I expect the lights to stay off

### How to reproduce it (minimal and precise)

Group this Phillips hue light along with other lights, I have tried multiple groups to see and I get the same behaviour. This is the model https://www.zigbee2mqtt.io/devices/9290012573A.html#philips-9290012573a

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20210708

### Adapter

son off stick p

### Debug log

nfo  2023-12-21 16:53:58: MQTT publish: topic 'zigbee2mqtt/Spare room side lights', payload '{"brightness":254,"color_mode":"color_temp","color_temp":317,"state":"ON"}'
info  2023-12-21 16:53:58: MQTT publish: topic 'zigbee2mqtt/Spare room desk lamp', payload '{"brightness":254,"color_mode":"color_temp","color_temp":317,"level_config":{"on_level":"previous"},"linkquality":87,"power_on_behavior":"previous","state":"ON","update":{"installed_version":269837568,"latest_version":537029168,"state":"available"},"update_available":true}'
info  2023-12-21 16:53:58: MQTT publish: topic 'zigbee2mqtt/Spare room shelf lamp', payload '{"brightness":254,"color_mode":"color_temp","color_temp":317,"level_config":{"on_level":"previous"},"linkquality":60,"power_on_behavior":"previous","state":"ON","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":false}'
debug 2023-12-21 16:53:58: Received MQTT message on 'zigbee2mqtt/Bedroom/set' with data '{"state":"OFF"}'
debug 2023-12-21 16:53:58: Publishing 'set' 'state' to 'Bedroom'
debug 2023-12-21 16:53:58: Received Zigbee message from 'Bedroom fairy lights', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2023-12-21 16:53:58: MQTT publish: topic 'zigbee2mqtt/Bedroom fairy lights', payload '{"linkquality":135,"power_on_behavior":null,"state":"OFF","update":{"installed_version":537019939,"latest_version":587765297,"state":"available"},"update_available":true}'
info  2023-12-21 16:53:58: MQTT publish: topic 'zigbee2mqtt/Bedside Ailis', payload '{"brightness":254,"color_mode":"color_temp","color_temp":316,"linkquality":96,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16786948,"latest_version":16786948,"state":"idle"},"update_available":false}'
info  2023-12-21 16:53:58: MQTT publish: topic 'zigbee2mqtt/Bedroom', payload '{"brightness":254,"color":{"h":32,"hue":32,"s":81,"saturation":81,"x":0.4574,"y":0.41},"color_mode":"color_temp","color_temp":316,"state":"OFF"}'
info  2023-12-21 16:53:58: MQTT publish: topic 'zigbee2mqtt/Bedside Luke', payload '{"brightness":254,"color_mode":"color_temp","color_temp":316,"linkquality":135,"power_on_behavior":null,"state":"OFF","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":false}'
info  2023-12-21 16:53:58: MQTT publish: topic 'zigbee2mqtt/Bedroom fairy lights', payload '{"linkquality":135,"power_on_behavior":null,"state":"OFF","update":{"installed_version":537019939,"latest_version":587765297,"state":"available"},"update_available":true}'
info  2023-12-21 16:53:58: MQTT publish: topic 'zigbee2mqtt/Klum', payload '{"brightness":254,"color_mode":"color_temp","color_temp":316,"linkquality":39,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16783874,"latest_version":16786948,"state":"available"},"update_available":true}'
info  2023-12-21 16:53:58: MQTT publish: topic 'zigbee2mqtt/Bedroom main', payload '{"brightness":254,"color":{"h":32,"hue":32,"s":81,"saturation":81,"x":0.4574,"y":0.41},"color_mode":"color_temp","color_temp":366,"color_temp_startup":366,"linkquality":75,"power_on_behavior":"on","state":"OFF","update":{"installed_version":16786948,"latest_version":16786948,"state":"idle"},"update_available":false}'
info  2023-12-21 16:53:58: MQTT publish: topic 'zigbee2mqtt/Bedroom side lights', payload '{"brightness":254,"color_mode":"color_temp","color_temp":316,"state":"OFF"}'
info  2023-12-21 16:53:58: MQTT publish: topic 'zigbee2mqtt/Bedroom 2', payload '{"brightness":254,"color":{"h":32,"hue":32,"s":81,"saturation":81,"x":0.4574,"y":0.41},"color_mode":"color_temp","color_temp":316,"state":"OFF"}'
debug 2023-12-21 16:54:03: Device 'Bedroom main' announced itself
info  2023-12-21 16:54:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Bedroom main","ieee_address":"0x001788010277c6b0"},"type":"device_announce"}'
info  2023-12-21 16:54:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Bedroom main"},"type":"device_announced"}'
debug 2023-12-21 16:54:05: Retrieving state of 'Bedroom main' after reconnect
debug 2023-12-21 16:54:05: Received Zigbee message from 'Bedroom main', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 11 with groupID 0
info  2023-12-21 16:54:05: MQTT publish: topic 'zigbee2mqtt/Bedroom', payload '{"brightness":254,"color":{"h":32,"hue":32,"s":81,"saturation":81,"x":0.4574,"y":0.41},"color_mode":"color_temp","color_temp":316,"state":"ON"}'
info  2023-12-21 16:54:05: MQTT publish: topic 'zigbee2mqtt/Bedroom main', payload '{"brightness":254,"color":{"h":32,"hue":32,"s":81,"saturation":81,"x":0.4574,"y":0.41},"color_mode":"color_temp","color_temp":366,"color_temp_startup":366,"linkquality":81,"power_on_behavior":"on","state":"ON","update":{"installed_version":16786948,"latest_version":16786948,"state":"idle"},"update_available":false}'
info  2023-12-21 16:54:05: MQTT publish: topic 'zigbee2mqtt/Bedroom 2', payload '{"brightness":254,"color":{"h":32,"hue":32,"s":81,"saturation":81,"x":0.4574,"y":0.41},"color_mode":"color_temp","color_temp":316,"state":"ON"}'

LukeDefeo, 2023-12-21:
Actually playing with this device some more I can reproduce it by just turning off this specific device, the command doesn't need to come to the group again it comes on about 2 seconds later. It happens about about 75% of the time

catduckgnaf, 2024-01-04:
I can confirm its the device and not groupings.

I believe this has to do with a recent firmware update on these bulbs. I am experiencing a similar issue, and I believe it is only for firmware  1.108.5. Can you confirm what firmware you are on. And can anyone with an older firmware confirm whether these bulbs are still having issues? I don't think we can downgrade firmware.

Let me know if you were able to solve this.

LukeDefeo, 2024-04-06:
firmware version is 1.108.7, Shame its firmware related and can't be downgraded.

The issue was very persistent for a while but now it very rarely happens. I guess I will just replace the bulb eventually

closed 2024-04-06 by LukeDefeo