Koenkk/zigbee2mqtt #20337: frient AQSZB-110  Hardware Version 1.1.0 Airquality sensor VOC

XDOM-JR, 2023-12-21:
### Link

https://frient.com/products/air-quality-sensor/

### Database entry

Pairing not working

### Comments

The friend Airquality sensor can't be paired with the zigbee2MQTT Dev Version

Device: https://zigbee.blakadder.com/Develco_AQSZB-110.html
Manufacter URL: https://frient.com/products/air-quality-sensor/ 
Manufacter: Develco
Supports: air quality, battery, humidity, temperature, voc

Error Zigbee2MQTT: 
Error 2023-12-21 19:59:40Failed to interview '0x0015bc0036001f1f', device has not successfully been paired
Info 2023-12-21 19:59:40MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc0036001f1f","ieee_address":"0x0015bc0036001f1f","status":"failed"},"type":"device_interview"}'

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['AQSZB-110'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'AQSZB-110', // Vendor model number, look on the device for a model number
    vendor: 'frient', // Vendor of the device (only used for documentation and startup logging)
    description: 'Frient temperature & humidity sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.temperature(), e.humidity(), e.voc()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

sodfenrir, 2024-01-03:
I have the exact same model, and the exact same issue. As I though it would'nt work, I gave it to a friend of mine (same installation, as I learn with him) and, at his place, the device got interviewed easily. 

Don't know what the problem is, to be honnest.

XDOM-JR, 2024-01-03:
Hello @sodfenrir 
It seems to be a problem that only occurs with specific Zigbee coordinator models. In my case, I use the Aeotec Z-Stick ZGA008. Which Zigbee controller are you using?

sodfenrir, 2024-01-03:
I have the Sonoff USB ZBDongle-E. I think that my friend has the SONOFF Zigbee 3.0 USB Dongle Plus. Will update if I'm wrong about that one.

XDOM-JR, 2024-01-03:
I believe I've identified the issue; I found multiple threads in forums with other devices experiencing the same problem when interacting with devices. In most cases, Zigbee coordinators/controllers with the [EFR32MG21] chip are involved. My device, the Aeotec Z-Stick, has this chip, and so does the ZBDongle-P, which features the chip from Texas Instruments

Source of pics: https://smarthomescene.com/blog/best-zigbee-dongles-for-home-assistant-2023/

[image]

**Difference between ZBDongle-E and ZBDongle-P**
[image]

Another factor that may have an impact is that I have the USB Dongle passed through ESXi to Zigbee2MQTT. Please let me know if your setup also involves virtualization, some UART Serial adapters used in the most of USB Zigbee coordinators have troubles with virtualization. Actually I order the ZBDongle-P to check if this solve my issue.

sodfenrir, 2024-01-03:
Since we're speaking about it, and based on your interesting infos, I'm trying new tests again. Nothing good for the moment.

No, my coordinator is directely connected to my pi. I tried changing port and serial in Zigbee2MQTT, to no avail.

Matthias-Beck, 2024-01-31:
Hi guys,
I had the same issue but with the sonoff dongle E as coordinator.
I could fix it with a firmware update of the sonoff dongle

XDOM-JR, 2024-01-31:
Hello @Matthias-Beck 

Can you please share which firmware is functional and where we can download it?

regards

Matthias-Beck, 2024-01-31:
@XDOM-JR 
I did it here with the web flasher. You can decide to install the current version 7.x or even switch to multiprotocol zigbee and thread ( multiprotocol needs re discover of zigbee devices)

https://dialedin.com.au/blog/sonoff-zbdongle-e-rcp-firmware

Matthias-Beck, 2024-01-31:
@XDOM-JR 
Yes, I also have this sensor. My coordinator is in the basement, and I did the interview in teh Room above it and afterwards I moved it to the destination. I encountered my Tuya plugs are not working as routers, so I put another in between and teh connection is now stable.
Here is the info from my Z2M device page:

[image]

XDOM-JR, 2024-01-31:
In my case the Devices can't be Joined because the Interview fails:
How you configure the Serial Port?
Here my conning:

[image]

The devices ist showed as unsupported:
[image]

Here another informations like Firmware(after update with the link you share) of the coordinator and Zigbee2MQTT

[image]

XDOM-JR, 2024-01-31:
Ok,
I will try to reset Z2M, I don't have other devices(no router or devices). Can you tell me witch FW you have exactly(its the last Multiprotokoll version)

XDOM-JR, 2024-01-31:
additional info about the Sensor:
frient AQSZB-110 HW: 1.1.0 FW Version: Unknow

kotope, 2024-05-28:
Got the same issue with Sonoff Dongle-E (EZSP v8 6.10.3.0).

XDOM-JR, 2024-07-02:
I change my Zigbee Coordinator to someone with the TI(Texas Instrument CC2652P) Chip and it works:

SONOFF Zigbee 3.0 USB Dongle Plus - https://www.amazon.de/dp/B09KXTCMSC?psc=1&ref=ppx_yo2ov_dt_b_product_details

[image]

closed 2025-01-09 by github-actions[bot]