Koenkk/zigbee2mqtt #20340: little endian or big endian?

bobuhito, 2023-12-22:
### What happened?

When I make the hex file in TI's Code Composer Studio, I get reversed bytes (I am most
focused on the bootloader-enabling bytes so I added spaces around it) in the
ending ccfg flash block:

:020000040005F5
:207FA80001800000FF8000100058FFFDF3F7FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7D
:207FC800FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF C5FE0DC5 FFFFFFFFFFFFFF00FFC5C50092
:187FE800FF00000000000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF92
:00000001FF

Your hex reverses these 4-byte words (your "CC2652RB_coordinator_20230507.hex"
bootloader bytes are C50DFEC5), but I am sure your code has been tested well and
therefore I suspect Code Composer Studio has this bug (but that also surprises me
since they should have tested this thoroughly...could their different JTAG
programming cause a byte reversal?).

Which has the bug, or am I going crazy?  I would bet this has already been discussed
and I missed it.

### Zigbee2MQTT version

all

### Adapter firmware version

CC2652RB_coordinator_20230507

### Adapter

Electrolama CC2652RB

Koenkk, 2023-12-22:
Please report this issue on the TI forum, I have no clue about this.

closed 2023-12-22 by Koenkk

bobuhito, 2023-12-22:
Thanks for thinking about this.  So, you don't compile your firmware with Code Composer Studio?
That's surprising.  I can see that working for the general ARM instructions, but that seems like
a lot of work to program the TI-specific configuration.

Koenkk, 2023-12-23:
I use CCS to compile the firmwares but do not have the knowledge to help you with this issue

bobuhito, 2023-12-24:
Well, this is incredibly weird then.  I will try some other options in CCS to see why I get these bytes reversed...maybe this is because I chose the "TI clang" option for the compiler (they had 3 options).