Koenkk/zigbee2mqtt #20342: IKEA LED2107C4 TRADFRI bulb E14 WS candle 470lm - Not discoveredd

mschut, 2023-12-22:
### What happened?

When trying to connect my new IKEA bulbs (LED2107C4) this version just wont be recognised. 

I have read other issues here about this bulb, and found out it was added in the Zigbee-herdsman-converters version 15.91.0. I have currently version 15.130.1 installed, so I would assume this is up to date. 

### What did you expect to happen?

It woulkd show up in my discovered devices. 

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20210708

### Adapter

usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

crudooper, 2023-12-26:
I have the same problem. How someone pair this device ? It's indicated into manual, toggle main switch 6x, but not in my case... I have a few other bulb from Ikea, no problem...

crudooper, 2023-12-28:
I tried to update adapter firmware to the last version available (20230507), but it's still the same.

nzottmann, 2024-03-07:
Have you been able to get the bulb discovered in the meanwhile @mschut ?

nzottmann, 2024-03-12:
I found a solution for this. Seems like the bulb is quite sensitive about the timing of the reset procedure (turning it off and on 6 times). I could not get this working with flipping the switch manually, but it worked using a smart plug.

Waiting **800ms** between each toggle worked reliably.

closed 2024-10-09 by github-actions[bot]