Koenkk/zigbee2mqtt #20348: [Feature request]: bindings between window sensor and thermostat valves

git-tail-d, 2023-12-22:
### Is your feature request related to a problem? Please describe

Hello,

I would like to lower the room temperature on the heating when the windows are open.
This can of course be set up via the control software, but if z2m is not running, the heating remains switched on.

This task could be solved much more effectively and with less effort using bindings.

I'm still missing the setting options in z2m.

### Describe the solution you'd like

bindings between window sensor and thermostat valves

### Describe alternatives you've considered

bindings also between switches and lamps, sensors and sockets, sensors and lamps.

### Additional context

Just significantly more functions with the bindings.

usma0118, 2023-12-23:
+1 want to combine Ikea switch with hue lights.

sjorge, 2023-12-24:
It depends on the actually devices if they support it or not, the only TRV I am aware of that supports this is the Ubisys H1 with 1.5.x firmware. (It got added in that fw revision) But i have not tested this myself.

closed 2024-07-22 by github-actions[bot]