Koenkk/zigbee2mqtt #20350: Lidl light model 14156506L does no longer support the "light" entitiy in home assistant. Power On behaviour is still there.

paulswerker, 2023-12-22:
### What happened?

I started updating from an older version of Zigbee2MQTT (1.26.0<=) to the latest version 1.34.0. The light remained unavailable from home assistant while other entities remained visible.

The light can still be controlled from the dashboard from mqtt. All other devices seem also unaffected

- I tried to look into the zigbee herdsman converters, but nothing changed to the light as it was first added
- I was not wiser from the changes to the [markdown changes on the documentation](https://github.com/Koenkk/zigbee2mqtt.io/commits/c6b08fe5a345014f2ef085e445a5f24b5d24e920/docs/devices/14156506L.md): 

If somebody can point me to where this changes might have occured I can take a look myself as well. For now solution to stay on 1.26.0

Note: When 1.26 was running and closed down to start 1.34 to create this issue, the light still worked from home assistant. But when restarting HA to test, the connection to the light was gone again. So it seems the availability message is lost

### What did you expect to happen?

Light to be supported

### How to reproduce it (minimal and precise)

1. Have the lidl mood light connected to zigbee2mqtt
2. Update zigbee2mqtt docker image
3. Lidl mood light becomes unavailable, even after deletion, reconnection or anything. 

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20210120

### Adapter

 Slaesh's CC2652RB

### Debug log

This is some of the stuff I could find on this specific version turning it on off

debug 2023-12-22 17:29:49: Saving state to file /app/data/state.json
debug 2023-12-22 17:30:55: Received MQTT message on 'zigbee2mqtt/vensterbank_gamekamer_kleur/set' with data '{"state":"OFF"}'
debug 2023-12-22 17:30:55: Publishing 'set' 'state' to 'vensterbank_gamekamer_kleur'
info  2023-12-22 17:30:55: MQTT publish: topic 'zigbee2mqtt/vensterbank_gamekamer_kleur', payload '{"brightness":132,"color":{"x":0.181,"y":0.091},"color_mode":"xy","color_power_on_behavior":null,"do_not_disturb":true,"linkquality":15,"state":"OFF"}'
debug 2023-12-22 17:30:56: Received MQTT message on 'zigbee2mqtt/vensterbank_gamekamer_kleur/set' with data '{"state":"ON"}'
debug 2023-12-22 17:30:56: Publishing 'set' 'state' to 'vensterbank_gamekamer_kleur'
info  2023-12-22 17:30:56: MQTT publish: topic 'zigbee2mqtt/vensterbank_gamekamer_kleur', payload '{"brightness":132,"color":{"x":0.181,"y":0.091},"color_mode":"xy","color_power_on_behavior":null,"do_not_disturb":true,"linkquality":15,"state":"ON"}'

Koenkk, 2023-12-23:
You are probably using an outdated HA version, updating should fix it.

paulswerker, 2023-12-23:
Thanks koen, that seems to do the trick. :)

closed 2023-12-23 by paulswerker