Koenkk/zigbee2mqtt #20351: [TuYa TH01Z] Zigbee Temperature & Humidity Sensor - Some troubles with Timezone

mb23843, 2023-12-22:
### What happened?

Divice connecting via z2mqtt. All working OK.
But some problems with date and time sync and displaying on device screen (something with Timezone, cant figure out problem).
1) Date sync OK - xx/xx: first part month, second part day.
2) Time sync NOT OK - 3 hours behind from Home Assistant Server. (for example: my local time Home Assistant time 20:17, on device screen 17:17)

I checked the time with Timezone on the Linux Home Assistant container, everything is correct there.
P.S. Sory for bad english

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Zigbee Dongl-E

mb23843, 2024-04-26:
Hello! Any updates? There is trouble with UTC-date I think. When u live in another timezone than UTC

alehuo, 2024-07-08:
I have the same Zigbee adapter as you, and a TH-01Z temperature sensor. No issues here. GMT+02 timezone is set correctly in the device.

closed 2025-01-09 by github-actions[bot]