Koenkk/zigbee2mqtt #20354: UNSUPPORTED_ATTRIBUTE error for TuYa TS0003

jvcsw, 2023-12-23:
### What happened?

Errors appear in the logs constantly.

Device: https://www.zigbee2mqtt.io/devices/TS0003.html

Error:

Failed to configure '0xXXXXXXXXXXXXXX', attempt 3 (Error: Read 0xXXXXXXXXXXXXXX/1 genOnOff(["startUpOnOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:463:22) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:62:9) at result.configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:113:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:139:25) at Configure.configure (/app/lib/extension/configure.ts:117:13))

### What did you expect to happen?

No errors logs

### Zigbee2MQTT version

1.34.0-dev (commit #6875146)

### Adapter firmware version

20211210

### Adapter

zzh

Cossid, 2024-01-02:
Tried removing my incomplete custom external component handler to see how the `Generate definition` feature would do.

Getting a similar `UNSUPPORTED_ATTRIBUTE` when removing my custom external component handler, which causes the entire process to crash.

```
lightingColorCtrl(["colorCapabilities"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
```

Cossid, 2024-01-02:
Looks like #20515 is also related.

Koenkk, 2024-01-02:
Fixed the error!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-02 by Koenkk