Koenkk/zigbee2mqtt #20361: [Wrong device]: Lonsonho QS-Zigbee-D02-TRIAC-LN wrong detected as https://www.zigbee2mqtt.io/devices/TS110E_1gang_2.html

mrmaximas, 2023-12-23:
### Link

https://aliexpress.ru/item/4001279149071.html

### Model

QS-Zigbee-D02-TRIAC-LN

### Description

1 Gang Dimmer Switch Module With Neutral

### Vendor

Lonsonho

### Picture (link)

https://a.allegroimg.com/s1024/0c3520/42b0d2244d588328831981088e3e

### Database entry

{"id":70,"type":"Router","ieeeAddr":"0xa4c138243e41c8dd","nwkAddr":5911,"manufId":4417,"manufName":"_TZ3210_ngqk6jia","powerSource":"Mains (single phase)","modelId":"TS110E","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[3,5,4,6,8,57345,4096,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS110E","manufacturerName":"_TZ3210_ngqk6jia","powerSource":1,"zclVersion":3,"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genLevelCtrl":{"attributes":{"61440":392,"64514":0}},"genOnOff":{"attributes":{"moesStartUpOnOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-18028522},"lastSeen":1703334210122,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-12-24:
Fixed.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

mrmaximas, 2023-12-24:
> Fixed.
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Thanks!
How ti fix it issue?
zigbee2mqtt % git pull
Updating 27ccd1a5..1a726501
error: Your local changes to the following files would be overwritten by merge:
	package.json
Please commit your changes or stash them before you merge.
Aborting
zigbee2mqtt %

Koenkk, 2023-12-25:
Forgot to push, added it now.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-25 by Koenkk

mrmaximas, 2025-02-22:
@Koenkk It looks like I found a bug in the converter and at initial configuration of devices of these two types:
https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-LN_1.html#lonsonho-qs-zigbee-d02-triac-ln_1
[image]

https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-L_1.html#lonsonho-qs-zigbee-d02-triac-l_1
[image]

Reporting is used from two identical clusters. These devices have only one channel each. I think the bug occurred when you made a converter from the converter for the same device but with two channels.