Koenkk/zigbee2mqtt #20362: Z2m crashes at adding devices

oleksii-kalinin, 2023-12-23:
### What happened?

I'm trying to add some devices and z2m crash

### Zigbee2MQTT version

1.34.0-1

### Adapter firmware version

7

### Adapter

Skyconnect

### Debug log

2023-12-23T12:56:26.979Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 3c90012400e6a589ee2dfeffea35900203249f
2023-12-23T12:56:26.981Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x24: {"_cls_":"trustCenterJoinHandler","_id_":36,"_isRequest_":false,"newNodeId":42470,"newNodeEui64":{"type":"Buffer","data":[144,53,234,255,254,45,238,137]},"status":2,"policyDecision":3,"parentOfNewNodeId":40740}
2023-12-23T12:56:26.986Z zigbee-herdsman:adapter:ezsp:debg Device left network request received: 42470 9035eafffe2dee89
2023-12-23T12:56:26.988Z zigbee-herdsman:controller:log Device leave '0x9035eafffe2dee89'
Zigbee2MQTT:warn  2023-12-23 13:56:26: Device '0x9035eafffe2dee89' left the network
Zigbee2MQTT:info  2023-12-23 13:56:27: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x9035eafffe2dee89","ieee_address":"0x9035eafffe2dee89"},"type":"device_leave"}'
2023-12-23T12:56:27.139Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,6,0): 167eb1a9702af317d07a67db55bfa7d99e4d846472687e
2023-12-23T12:56:27.147Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2023-12-23T12:56:27.149Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
2023-12-23T12:56:27.150Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 3c90012400e6a589ee2dfeffea35900203a3cf
2023-12-23T12:56:27.150Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x24: {"_cls_":"trustCenterJoinHandler","_id_":36,"_isRequest_":false,"newNodeId":42470,"newNodeEui64":{"type":"Buffer","data":[144,53,234,255,254,45,238,137]},"status":2,"policyDecision":3,"parentOfNewNodeId":53155}
2023-12-23T12:56:27.153Z zigbee-herdsman:adapter:ezsp:debg Device left network request received: 42470 9035eafffe2dee89
2023-12-23T12:56:27.153Z zigbee-herdsman:controller:log Device leave '0x9035eafffe2dee89'
Zigbee2MQTT:warn  2023-12-23 13:56:27: Device '0x9035eafffe2dee89' left the network
Zigbee2MQTT:info  2023-12-23 13:56:27: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x9035eafffe2dee89","ieee_address":"0x9035eafffe2dee89"},"type":"device_leave"}'
2023-12-23T12:56:27.306Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (2,6,0): 267eb1a9702af317d07a67db55bfa7d99e4d8464cf9d7e
2023-12-23T12:56:27.307Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (3)
2023-12-23T12:56:27.309Z zigbee-herdsman:adapter:ezsp:uart --> [83401b7e]
2023-12-23T12:56:27.310Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 3c90012400e6a589ee2dfeffea35900203a3cf
2023-12-23T12:56:27.310Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x24: {"_cls_":"trustCenterJoinHandler","_id_":36,"_isRequest_":false,"newNodeId":42470,"newNodeEui64":{"type":"Buffer","data":[144,53,234,255,254,45,238,137]},"status":2,"policyDecision":3,"parentOfNewNodeId":53155}
Zigbee2MQTT:warn  2023-12-23 13:56:27: Device '0x9035eafffe2dee89' left the network
2023-12-23T12:56:27.311Z zigbee-herdsman:adapter:ezsp:debg Device left network request received: 42470 9035eafffe2dee89
2023-12-23T12:56:27.311Z zigbee-herdsman:controller:log Device leave '0x9035eafffe2dee89'
Zigbee2MQTT:info  2023-12-23 13:56:27: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x9035eafffe2dee89","ieee_address":"0x9035eafffe2dee89"},"type":"device_leave"}'
2023-12-23T12:56:27.468Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"lookupEui64ByNodeId","_id_":97,"_isRequest_":true,"nodeId":53954}
2023-12-23T12:56:27.471Z zigbee-herdsman:adapter:ezsp:uart --> DATA (6,3,0): 3e00016100c2d2
2023-12-23T12:56:27.472Z zigbee-herdsman:adapter:ezsp:uart --> [637c21a9352ad76061f47e]
2023-12-23T12:56:27.472Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (7)
Zigbee2MQTT:warn  2023-12-23 13:56:27: Device '0x9035eafffe2dee89' left the network
2023-12-23T12:56:27.473Z zigbee-herdsman:adapter:ezsp:ezsp ==> sendUnicast: {"type":0,"indexOrDestination":42470,"apsFrame":{"clusterId":2,"profileId":0,"sequence":4,"sourceEndpoint":0,"destinationEndpoint":0,"groupId":0,"options":256},"messageTag":5,"message":{"type":"Buffer","data":[4,230,165]}}
2023-12-23T12:56:27.511Z zigbee-herdsman:adapter:ezsp:uart <-- [367d5eb1a9702af317d07a67db55bfa7d99e4df3447d31627e467d5eb1a9702ad7608c5e8bdb55405f4d9e4d03346ecd7e567d5eb1a9702af317d07a67db55bfa7d99e4df3447aa97e667d5eb1a96b2a7d334fa6944a36aa559249984e27d612ce674be87e2e7d5eb1a9702af317d07a67db55bfa7d99e4d846472247e3e7d5eb1a9702af317d07a67db55bfa7d99e4df344acdb7e4e7d5eb1a9702ad7608c5e8bdb55405f4d9e4d0334d3747e5e7d5eb1a9702af317d07a67db55bfa7d99e4df344c7107e6e7d5eb1a96b2a7d334fa6944a36aa559249984e27d612ce6753d37e]
2023-12-23T12:56:27.512Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (3,6,0): 367eb1a9702af317d07a67db55bfa7d99e4df34411627e
2023-12-23T12:56:27.512Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (4)
2023-12-23T12:56:27.512Z zigbee-herdsman:adapter:ezsp:uart --> [8430fc7e]
2023-12-23T12:56:27.513Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 6 | 7
2023-12-23T12:56:27.513Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 3c90012400e6a589ee2dfeffea35900203d4ef
2023-12-23T12:56:27.515Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x24: {"_cls_":"trustCenterJoinHandler","_id_":36,"_isRequest_":false,"newNodeId":42470,"newNodeEui64":{"type":"Buffer","data":[144,53,234,255,254,45,238,137]},"status":2,"policyDecision":3,"parentOfNewNodeId":61396}
2023-12-23T12:56:27.516Z zigbee-herdsman:adapter:ezsp:debg Device left network request received: 42470 9035eafffe2dee89
2023-12-23T12:56:27.517Z zigbee-herdsman:controller:log Device leave '0x9035eafffe2dee89'
Zigbee2MQTT:info  2023-12-23 13:56:27: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x9035eafffe2dee89","ieee_address":"0x9035eafffe2dee89"},"type":"device_leave"}'
2023-12-23T12:56:27.757Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (4,6,0): 467eb1a9702ad7608c5e8bdb55405f4d9e4d03346ecd7e
2023-12-23T12:56:27.757Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (5)
2023-12-23T12:56:27.758Z zigbee-herdsman:adapter:ezsp:uart --> [8520dd7e]
2023-12-23T12:56:27.758Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 6 | 7
2023-12-23T12:56:27.758Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 3c90012400c2d2d5cac1feff15cd040203249f
2023-12-23T12:56:27.758Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x24: {"_cls_":"trustCenterJoinHandler","_id_":36,"_isRequest_":false,"newNodeId":53954,"newNodeEui64":{"type":"Buffer","data":[4,205,21,255,254,193,202,213]},"status":2,"policyDecision":3,"parentOfNewNodeId":40740}
2023-12-23T12:56:27.758Z zigbee-herdsman:adapter:ezsp:debg Device left network request received: 53954 04cd15fffec1cad5
2023-12-23T12:56:27.758Z zigbee-herdsman:controller:log Device leave '0x04cd15fffec1cad5'
Zigbee2MQTT:warn  2023-12-23 13:56:27: Device '0x04cd15fffec1cad5' left the network
Zigbee2MQTT:info  2023-12-23 13:56:27: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cd15fffec1cad5","ieee_address":"0x04cd15fffec1cad5"},"type":"device_leave"}'
Zigbee2MQTT:warn  2023-12-23 13:56:27: Device '0x9035eafffe2dee89' left the network
2023-12-23T12:56:27.880Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (5,6,0): 567eb1a9702af317d07a67db55bfa7d99e4df3447aa97e
[... 62 lines cut ...]
2023-12-23T12:56:32.190Z zigbee-herdsman:adapter:ezsp:debg Requesting 'Node Descriptor' for '53954'
2023-12-23T12:56:32.192Z zigbee-herdsman:adapter:ezsp:driv ZDO Node_Desc_req params: {"dstaddr":53954}
2023-12-23T12:56:32.194Z zigbee-herdsman:adapter:ezsp:ezsp ==> lookupEui64ByNodeId: {"nodeId":53954}
2023-12-23T12:56:32.199Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"lookupEui64ByNodeId","_id_":97,"_isRequest_":true,"nodeId":53954}
2023-12-23T12:56:32.200Z zigbee-herdsman:adapter:ezsp:uart --> DATA (7,7,0): 3f00016100c2d2
2023-12-23T12:56:32.202Z zigbee-herdsman:adapter:ezsp:uart --> [777d5d21a9352ad760eb4c7e]
2023-12-23T12:56:32.204Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (0)
2023-12-23T12:56:32.228Z zigbee-herdsman:adapter:ezsp:uart Port closed
2023-12-23T12:56:32.228Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2023-12-23T12:56:32.228Z zigbee-herdsman:adapter:ezsp:driv Close driver
2023-12-23T12:56:32.228Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2023-12-23T12:56:32.228Z zigbee-herdsman:adapter:ezsp:driv Close driver
[... cut ...]
2023-12-23T12:56:34.204Z zigbee-herdsman:adapter:ezsp:uart --> Error: Error: {"sequence":0} after 2000ms
2023-12-23T12:56:34.204Z zigbee-herdsman:adapter:ezsp:uart -!- break waiting (0)
2023-12-23T12:56:34.204Z zigbee-herdsman:adapter:ezsp:uart Can't send DATA frame (7,7,0): 3f00016100c2d2
Error: {"address":42470,"clusterId":32770,"sequence":4} after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)

oleksii-kalinin, 2023-12-23:
```
Zigbee2MQTT:info  2023-12-23 14:05:04: Device '0x9035eafffe3285b3' joined
Zigbee2MQTT:info  2023-12-23 14:05:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x9035eafffe3285b3","ieee_address":"0x9035eafffe3285b3"},"type":"device_joined"}'
Zigbee2MQTT:info  2023-12-23 14:05:05: MQTT publish: topic 'zigbee2mqtt/0x9035eafffe3285b3/availability', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-12-23 14:05:05: MQTT publish: topic 'zigbee2mqtt/0x9035eafffe3285b3', payload '{"last_seen":"2023-12-23T14:05:05+01:00"}'
Zigbee2MQTT:info  2023-12-23 14:05:05: Starting interview of '0x9035eafffe3285b3'
Zigbee2MQTT:info  2023-12-23 14:05:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x9035eafffe3285b3","ieee_address":"0x9035eafffe3285b3","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2023-12-23 14:05:05: MQTT publish: topic 'zigbee2mqtt/0x9035eafffe3285b3', payload '{"last_seen":"2023-12-23T14:05:05+01:00"}'
Error: Failure send findKeyTableEntry:{"type":"Buffer","data":[197,0,1,117,0,179,133,50,254,255,234,53,144,1]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:562:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

oleksii-kalinin, 2023-12-23:
```
Zigbee2MQTT:info  2023-12-23 14:12:04: Device '0x9035eafffe67ee68' joined
Zigbee2MQTT:info  2023-12-23 14:12:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x9035eafffe67ee68","ieee_address":"0x9035eafffe67ee68"},"type":"device_joined"}'
Zigbee2MQTT:info  2023-12-23 14:12:04: MQTT publish: topic 'zigbee2mqtt/0x9035eafffe67ee68/availability', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-12-23 14:12:04: MQTT publish: topic 'zigbee2mqtt/0x9035eafffe67ee68', payload '{"last_seen":"2023-12-23T14:12:04+01:00"}'
Zigbee2MQTT:info  2023-12-23 14:12:04: Starting interview of '0x9035eafffe67ee68'
Zigbee2MQTT:info  2023-12-23 14:12:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x9035eafffe67ee68","ieee_address":"0x9035eafffe67ee68","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2023-12-23 14:12:04: MQTT publish: topic 'zigbee2mqtt/0x9035eafffe67ee68', payload '{"last_seen":"2023-12-23T14:12:04+01:00"}'
Zigbee2MQTT:info  2023-12-23 14:12:06: Device '0x04cd15fffe7e14c3' joined
Zigbee2MQTT:info  2023-12-23 14:12:06: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cd15fffe7e14c3","ieee_address":"0x04cd15fffe7e14c3"},"type":"device_joined"}'
Zigbee2MQTT:info  2023-12-23 14:12:06: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe7e14c3/availability', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-12-23 14:12:06: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe7e14c3', payload '{"last_seen":"2023-12-23T14:12:06+01:00"}'
Zigbee2MQTT:info  2023-12-23 14:12:06: Starting interview of '0x04cd15fffe7e14c3'
Zigbee2MQTT:info  2023-12-23 14:12:06: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cd15fffe7e14c3","ieee_address":"0x04cd15fffe7e14c3","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2023-12-23 14:12:06: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe7e14c3', payload '{"last_seen":"2023-12-23T14:12:06+01:00"}'
Zigbee2MQTT:info  2023-12-23 14:12:06: MQTT publish: topic 'zigbee2mqtt/0x9035eafffe67ee68', payload '{"last_seen":"2023-12-23T14:12:06+01:00"}'
Zigbee2MQTT:info  2023-12-23 14:12:06: MQTT publish: topic 'zigbee2mqtt/Climate Bedroom', payload '{"battery":96.5,"humidity":64.88,"last_seen":"2023-12-23T14:12:06+01:00","linkquality":244,"temperature":20.61,"voltage":2900}'
Error: Failure send findKeyTableEntry:{"type":"Buffer","data":[93,0,1,117,0,104,238,103,254,255,234,53,144,1]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:562:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

oleksii-kalinin, 2023-12-23:
```
Zigbee2MQTT:info  2023-12-23 14:21:27: Zigbee: allowing new devices to join.
Zigbee2MQTT:error 2023-12-23 14:21:27: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Connection not initialized'
Zigbee2MQTT:info  2023-12-23 14:21:27: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{},"error":"Connection not initialized","status":"error","transaction":"nhj0k-2"}'
Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:551:19)
    at Driver.addTransientLinkKey (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:678:26)
    at Driver.preJoining (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:610:35)
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:244:29)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:42)
    at /app/node_modules/zigbee-herdsman/src/utils/queue.ts:21:18
    at new Promise (<anonymous>)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:19:16)
    at EZSPAdapter.permitJoin (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:241:27)
    at Controller.permitJoinInternal (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:246:32)
```

Tiremo, 2023-12-28:
Can confirm the exact same issue on my end. Same Adapter, Firmware, Z2M Version etc.

irata-coder, 2023-12-30:
I think this might be the caused by the same underlying issue as one I have reported here:

[https://github.com/Koenkk/zigbee-herdsman/issues/834](url)

corporategoth, 2024-01-07:
For reference, an older ticket with a lot of other people commenting on the same - https://github.com/Koenkk/zigbee2mqtt/issues/17532

closed 2024-01-09 by oleksii-kalinin

pathia, 2024-02-06:
@oleksii-kalinin you closed the issue in januari, but I'm curious: is the problem solved for you?

oleksii-kalinin, 2024-02-06:
@pathia I closed the issue b/c similar issues have already been created.

pathia, 2024-02-06:
Yes so I saw, but none of them have a real solution for the issue you are facing. It's the same as I experience over here by the way.
So I was curious how things went.

oleksii-kalinin, 2024-02-06:
@pathia my issue was that I tried to add 6 IKEA leds at one time, so I add one by one and everything went ok. I had no issue with adding 6 LEDs with the conbee2 adapter, so it's probably an issue with z2m <> ezsp.

pathia, 2024-02-06:
Unfortunately my problem is not so easy to solve. Over here it's a sonoff temp hum sensor. I will open a new issue for it later.