Koenkk/zigbee2mqtt #20363: Applying zigbee scene to 24 INNR RS 227 T  crashes zigbee network - reproducable

mogim0gi, 2023-12-23:
### What happened?

after applying a scene to 3 zigbee groups (24 devices ) 3 times in 10 seconds:
1. Zigbee network went down, no devices on network responsive, however zigbee2mqtt remains active.
2.  INNR RS 227 T lights started flickering randomly, only fixable by power cycle.

### What did you expect to happen?

Scene to be applied (brightness change) without crashing zigbee network

### How to reproduce it (minimal and precise)

1. Apply scene 1 affecting 24 zigbee devices
2. Apply scene 2 affecting 24 zigbee devices
3. Apply scene 1 again

### Zigbee2MQTT version

1.34.0 

### Adapter firmware version

20221226

### Adapter

Sonoff adapter P

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13759414/log.txt)

mogim0gi, 2023-12-23:
I usually get: 

{"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))`

sometimes followed by NWK_full errors.

`[Info](http://192.168.1.236:8123/hassio/addon/45df7312_zigbee2mqtt/info)
[Documentation](http://192.168.1.236:8123/hassio/addon/45df7312_zigbee2mqtt/documentation)
[Configuration](http://192.168.1.236:8123/hassio/addon/45df7312_zigbee2mqtt/config)
[Log](http://192.168.1.236:8123/hassio/addon/45df7312_zigbee2mqtt/logs)
Zigbee2MQTT
2023-12-23T18:48:03.234Z zigbee-herdsman:controller:log Device announce '0xc498860000977c92'
Zigbee2MQTT:info  2023-12-23 20:48:03: MQTT publish: topic 'zigbee2mqtt/0xc498860000977c92', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"color_temp_startup":65535,"last_seen":"2023-12-23T18:48:03.234Z","linkquality":61,"power_on_behavior":"previous","state":"ON","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":false}'
Zigbee2MQTT:debug 2023-12-23 20:48:03: Device '0xc498860000977c92' announced itself
Zigbee2MQTT:info  2023-12-23 20:48:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xc498860000977c92","ieee_address":"0xc498860000977c92"},"type":"device_announce"}'
Zigbee2MQTT:info  2023-12-23 20:48:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xc498860000977c92"},"type":"device_announced"}'
Zigbee2MQTT:info  2023-12-23 20:48:03: MQTT publish: topic 'zigbee2mqtt/0xc4988600000d7fb7', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"color_temp_startup":65535,"last_seen":"2023-12-23T18:48:03.365Z","linkquality":36,"power_on_behavior":"previous","state":"ON","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":false}'
2023-12-23T18:48:03.364Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,57,213,57,213,183,127,13,0,0,134,152,196,142,24,254,13,69,193,244,24,244,24,178,117,13,0,0,134,152,196,142,23,254,1,100,1,0,100,254,3,68,128,0,1,243,53,254,28,68,129,0,0,6,0,59,68,1,1,0,25,0,128,163,170,0,0,8,24,126,1,0,0,0,16,0,59,68,29,45]
2023-12-23T18:48:03.365Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [57,213,57,213,183,127,13,0,0,134,152,196,142] - 24
2023-12-23T18:48:03.365Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":54585,"nwkaddr":54585,"ieeeaddr":"0xc4988600000d7fb7","capabilities":142}
2023-12-23T18:48:03.365Z zigbee-herdsman:controller:log Device announce '0xc4988600000d7fb7'
Zigbee2MQTT:debug 2023-12-23 20:48:03: Device '0xc4988600000d7fb7' announced itself
Zigbee2MQTT:info  2023-12-23 20:48:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xc4988600000d7fb7","ieee_address":"0xc4988600000d7fb7"},"type":"device_announce"}'
Zigbee2MQTT:info  2023-12-23 20:48:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xc4988600000d7fb7"},"type":"device_announced"}'
2023-12-23T18:48:03.452Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,244,24,244,24,178,117,13,0,0,134,152,196,142,23,254,1,100,1,0,100,254,3,68,128,0,1,243,53,254,28,68,129,0,0,6,0,59,68,1,1,0,25,0,128,163,170,0,0,8,24,126,1,0,0,0,16,0,59,68,29,45]
2023-12-23T18:48:03.452Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [244,24,244,24,178,117,13,0,0,134,152,196,142] - 23
2023-12-23T18:48:03.453Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":6388,"nwkaddr":6388,"ieeeaddr":"0xc4988600000d75b2","capabilities":142}
2023-12-23T18:48:03.453Z zigbee-herdsman:controller:log Device announce '0xc4988600000d75b2'
Zigbee2MQTT:info  2023-12-23 20:48:03: MQTT publish: topic 'zigbee2mqtt/0xc4988600000d75b2', payload '{"brightness":254,"color_mode":"color_temp","color_temp":332,"color_temp_startup":65535,"last_seen":"2023-12-23T18:48:03.453Z","linkquality":145,"power_on_behavior":"previous","state":"ON","update":{"installed_version":604050705,"latest_version":604050705,"state":"idle"},"update_available":false}'
Zigbee2MQTT:debug 2023-12-23 20:48:03: Device '0xc4988600000d75b2' announced itself
Zigbee2MQTT:info  2023-12-23 20:48:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xc4988600000d75b2","ieee_address":"0xc4988600000d75b2"},"type":"device_announce"}'
Zigbee2MQTT:info  2023-12-23 20:48:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xc4988600000d75b2"},"type":"device_announced"}'
2023-12-23T18:48:03.543Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100,254,3,68,128,0,1,243,53,254,28,68,129,0,0,6,0,59,68,1,1,0,25,0,128,163,170,0,0,8,24,126,1,0,0,0,16,0,59,68,29,45]
2023-12-23T18:48:03.543Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2023-12-23T18:48:03.543Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2023-12-23T18:48:03.543Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,243,53,254,28,68,129,0,0,6,0,59,68,1,1,0,25,0,128,163,170,0,0,8,24,126,1,0,0,0,16,0,59,68,29,45]
[... cut ...] ndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
Zigbee2MQTT:error 2023-12-23 20:48:58: Failed to read state of '0xc4988600000d75b2' after reconnect (Read 0xc4988600000d75b2/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))`

mogim0gi, 2024-01-14:
@Koenkk what do you need to take a look at this? Basically can reproduce anytime with moving color temperature slider slowly, 5-6 increments and it crashes.

Koenkk, 2024-01-15:
I will publish a new firmware soon, keep a look at https://github.com/Koenkk/Z-Stack-firmware/discussions (it may fix your issue)

mogim0gi, 2024-07-14:
Thanks, fw update on INNR fixed it

closed 2024-07-15 by Koenkk