Koenkk/zigbee2mqtt #20365: BSEED 1 Gang Switch with moore Functions

HobboRobin, 2023-12-23:
### Link

https://www.bseed.com/products/wifi-light-switch-function-key-diy-parts?variant=42014280024219

### Database entry

{ 	"id":120, 	"type":"Router", 	"ieeeAddr":"0xa4c1389715ee4f65", 	"nwkAddr":24228, 	"manufId":4417, 	"manufName":"_TZ3000_ysdv91bk", 	"powerSource":"Mains (single phase)", 	"modelId":"TS0001", 	"epList":[1,242], 	"endpoints":{ 		"1":{ 			"profId":260, 			"epId":1, 			"devId":256, 			"inClusterList":[3,4,5,6,57344,57345,0], 			"outClusterList":[25,10], 			"clusters":{ 				"genBasic":{ 					"attributes":{ 						"65503":"\t\u0000\u0000\u0000\u0012\u000f\u0000\u0000\u0000\u0016", 						"65506":54, 						"65508":1, 						"65534":0, 						"modelId":"TS0001", 						"manufacturerName":"_TZ3000_ysdv91bk", 						"powerSource":1, 						"zclVersion":3, 						"appVersion":82, 						"stackVersion":0, 						"hwVersion":1, 						"dateCode":"" 					} 				}, 				"manuSpecificBosch":{ 					"attributes":{ 						"53251":"AAAA" 					} 				}, 				"genOnOff":{ 					"attributes":{ 						"32768":0, 						"onOff":0, 						"onTime":0, 						"offWaitTime":0, 						"tuyaBacklightMode":1, 						"moesStartUpOnOff":0, 						"tuyaBacklightSwitch":1 					} 				}, 				"manuSpecificTuya_3":{ 					"attributes":{ 						"powerOnBehavior":0, 						"switchType":0 					} 				} 			}, 			"binds":[{ 					"cluster":6, 					"type":"endpoint", 					"deviceIeeeAddress":"0x00124b0026b74820", 					"endpointID":1 				} 			], 			"configuredReportings":[], 			"meta":{} 		}, 		"242":{ 			"profId":41440, 			"epId":242, 			"devId":97, 			"inClusterList":[], 			"outClusterList":[33], 			"clusters":{}, 			"binds":[], 			"configuredReportings":[], 			"meta":{} 		} 	}, 	"appVersion":82, 	"stackVersion":0, 	"hwVersion":1, 	"dateCode":"", 	"zclVersion":3, 	"interviewCompleted":true, 	"meta":{ 		"configured":153093895 	}, 	"lastSeen":1703355693140, 	"defaultSendRequestWhen":"immediate" }

### Comments

I tried to make the external converter but i stuck with almost everything that begins with Tuya
The Device is working basicly with the already existing TS0001 switch but the backlight is not even showing 

### Supported color modes

on, off (red (zigbee connection) and blue (Switch state))

HobboRobin, 2023-12-30:
Nothing ? i am really no idea what i can do

MephistoJB, 2024-01-20:
In [this](https://github.com/Koenkk/zigbee2mqtt/issues/16311#issuecomment-1413969515) post @NoAdO stated, that he was able to integrate the TS0011 from BSEED. This seems to be very similar to your device. Maybe he can help here by sharing the external converter. I would also appreciate help here.

NoAdO, 2024-01-20:
> In [this](https://github.com/Koenkk/zigbee2mqtt/issues/16311#issuecomment-1413969515) post @NoAdO stated, that he was able to integrate the TS0011 from BSEED. This seems to be very similar to your device. Maybe he can help here by sharing the external converter. I would also appreciate help here.

Sorry, i can't help. I just excluded device from network to locate the problem and later switched stick, nothing more.

MephistoJB, 2024-01-20:
Oh that is a pitty.

@NoAdO Do you have any other BSEED devices integrated. Maybe we can use them as a reference, so we won´t have to start from scratch

NoAdO, 2024-01-20:
> Oh that is a pitty.
> 
> @NoAdO Do you have any other BSEED devices integrated. Maybe we can use them as a reference, so we won´t have to start from scratch

No, only that one. And i do not own efr stick anymore, i gifted it to friend :D And he has only tuya and yandex, as i know

MephistoJB, 2024-01-22:
I tried to develop an external converter by following the guide for Tuya on the website. No success here. Nothing is printed out. I would appreciate any help here. What's the usual procedure for such a case?

closed 2024-08-21 by github-actions[bot]