Koenkk/zigbee2mqtt #20366: BSEED 2 Gang Switch with moore Functions

HobboRobin, 2023-12-23:
### Link

https://www.bseed.com/products/wifi-light-switch-function-key-diy-parts?variant=42014280024219

### Database entry

{ 	"id":123, 	"type":"Router", 	"ieeeAddr":"0xa4c138ce47a2c086", 	"nwkAddr":28434, 	"manufId":4417, 	"manufName":"_TZ3000_mklgayek", 	"powerSource":"Mains (single phase)", 	"modelId":"TS0002", 	"epList":[1,2,242], 	"endpoints":{ 		"1":{ 			"profId":260, 			"epId":1, 			"devId":256, 			"inClusterList":[3,4,5,6,57344,57345,0], 			"outClusterList":[25,10], 			"clusters":{ 				"genBasic":{ 					"attributes":{ 						"65503":"6 \u0019-\u0012I \u0019-\u0016", 						"65506":54, 						"65508":0, 						"65534":0, 						"stackVersion":0, 						"dateCode":"", 						"manufacturerName":"_TZ3000_mklgayek", 						"zclVersion":3, 						"appVersion":82, 						"modelId":"TS0002", 						"powerSource":1 					} 				}, 				"genOnOff":{ 					"attributes":{ 						"32768":0, 						"onOff":0, 						"onTime":0, 						"offWaitTime":0, 						"tuyaBacklightMode":1, 						"moesStartUpOnOff":0, 						"tuyaBacklightSwitch":1 					} 				}, 				"manuSpecificBosch":{ 					"attributes":{ 						"53251":"AAAAAgAA" 					} 				}, 				"manuSpecificTuya_3":{ 					"attributes":{ 						"powerOnBehavior":0, 						"switchType":0 					} 				} 			}, 			"binds":[{ 					"cluster":6, 					"type":"endpoint", 					"deviceIeeeAddress":"0x00124b0026b74820", 					"endpointID":1 				} 			], 			"configuredReportings":[], 			"meta":{} 		}, 		"2":{ 			"profId":260, 			"epId":2, 			"devId":256, 			"inClusterList":[4,5,6,57345], 			"outClusterList":[], 			"clusters":{ 				"genOnOff":{ 					"attributes":{ 						"onOff":0, 						"onTime":0, 						"offWaitTime":0 					} 				}, 				"manuSpecificTuya_3":{ 					"attributes":{ 						"powerOnBehavior":0 					} 				} 			}, 			"binds":[{ 					"cluster":6, 					"type":"endpoint", 					"deviceIeeeAddress":"0x00124b0026b74820", 					"endpointID":1 				} 			], 			"configuredReportings":[], 			"meta":{} 		}, 		"242":{ 			"profId":41440, 			"epId":242, 			"devId":97, 			"inClusterList":[], 			"outClusterList":[33], 			"clusters":{}, 			"binds":[], 			"configuredReportings":[], 			"meta":{} 		} 	}, 	"appVersion":82, 	"stackVersion":0, 	"hwVersion":1, 	"dateCode":"", 	"zclVersion":3, 	"interviewCompleted":true, 	"meta":{ 		"configured":-1030697865 	}, 	"lastSeen":1703351750345, 	"defaultSendRequestWhen":"immediate" }

### Comments

I tried to make the external converter but i stuck with almost everything that begins with Tuya
The Device is working basicly with the already existing TS0002 switch but the backlight is not even showing

i also tried it with the TS0001 (different post)

### Supported color modes

on, off (red (zigbee connection) and blue (Switch state))

HobboRobin, 2023-12-30:
Nothing ? i am really no idea what i can do

rosicenko, 2024-03-18:
Still not working? I think I have the same device and it shows in Z2MQTT as this and works well. 
[image]

closed 2024-08-26 by HobboRobin