Koenkk/zigbee2mqtt #20367: Interview - node descriptor request failed Ikea E1743 / E1603

geeano, 2023-12-23:
### What happened?

After joining device, interview is started but ends up in fail
Various attempts at two different E1603 devices and one E1743 device. No one succeeded

### What did you expect to happen?

Interview would be succesful and device would be paired 

### How to reproduce it (minimal and precise)

follow pairing instruction in accordance to supported devices information . 

### Zigbee2MQTT version

1.33.2-dev 

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

[log_ikea_E1743.txt](https://github.com/Koenkk/zigbee2mqtt/files/13760246/log_ikea_E1743.txt)

relevant device 0xbc026efffe6a868b
2023-12-23T21:08:21.248Z zigbee-herdsman:controller:log Device '0xbc026efffe6a868b' joined
2023-12-23T21:08:21.249Z zigbee-herdsman:controller:log Device '0xbc026efffe6a868b' accepted by handler
2023-12-23T21:08:21.250Z zigbee-herdsman:controller:log Interview '0xbc026efffe6a868b' start
Zigbee2MQTT:info  2023-12-23 22:08:21: Starting interview of '0xbc026efffe6a868b'
Zigbee2MQTT:info  2023-12-23 22:08:21: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc026efffe6a868b","ieee_address":"0xbc026efffe6a868b","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2023-12-23 22:08:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xbc026efffe6a868b"},"type":"pairing"}'
2023-12-23T21:08:21.409Z zigbee-herdsman:controller:device:log Interview - start device '0xbc026efffe6a868b'
2023-12-23T21:08:44.443Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0xbc026efffe6a868b', attempt 1

geeano, 2023-12-30:
Coen, do you have any idea?

Koenkk, 2023-12-30:
https://www.zigbee2mqtt.io/guide/faq/#interview-fails might help

geeano, 2023-12-30:
- Try pairing the device closer to the coordinator. Done no result

-There can be too much interference, try connecting the coordinator USB through an USB extension cable. This problem occurs a lot when used in combination with a Raspberry Pi 4. Since beginning running Z2M I'm using an extension cable. 

- If it’s a battery powered device, try replacing the battery. Try to keep the device awake by pressing the button of the device (if any) every 3 seconds.
Try re-pairing the device again for 2 or 3 times. Device with 230V power supply. Various attempts done with two different Ikea power sockets. I'm using another power socket same model already a couple of years, a second and a third one failed. First assumption was device fail, so I bought a third one. Same symptoms.

-This might be a Zigbee2MQTT bug, Create a new issue with the zigbee-herdsman debug logging attached to it. [How to enable zigbee-herdsman debug logging](https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging). Done, log uploaded

-If device joins with 0x000000000000000 as ieeeAddress (you will see: Starting interview of '0x0000000000000000' in the Zigbee2MQTT log) your CC253X might be broken. See issue #2761. Not the case

In case the device is a bulb, try resetting it through [Touchlink](https://www.zigbee2mqtt.io/guide/usage/touchlink.html)
Try pairing close to a bulb (light) router instead of the coordinator. Not the case. 

For this reason I opened the issue

geeano, 2024-01-05:
Hi Koen, any suggestions after doing all checks as indicated above ?

Koenkk, 2024-01-06:
One thing you can also try is updating the coordinator firmware

geeano, 2024-02-02:
I bought a new sonoff  adapter that will be provided with the latest version of the firmware, to check if it will solve the issue.

geeano, 2024-02-02:
I also created a sniffer adapter that can be used with wireshark in order to check the communication on the various layers.

closed 2024-09-01 by github-actions[bot]

geeano, 2024-11-21:
Problem with the IKEA devices E1743 / E1603 are still not resolved. I've been able tosetup a sniffing environment but are not able to catch the node descriptor request.. 

From the Zigbee-herdsman*  debugging to follow log is given 

[2024-11-21 13:38:02] debug:    zh:zstack: Discovering route to 12536
[2024-11-21 13:38:02] debug:    zh:zstack:znp: SREQ: --> ZDO - extRouteDisc - {"dstAddr":12536,"options":0,"radius":30}
[2024-11-21 13:38:02] debug:    zh:zstack:unpi:writer: --> frame [254,4,37,69,248,48,0,30,178]
[2024-11-21 13:38:02] debug:    zh:zstack:unpi:parser: <-- [254,1,101,69,0,33]
[2024-11-21 13:38:02] debug:    zh:zstack:unpi:parser: --- parseNext [254,1,101,69,0,33]
[2024-11-21 13:38:02] debug:    zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 69 - [0] - 33
[2024-11-21 13:38:02] debug:    zh:zstack:znp: SRSP: <-- ZDO - extRouteDisc - {"status":0}
[2024-11-21 13:38:02] debug:    zh:zstack:unpi:parser: --- parseNext []
[2024-11-21 13:38:05] debug:    zh:zstack:znp: SREQ: --> ZDO - nodeDescReq - {"dstaddr":12536,"nwkaddrofinterest":12536}
[2024-11-21 13:38:05] debug:    zh:zstack:unpi:writer: --> frame [254,4,37,2,248,48,248,48,35]
[2024-11-21 13:38:05] debug:    zh:zstack:unpi:parser: <-- [254,1,101,2,0,102]
[2024-11-21 13:38:05] debug:    zh:zstack:unpi:parser: --- parseNext [254,1,101,2,0,102]
[2024-11-21 13:38:05] debug:    zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 2 - [0] - 102
[2024-11-21 13:38:05] debug:    zh:zstack:znp: SRSP: <-- ZDO - nodeDescReq - {"status":0}
[2024-11-21 13:38:05] debug:    zh:zstack:unpi:parser: --- parseNext []
[2024-11-21 13:38:15] debug:    zh:controller:device: Interview - quirks check for 'undefined'-'undefined'-'Unknown'
[2024-11-21 13:38:15] debug:    zh:controller:device: Interview - quirks did not match
[2024-11-21 13:38:15] debug:    zh:controller:device: Interview - node descriptor request failed for '0xbc026efffe6a868b', attempt 6

The problem occurs at two E1603 device and one E1743 device. In the past I joined my first E1603 that is still in operation. Unfortunately the new ones fail to join. Koen do you have any suggestion / hints to perform further investigation. I flashed a new Sonoff Dongle Plus "ZBDongle-P" but  it did not result in a fix.

Thx in advance, your help would be appreciated