Koenkk/zigbee2mqtt #20369: [New device support]:

g1335333249, 2023-12-24:
### Link

https://www.aqara.com/Canon_Smart_Wall_Switch_Z1_overview

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0x54ef4410009ea053","nwkAddr":29649,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.switch.acn055","epList":[1,2,3,4,21],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[2820,1794,18,5,4,6,3,0,64704],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"appVersion":31,"modelId":"lumi.switch.acn055"}},"genOnOff":{"attributes":{"245":57921792,"onOff":0}},"genMultistateInput":{"attributes":{"presentValue":1}},"aqaraOpple":{"attributes":{"4":4,"223":{"type":"Buffer","data":[3,35,36,211,0,0,4,33,10,2,5,33,33,0,41,35,194,116,46,105,42,35,71,179,0,8,40,35,64,0,0,0,43,33,0,0,44,41,191,255]},"247":{"type":"Buffer","data":[3,40,21,5,33,20,0,9,33,0,0,10,33,0,0,12,32,10,13,35,31,0,0,0,17,35,1,0,0,0,100,16,0,101,16,0,102,16,0,149,57,0,0,0,0,154,32,0]},"512":1,"617":1,"645":0,"646":1,"648":1,"651":1,"1303":1}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[18,5,4,6,64704],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"245":117440515,"onOff":0}},"genMultistateInput":{"attributes":{"presentValue":1}},"aqaraOpple":{"attributes":{"512":1,"617":1,"645":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[18,5,4,6,64704],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"245":117440716,"onOff":0}},"genMultistateInput":{"attributes":{"presentValue":1}},"aqaraOpple":{"attributes":{"512":1,"617":1,"645":0}}},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[18,64704],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}},"aqaraOpple":{"attributes":{"617":0}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":0,"inClusterList":[12],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":31,"stackVersion":27,"hwVersion":1,"dateCode":"20231102","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1703400823943,"defaultSendRequestWhen":"immediate"}

### Comments

The switch can currently be controlled, but I don't know how to parse aqaraOpple, and I don't understand what it means. I need help, thank you.

{"attributes":{"4":4,"223":{"type":"Buffer","data":[3,35,36,211,0,0,4,33,10,2,5,33,33,0,41,35,194,116,46,105,42,35,71,179,0,8,40,35,64,0,0,0,43,33,0,0,44,41,191,255]},"247":{"type":"Buffer","data":[3,40,21,5,33,20,0,9,33,0,0,10,33,0,0,12,32,10,13,35,31,0,0,0,17,35,1,0,0,0,100,16,0,101,16,0,102,16,0,149,57,0,0,0,0,154,32,0]},"512":1,"617":1,"645":0,"646":1,"648":1,"651":1,"1303":1}}

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    aqara_z1_opple: {
        cluster: 'aqaraOpple',
        type: ['readResponse', 'attributeReport'],
        convert: (model, msg, publish, options, meta) => {
                console.log('model:' + JSON.stringify(model));
                console.log('msg:' + JSON.stringify(msg));
                console.log('publish:' + JSON.stringify(publish));
                console.log('options:' + JSON.stringify(options));
                console.log('meta:' + JSON.stringify(meta));
        //      console.log('---' + Buffer.from(msg.data['247'], 'ascii').toString('hex'));
        },
    },
};

const definition = {
    zigbeeModel: ['lumi.switch.acn055'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ZNQBKG41LM', // Vendor model number, look on the device for a model number
    vendor: 'Xiaomi', // Vendor of the device (only used for documentation and startup logging)
    description: 'Aqara Z1', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [
            fz.on_off,
            fzLocal.aqara_z1_opple,
            fz.xiaomi_multistate_action,
            fz.aqara_opple_multistate,
    ], // We will add this later
    toZigbee: [
            tz.on_off,
            tz.xiaomi_switch_operation_mode_opple,
            tz.xiaomi_switch_power_outage_memory,
            tz.aqara_switch_mode_switch,
            tz.xiaomi_flip_indicator_light
    ], // Should be empty, unless device can be controlled (e.g. lights, switches).
    meta: {multiEndpoint: true},
    exposes: [
        e.switch().withEndpoint('1'),
        e.switch().withEndpoint('2'),
        e.switch().withEndpoint('3'),

    
    ], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

Koenkk, 2023-12-25:
Try with the following fromZigbee converter: `xiaomi.fromZigbee.aqara_opple`, make sure to import it: `const xiaomi = require('zigbee-herdsman-converters/lib/xiaomi');`

g1335333249, 2023-12-27:
Thank you, this can be parsed

closed 2023-12-27 by g1335333249

Koenkk, 2023-12-27:
Great, please create a pull request to add out-of-the-box support for this device.