Koenkk/zigbee2mqtt #20370: Failed to start zigbee - slae.sh_cc2652rb

lordmuni, 2023-12-24:
### What happened?

Hello everyone,

I'm encountering an issue when trying to start Zigbee2MQTT on my Debian GNU/Linux 12 system, and I'm seeking assistance to resolve it. Here is the detailed error I'm facing
I have followed the official documentation and checked for possible issues on the [Zigbee2MQTT troubleshooting page](https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html), but I haven't been able to resolve the problem.

My current setup uses Proxmox and LXC, and I have followed the steps outlined on the [Zigbee2MQTT installation page for Proxmox](https://tteck.github.io/Proxmox/).

I would appreciate any guidance or suggestions to resolve this issue. Has anyone encountered a similar problem or have any ideas on addressing this specific error?

Thank you in advance for your assistance!

### Zigbee2MQTT version

zigbee2mqtt@1.34.0 start

### Adapter firmware version

usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_25_9B_EB_82-if00

### Adapter

Slaesh's CC2652RB stick

### Debug log

Debian GNU/Linux 12 zigbee2mqtt tty1

zigbee2mqtt login: root (automatic login)

Zigbee2MQTT LXC provided by https://tteck.github.io/Proxmox/

Last login: Sat Dec 23 16:32:53 CET 2023 on tty1
You have mail.
root@zigbee2mqtt:~# cd /opt/zigbee2mqtt && npm start

zigbee2mqtt@1.34.0 start
node index.js

Zigbee2MQTT:info 2023-12-23 16:50:45: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2023-12-23.16-50-45' filename: log.txt
Zigbee2MQTT:info 2023-12-23 16:50:45: Starting Zigbee2MQTT version 1.34.0 (commit #56589dcc)
Zigbee2MQTT:info 2023-12-23 16:50:45: Starting zigbee-herdsman (0.25.0)
Zigbee2MQTT:error 2023-12-23 16:50:45: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-12-23 16:50:45: Failed to start zigbee
Zigbee2MQTT:error 2023-12-23 16:50:45: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-12-23 16:50:45: Exiting...
Zigbee2MQTT:error 2023-12-23 16:50:46: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_25_9B_EB_82-if00'
at SerialPort. (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
at SerialPort._error (/opt/zigbee2mqtt/node_modules/@serialport/stream/dist/index.js:82:22)
at /opt/zigbee2mqtt/node_modules/@serialport/stream/dist/index.js:118:18
root@zigbee2mqtt:/opt/zigbee2mqtt# `

configuration.yaml
homeassistant: true
permit_join: false
mqtt:
base_topic: zigbee2mqtt
server: mqtt://192.168.100.16:1883
user: xxxxx
password: xxxxx
keepalive: 60
reject_unauthorized: true
version: 4
serial:
port: >-
/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_25_9B_EB_82-if00
advanced:
pan_id: 44852
network_key:
- 114
- 73
- 0
- 35
- 252
- 3
- 215
- 240
- 45
- 203
- 182
- 59
- 45
- 159
- 58
- 184
channel: 20

T-IOOO, 2023-12-25:
Go to your /dev/serial folder and see if you have a folder named ”by-path” there. Use that location instead of the ”by-serial”.

lordmuni, 2023-12-25:
Thank you for your suggestion, T-IOOO. I appreciate your input and have already tried using the /dev/serial/by-path directory. Here is what I did and the outcome:

Checked for /dev/serial/by-path Directory:

I executed the command ls -l /dev/serial/by-path to check if the directory existed and to list the available devices there.
Updated configuration.yaml File:

I found a link in /dev/serial/by-path corresponding to my Zigbee adapter (Slaesh's CC2652RB stick) and updated the configuration.yaml file to use this path instead of /dev/serial/by-id/....
The change in the configuration.yaml looked like this:
yaml
Copy code
serial:
  port: /dev/serial/by-path/[found-path]
Restarted Zigbee2MQTT:

After making this change, I restarted Zigbee2MQTT to apply the new configuration.
Unfortunately, this approach did not resolve the issue, and I'm still experiencing the same problem. If you or anyone else has any other suggestions or ideas, I would be grateful to hear them. Thank you again for your assistance!

lordmuni, 2023-12-25:
I wanted to provide an update and happily report that I have resolved the issue with Zigbee2MQTT on my system. After further investigation and trial, I found that the solution was to delete the database.db, state.json, and log files, and then perform a restart. Here are the specific steps I took:

Delete Specific Files:

I navigated to the Zigbee2MQTT data directory and removed the database.db, state.json, and all log files. These files can sometimes contain corrupt data or state information that prevents Zigbee2MQTT from starting correctly.
Restart Zigbee2MQTT:

After deleting these files, I restarted Zigbee2MQTT. This allowed the service to create fresh instances of these files without any previous errors or corrupted data.
This process successfully resolved the startup issue I was experiencing. Zigbee2MQTT is now running smoothly on my system.

I wanted to share this solution in case anyone else encounters a similar problem. Thank you to everyone who provided suggestions and assistance!

closed 2023-12-25 by lordmuni