Koenkk/zigbee2mqtt #20372: TO-Q-SY1-JZT precision/decimal places not working

nireins, 2023-12-24:
### What happened?

Hi, i use some TO-Q-SY1-JZT and i change the power_precision, current_precision and voltage_precision to 3, but only the current_precision shows 3 decimal places. 

[image]

Am I doing something wrong or can the device not do this?

### What did you expect to happen?

That the decimal places are displayed after setting.

### How to reproduce it (minimal and precise)

[image]

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

7.3.1.0 build 176

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

mrmaximas, 2023-12-24:
@nireins i'ts absolutely normal situation. If the device outputs integers, then whatever precision you set it to, it will be integers.

Koenkk, 2024-06-22:
Z2M can only reduce the precision, not increase it (since the device simply doesn't report it).

closed 2024-06-22 by Koenkk