Koenkk/zigbee2mqtt #20376: [Feature request]: add an option to assign Home Assistant entity_category from custom converter

osnwt, 2023-12-25:
### Is your feature request related to a problem? Please describe

Writing converter for DIY device for Home Assistant, I have not found an option to declare some entities as configuration or setting, etc. It seems to be hardcoded for known cases, but would be nice to have such option for converter, too.

### Describe the solution you'd like

Add an option to override entity_category in converters for Home Assistant.

### Describe alternatives you've considered

Maybe I missed such an option?

### Additional context

N/A

slugzero, 2024-01-09:
Thank you for linking this to my feature request, and sorry, I wasn't aware of yours when I opened it.
With the latest `dev` version, you should now be able to do what you like by adding `.withCategory('config')` or `'diagnostic'` to your exposes.

osnwt, 2024-01-09:
Thanks, excellent solution.
I'd close it as soon as it comes to a released version.

closed 2024-08-07 by github-actions[bot]