Koenkk/zigbee2mqtt #20381: Conbee III

sw224, 2023-12-25:
### What happened?

Today I installed the conbee III coordinator, all devices work properly except the Aqara E1 roller shutter motor. The system adds it as an unsupported device. Everything worked correctly on the conbee II coordinator.

### Zigbee2MQTT version

 1.34.0-1

### Adapter firmware version

DE03110417

### Adapter

Conbee III

T-IOOO, 2023-12-25:
I have slae.sh stick and homebridge adds an unsupported device ”coordinator” that crashes the entire z2m-plugin. A manual removal of the device from homebridge solves the problem temporally

GustavFBrandsen, 2023-12-26:
I have exactly the same problem with the E1 Roller motor, and the ConBee III.
Have you found a solution to the problem?

sw224, 2023-12-26:
Unfortunately not yet.

GustavFBrandsen, 2023-12-26:
https://phoscon.de/en/conbee3/compatible

This list doesn't have the E1 Driver on it... It seems like someone needs to make it compatible first...

GustavFBrandsen, 2024-02-09:
To anyone needing this: I managed to get it paired by updating the ConBee 3 to firmware 0x26500900 and Z2M at 1.35.3

dyvyak, 2024-05-08:
conbee III doesn't work well with z2m. conbee II is much better. i learned that by hard way...

closed 2024-12-05 by github-actions[bot]