Koenkk/zigbee2mqtt #20383: [New device support]:

fresnoboy, 2023-12-26:
### Link

https://parts.andersenwindows.com/resources/pdf/Yale%20Assure%20Lock_Install%20Guide.pdf

### Database entry

{"id":71,"type":"EndDevice","ieeeAddr":"0x000d6f0018adca33","nwkAddr":40829,"manufId":4125,"manufName":"Yale","powerSource":"Battery","modelId":"YRM476 TS BLE","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":10,"inClusterList":[0,1,3,9,10,257,32,2821],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"YRM476 TS BLE","manufacturerName":"Yale","powerSource":3,"zclVersion":1,"appVersion":255,"hwVersion":2}},"genPollCtrl":{"attributes":{"checkinInterval":14400}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c2b90e","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":255,"hwVersion":2,"zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1703559843151,"defaultSendRequestWhen":"active","checkinInterval":3600}

### Comments

This lock, like the slightly different YRM276, are yale assure locks like the Yale YRD246HA20BP which is supported: https://www.zigbee2mqtt.io/devices/YRD246HA20BP.html  These locks are special models of assure locks that fit in Andersen Patio doors: https://shopyalehome.com/products/assure-lock-for-andersen-patio-doors-with-wi-fi-and-bluetooth?variant=39337803710596

How can I clone the support for that lock to the YRM 476 (which should also work on the YRM 276)?

Koenkk, 2023-12-26:
Could you check if the device works out-of-the-box in the latest dev branch? Both battery and locking should work.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

fresnoboy, 2023-12-26:
I just updated to latest-dev (I run the docker version), and the when trying to reconfigure it, it says this device cannot be configured.  Do I have to remove the device and rejoin it?

Koenkk, 2023-12-26:
Try updating to latest-dev again, I pushed the changes 30 mins ago.

fresnoboy, 2023-12-26:
Just tried it.  Same result.  Device cannot be configured error.

Koenkk, 2023-12-26:
What zigbee-herdsman-converters version shows up in the z2m frontend -> settings -> about? Should be 17.1.1 or higher

fresnoboy, 2023-12-26:
You are right, it's not updated.  But I did the pull with latest-dev as the option, and even did a docker rm -f.  But that didn't seem to trigger the update properly.  What am I missing?

fresnoboy, 2023-12-26:
NM.  Changed the docker .yml image statement to koenkk/zigbee2mqtt:latest-dev  The front now shows version 18.0.0 for the Zigbee-herdsman-converters.  And every device looks like it got reconfigured.  

The device is also now configurable, but the device description shows as not supported, and it doesn't show up in the dashbaord at all.

Koenkk, 2023-12-27:
What external definition is now generated? ([docs](https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/advanced/support-new-devices/01_support_new_devices.md#2-creating-the-external-definition))

fresnoboy, 2023-12-28:
Here is the generated definition:

const {batteryPercentage, lock} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['YRM476 TS BLE'],
    model: 'YRM476 TS BLE',
    vendor: 'Yale',
    description: 'Automatically generated definition',
    extend: [batteryPercentage(), lock({pinCodeCount: 250})],
};

module.exports = definition;

And here's what's exposed:

[image]

Koenkk, 2023-12-28:
Thanks, added it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-28 by Koenkk

fresnoboy, 2023-12-29:
Just tried it.  Version is now 18.02.  Still doesn't configure properly

Here is the generated definition:

const {batteryPercentage, lock} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['YRM476 TS BLE'],
    model: 'YRM476 TS BLE',
    vendor: 'Yale',
    description: 'Automatically generated definition',
    extend: [batteryPercentage(), lock({pinCodeCount: 250})],
};

module.exports = definition;

Koenkk, 2023-12-29:
Forgot to merge yesterday, it's not included in 18.0.2,will do it today.

fresnoboy, 2023-12-29:
As of 18.1.0, the YRM 476 does configure properly!  Basic lock function seems to work just fine, but I don't see any configuration  info for things like users and PINs in the HA device  created through MQTT.  I see that a bunch of info is exposed in the Z2M device information, but how do I get at it in HA?

BTW, this is super useful.  I have replaced most of my Zwave stuff with zigbee devices, but I have a set of these locks that forces me to deal with the mediocre Zwave support in HA.  If I can get full functionality via zigbee, I'll replace all of the lock modules and everything else still on Zwave and be done with that.

Koenkk, 2023-12-30:
HA does not support the more complex entities, you can use the `mqtt.publish` service of HA to publish MQTT messages ([docs of a similar lock](https://www.zigbee2mqtt.io/devices/YDD-D4F0-TSDB.html))