Koenkk/zigbee2mqtt #20384: The battery status of QBKG20LM is not recognized correctly.

hongliangl, 2023-12-26:
### What happened?

I have some devices whose model is `QBKG20LM`. The `Device type` of the devices should be recognized as `Router`. However, some of the them are recognized as `EndDevice` and the value of `Power` is `Battery ?`. I have tried repaired them several times but got the same result every time.

The following image is the device with the issue:

[image]

The following image is the device as expected:

[image]

### Zigbee2MQTT version

1.34.0 commit: aae7312

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

Koenkk, 2023-12-26:
Could you provide the data/database.db entry of this device?

hongliangl, 2023-12-26:
> Could you provide the data/database.db entry of this device?

Thank you for the prompt reply @Koenkk !

The entries are listed below:

```json
{
   "id":14,
   "type":"EndDevice",
   "ieeeAddr":"0x54ef44100060406e",
   "nwkAddr":431,
   "manufId":4151,
   "manufName":"LUMI",
   "powerSource":"Battery",
   "modelId":"lumi.switch.b2nacn01",
   "epList":[
      1,
      2,
      21,
      41,
      42
   ],
   "endpoints":{
      "1":{
         "epId":1,
         "inClusterList":[],
         "outClusterList":[],
         "clusters":{
            "aqaraOpple":{
               "attributes":{
                  "220":{
                     "type":"Buffer",
                     "data":[
                        197,
                        253,
                        3,
                        0,
                        202,
                        74,
                        175,
                        1,
                        0,
                        0,
                        255,
                        233
                     ]
                  },
                  "223":{
                     "type":"Buffer",
                     "data":[
                        22,
                        33,
                        0,
                        0,
                        23,
                        33,
                        0,
                        0,
                        12,
                        33,
                        26,
                        0,
                        24,
                        33,
                        235,
                        11,
                        30,
                        33,
                        50,
[... 687 lines cut ...]
      "242":{
         "profId":41440,
         "epId":242,
         "devId":97,
         "inClusterList":[],
         "outClusterList":[
            33
         ],
         "clusters":{},
         "binds":[],
         "configuredReportings":[],
         "meta":{}
      }
   },
   "appVersion":26,
   "stackVersion":2,
   "hwVersion":1,
   "dateCode":"11-01-2021",
   "zclVersion":3,
   "interviewCompleted":true,
   "meta":{},
   "lastSeen":1703586201981,
   "defaultSendRequestWhen":"immediate"
}
```

hongliangl, 2023-12-26:
This is the original data:

```json
{"id":13,"type":"EndDevice","ieeeAddr":"0x54ef441000603d80","nwkAddr":63417,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.switch.b2nacn01","epList":[1,2,21,31,41,42,51,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2,3,4,5,6,18,64704],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.switch.b2nacn01","appVersion":26,"dateCode":"11-01-2021"}},"aqaraOpple":{"attributes":{"220":{"type":"Buffer","data":[239,244,3,0,202,74,185,247,0,0,255,233]},"223":{"type":"Buffer","data":[22,33,0,0,23,33,0,0,12,33,19,0,24,33,137,7,30,33,33,0,31,33,150,13,32,33,211,198,34,33,150,70,35,33,0,0,36,32,0,37,35,248,21,240,226,250,33,0,0,251,33,0,0,252,35,0,0,0,0,253,35,0,0,0,0,254,35,16,0,0,2]},"229":{"type":"Buffer","data":[16,0,16,202,74,5,119,237,221,18,98,36,45,18,58,105,9,37,78,230,34,37,59,51,46,37,87,29,172,37,92,195,253,37,104,6,234,37,83,165,250,37,77,228,47,37,74,175,1,37,92,248,17,37,40,0,0,36,104,216,75,37,46,244,239,37,67]},"240":0,"247":{"type":"Buffer","data":[100,16,0,101,16,1,3,40,26,152,57,195,117,79,66,149,57,183,153,82,64,150,57,0,32,20,69,151,57,245,214,90,67,5,33,29,0,154,32,0,8,33,26,1,9,33,2,17,11,32,0,10,33,0,0,12,32,1]},"513":1,"515":0,"670":0}},"genOnOff":{"attributes":{"245":117440571,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6,18,64704],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"245":66566400,"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":256,"inClusterList":[12],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":53.21799850463867}}},"binds":[],"configuredReportings":[],"meta":{}},"31":{"profId":260,"epId":31,"devId":256,"inClusterList":[12],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"41":{"profId":260,"epId":41,"devId":256,"inClusterList":[18],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"42":{"profId":260,"epId":42,"devId":256,"inClusterList":[18],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"51":{"profId":260,"epId":51,"devId":256,"inClusterList":[18],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":26,"stackVersion":2,"hwVersion":1,"dateCode":"11-01-2021","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1703586201981,"defaultSendRequestWhen":"immediate"}
{"id":14,"type":"EndDevice","ieeeAddr":"0x54ef44100060406e","nwkAddr":431,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.switch.b2nacn01","epList":[1,2,21,41,42],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"aqaraOpple":{"attributes":{"220":{"type":"Buffer","data":[197,253,3,0,202,74,175,1,0,0,255,233]},"223":{"type":"Buffer","data":[22,33,0,0,23,33,0,0,12,33,26,0,24,33,235,11,30,33,50,0,31,33,9,14,32,33,250,193,34,33,67,70,35,33,1,0,36,32,0,37,35,34,25,156,221,250,33,0,0,251,33,0,0,252,35,0,0,0,0,253,35,0,0,0,0,254,35,16,0,0,2]},"229":{"type":"Buffer","data":[16,0,16,202,74,5,75,56,159,18,98,105,9,37,61,230,34,37,89,51,46,37,70,165,250,37,40,216,75,37,106,242,229,37,55,29,172,37,54,0,0,36,79,195,253,37,37,248,17,37,42,185,247,37,93,6,234,37,62,244,239,37,96,228,47,37,82]},"247":{"type":"Buffer","data":[100,16,0,101,16,0,3,40,25,152,57,0,0,0,0,149,57,51,83,1,63,150,57,0,64,18,69,151,57,0,0,0,0,5,33,30,0,154,32,0,8,33,26,1,9,33,1,16,11,32,0,10,33,0,0,12,32,1]},"512":0,"670":0}},"genBasic":{"attributes":{"modelId":"lumi.switch.b2nacn01","appVersion":26,"dateCode":"11-01-2021"}},"genOnOff":{"attributes":{"245":33664768,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"245":117440565,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"epId":21,"inClusterList":[],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[],"configuredReportings":[],"meta":{}},"41":{"epId":41,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"42":{"epId":42,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":26,"dateCode":"11-01-2021","interviewCompleted":true,"meta":{},"lastSeen":1703586107343,"defaultSendRequestWhen":"immediate"}
```

Koenkk, 2023-12-26:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) tomorrow.

closed 2023-12-26 by Koenkk

hongliangl, 2023-12-27:
The devices work as `Router` right now after updating to the latest-dev branch!