Koenkk/zigbee2mqtt #20387: Device ”coordinator” shows up in homekit

T-IOOO, 2023-12-26:
### What happened?

Device named ”coordinator” shows up in homekit. I am using z2m-homebridge. I am unaböe to remove the device from chached devices since it will only reappear and make the plugin crash. configuration.yaml is not showing this device at all. What could cause this?

### What did you expect to happen?

Why would the coordinator show up in the first place?

### How to reproduce it (minimal and precise)

I just updated to 1.34.0-dev commit: 56f57625

### Zigbee2MQTT version

1.34.0-dev commit: 56f57625

### Adapter firmware version

20230507

### Adapter

Slase.sh stick

Koenkk, 2023-12-26:
Please create an issue in the z2m-homebridge project, this is not something that can be fixed directly from z2m.

closed 2023-12-26 by Koenkk

T-IOOO, 2023-12-26:
> Please create an issue in the z2m-homebridge project, this is not something that can be fixed directly from z2m.

I wasn’t sure where to post this and I figured since I updated to the latest dev release and it all started. Besides I have tried to remove the device from the cached devices and still it reappears as if zigbee2mqtt is sending out this device somehow. But I will ask in the homebridge community