Koenkk/zigbee2mqtt #20388: [Wrong device]: TuYa TH02Z wrongly being detected as TuYa TS0201

MatthewMiele, 2023-12-26:
### Link

https://www.zigbee2mqtt.io/devices/TH02Z.html

### Model

TH02Z

### Description

Temperature and humidity sensor

### Vendor

TuYa

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/TH02Z.jpg

### Database entry

{             "date_code": "",             "definition": {                 "description": "Temperature & humidity sensor with display",                 "exposes": [                     {                         "access": 1,                         "description": "Remaining battery in %",                         "name": "battery",                         "property": "battery",                         "type": "numeric",                         "unit": "%",                         "value_max": 100,                         "value_min": 0                     },                     {                         "access": 1,                         "description": "Measured temperature value",                         "name": "temperature",                         "property": "temperature",                         "type": "numeric",                         "unit": "°C"                     },                     {                         "access": 1,                         "description": "Measured relative humidity",                         "name": "humidity",                         "property": "humidity",                         "type": "numeric",                         "unit": "%"                     },                     {                         "access": 1,                         "description": "Voltage of the battery in millivolts",                         "name": "voltage",                         "property": "voltage",                         "type": "numeric",                         "unit": "mV"                     },                     {                         "access": 1,                         "description": "Link quality (signal strength)",                         "name": "linkquality",                         "property": "linkquality",                         "type": "numeric",                         "unit": "lqi",                         "value_max": 255,                         "value_min": 0                     }                 ],                 "model": "TS0201",                 "options": [                     {                         "access": 2,                         "description": "Number of digits after decimal point for temperature, takes into effect on next report of device.",                         "name": "temperature_precision",                         "property": "temperature_precision",                         "type": "numeric",                         "value_max": 3,                         "value_min": 0                     },                     {                         "access": 2,                         "description": "Calibrates the temperature value (absolute offset), takes into effect on next report of device.",                         "name": "temperature_calibration",                         "property": "temperature_calibration",                         "type": "numeric"                     },                     {                         "access": 2,                         "description": "Number of digits after decimal point for humidity, takes into effect on next report of device.",                         "name": "humidity_precision",                         "property": "humidity_precision",                         "type": "numeric",                         "value_max": 3,                         "value_min": 0                     },                     {                         "access": 2,                         "description": "Calibrates the humidity value (absolute offset), takes into effect on next report of device.",                         "name": "humidity_calibration",                         "property": "humidity_calibration",                         "type": "numeric"                     }                 ],                 "supports_ota": false,                 "vendor": "TuYa"             },             "endpoints": {                 "1": {                     "bindings": [],                     "clusters": {                         "input": [                             "genPowerCfg",                             "genIdentify",                             "msTemperatureMeasurement",                             "msRelativeHumidity",                             "genBasic"                         ],                         "output": [                             "genIdentify",                             "genOta",                             "genTime"                         ]                     },                     "configured_reportings": [],                     "scenes": []                 }             },             "friendly_name": "0xa4c138a70b98dd6f",             "ieee_address": "0xa4c138a70b98dd6f",             "interview_completed": true,             "interviewing": false,             "manufacturer": "_TZ3000_xr3htd96",             "model_id": "TS0201",             "network_address": 28050,             "power_source": "Battery",             "supported": true,             "type": "EndDevice"         }

### Notes

I have set up 10 new [TuYa TH02Z
]([url](https://www.zigbee2mqtt.io/devices/TH02Z.html#tuya-th02z)) but they have all been detected as [TS0201](https://www.zigbee2mqtt.io/devices/TS0201.html#tuya-ts0201), showing the wrong images and details.

Also, the humidity, temperature and link quality appears to be working, but the battery level and voltage is not reported.

MatthewMiele, 2023-12-27:
closing this issue as updating zigbee2mqtt has resolved the issue. its now correctly identifying the sensors

closed 2023-12-27 by MatthewMiele

tanus10, 2024-12-15:
Sorry to awake an old thread.

_TZ3000_yujem9ee 

has the very same problem.