Koenkk/zigbee2mqtt #20396: The TS0601_cover_1 （TZE200_ol5jlkkr） device does not report the 'commandDataResponse' message.

1020844469, 2023-12-27:
### What happened?

The TS0601_cover_1 （TZE200_ol5jlkkr） device does not report the 'commandDataResponse' message.

When first joining the network, everything is normal. However, after a certain period of use, although it can be controlled, it does not report the 'commandDataResponse' Zigbee message after control. In terms of operation, for example, if I control it to open to 70%, and then stop it when the curtain is at 30%, under normal circumstances, it should display the correct percentage of 30%. However, because it did not receive the 'commandDataResponse' message, it does not show the correct position.

The normal behavior should be like this:
Received Zigbee message from 'livingroom cover', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1},{"data":{"data":[0,0,0,75],"type":"Buffer"},"datatype":2,"dp":2},{"data":{"data":[0,0,0,30],"type":"Buffer"},"datatype":2,"dp":3}],"seq":0}' from endpoint 1 with groupID 0
But since this message is not received, the current position of the curtain is never known

At this point, after re-pairing (without removing from the network), it returns to normal and can report the status. The occurrence of this issue varies, sometimes taking a few days, sometimes a dozen days. For this test, specifically using three devices, two of them experienced this problem after a week, while the other one is temporarily normal.

To rule out issues with the motor itself, three additional devices are being tested with UYA's own gateway. So far, they are all normal and have not experienced this situation.

### What did you expect to happen?

After you control the curtain in Zigbee2MQTT or Home Assistant
The normal behavior should be like this:
Received Zigbee message from 'livingroom cover', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1},{"data":{"data":[0,0,0,75],"type":"Buffer"},"datatype":2,"dp":2},{"data":{"data":[0,0,0,30],"type":"Buffer"},"datatype":2,"dp":3}],"seq":0}' from endpoint 1 with groupID 0
But since this message is not received, the current position of the curtain is never known

### How to reproduce it (minimal and precise)

At this point, after re-pairing (without removing from the network), it returns to normal and can report the status. The occurrence of this issue varies, sometimes taking a few days, sometimes a dozen days. For this test, specifically using three devices, two of them experienced this problem after a week, while the other one is temporarily normal.

To rule out issues with the motor itself, three additional devices are being tested with UYA's own gateway. So far, they are all normal and have not experienced this situation.

### Zigbee2MQTT version

1.33.0 commit: 7ee207f

### Adapter firmware version

20221226

### Adapter

SONOFF CC2652P Zigbee 3.0 USB Dongle Plus

### Debug log

incorrect:
Debug 2023-12-27 13:16:22Received MQTT message on 'zigbee2mqtt2/liveroom_cover/set' with data '{"position":32}'
Debug 2023-12-27 13:16:22Publishing 'set' 'position' to 'liveroom_cover'
Debug 2023-12-27 13:16:23Received MQTT message on 'zigbee2mqtt2/liveroom_cover/set' with data '{"state":"STOP"}'
Debug 2023-12-27 13:16:23Publishing 'set' 'state' to 'liveroom_cover'

correct:
ebug 2023-12-27 13:18:19Received MQTT message on 'zigbee2mqtt2/masterroom_cover/set' with data '{"position":43}'
Debug 2023-12-27 13:18:19Publishing 'set' 'position' to 'masterroom_cover'
Debug 2023-12-27 13:18:19Received Zigbee message from 'masterroom_cover', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,57],"type":"Buffer"},"datatype":2,"dp":2}],"seq":0}' from endpoint 1 with groupID 0
Info 2023-12-27 13:18:19MQTT publish: topic 'zigbee2mqtt2/masterroom_cover', payload '{"last_seen":"2023-12-27T13:18:19+08:00","linkquality":112,"options":null,"position":43,"running":true,"state":"OPEN"}'
Debug 2023-12-27 13:18:19Received Zigbee message from 'masterroom_cover', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":7}],"seq":0}' from endpoint 1 with groupID 0
Info 2023-12-27 13:18:19MQTT publish: topic 'zigbee2mqtt2/masterroom_cover', payload '{"last_seen":"2023-12-27T13:18:19+08:00","linkquality":112,"options":null,"position":43,"running":true,"state":"OPEN"}'
Debug 2023-12-27 13:18:20Received MQTT message on 'zigbee2mqtt2/masterroom_cover/set' with data '{"state":"STOP"}'
Debug 2023-12-27 13:18:20Publishing 'set' 'state' to 'masterroom_cover'
Debug 2023-12-27 13:18:20Received Zigbee message from 'masterroom_cover', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1}],"seq":0}' from endpoint 1 with groupID 0
Info 2023-12-27 13:18:20MQTT publish: topic 'zigbee2mqtt2/masterroom_cover', payload '{"last_seen":"2023-12-27T13:18:20+08:00","linkquality":112,"options":null,"position":43,"running":true,"state":"OPEN"}'
Debug 2023-12-27 13:18:20Received Zigbee message from 'masterroom_cover', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1},{"data":{"data":[0,0,0,73],"type":"Buffer"},"datatype":2,"dp":2},{"data":{"data":[0,0,0,73],"type":"Buffer"},"datatype":2,"dp":3}],"seq":0}' from endpoint 1 with groupID 0
Info 2023-12-27 13:18:20MQTT publish: topic 'zigbee2mqtt2/masterroom_cover', payload '{"last_seen":"2023-12-27T13:18:20+08:00","linkquality":112,"options":null,"position":27,"running":false,"state":"OPEN"}'

zvezdochet2004, 2024-01-25:
Hello. I have the same problem

closed 2024-08-23 by github-actions[bot]

rauko27, 2024-11-05:
Hello. I have the same problem too.