Koenkk/zigbee2mqtt #20400: IKEA TRÅDFRI LED2103G5 : No converter available for 'state' ("")

heviiguy, 2023-12-27:
### What happened?

Bulb does not toggle

- Initial pairing attempt resulted in bulb being described as "Unsupported".
- After creating the external converter file, the bulb was recognised but, there are no results when toggling the state in Zigbee2MQTT under "Exposes". Likewise when attempted in Home Assistance dashboard.
- MQTT Explorer shows that the message had been published to the bulb. 

### What did you expect to happen?

Expected the bulb to toggle between "on" and "off"

### How to reproduce it (minimal and precise)

As abovedebug 2023-12-27 10:25:40Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,165,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
debug 2023-12-27 10:27:13Saving state to file /app/data/state.json
debug 2023-12-27 10:27:32Received MQTT message on 'zigbee2mqtt/Light - (hevii Office - overhead)/set' with data '{"state":"OFF"}'
error 2023-12-27 10:27:32No converter available for 'state' ("OFF")
debug 2023-12-27 10:27:45Received MQTT message on 'zigbee2mqtt/Light - (hevii Office - overhead)/set' with data '{"state":"ON"}'
error 2023-12-27 10:27:45No converter available for 'state' ("ON")
debug 2023-12-27 10:27:48Received MQTT message on 'zigbee2mqtt/Light - (hevii Office - overhead)/get' with data '{"state":""}'
error 2023-12-27 10:27:48No converter available for 'state' ("")
debug 2023-12-27 10:27:56Received MQTT message on 'zigbee2mqtt/Light - (hevii Office - overhead)/get' with data '{"state":""}'
error 2023-12-27 10:27:56No converter available for 'state' ("")

### Zigbee2MQTT version

1.34.0 commit: aae7312

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff ZBDongle-E

### Debug log

```
debug 2023-12-27 10:25:40Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,165,2,11,254,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
debug 2023-12-27 10:27:13Saving state to file /app/data/state.json
debug 2023-12-27 10:27:32Received MQTT message on 'zigbee2mqtt/Light - (hevii Office - overhead)/set' with data '{"state":"OFF"}'
error 2023-12-27 10:27:32No converter available for 'state' ("OFF")
debug 2023-12-27 10:27:45Received MQTT message on 'zigbee2mqtt/Light - (hevii Office - overhead)/set' with data '{"state":"ON"}'
error 2023-12-27 10:27:45No converter available for 'state' ("ON")
debug 2023-12-27 10:27:48Received MQTT message on 'zigbee2mqtt/Light - (hevii Office - overhead)/get' with data '{"state":""}'
error 2023-12-27 10:27:48No converter available for 'state' ("")
debug 2023-12-27 10:27:56Received MQTT message on 'zigbee2mqtt/Light - (hevii Office - overhead)/get' with data '{"state":""}'
error 2023-12-27 10:27:56No converter available for 'state' ("")
```

Koenkk, 2023-12-27:
Try with the latest [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) and no external converter

heviiguy, 2023-12-27:
Tried it with [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) to no avail.

I now get:
"Device '0x90395efffe6b12af' with Zigbee model 'TRADFRI bulb E26 WW globe 800lm' and manufacturer name 'IKEA of Sweden' is NOT supported"

Unless the folks at the factory screwed-up and mis-labled to bulb, I'm certain that this is indeed a TRÅDFRI LED2103G5. I'll invoke the external converter again...

Koenkk, 2023-12-27:
Added it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) tomorrow

closed 2023-12-27 by Koenkk

heviiguy, 2023-12-27:
Thanks, Koen!

And, "Happy Holidays" 🎅🏼

heviiguy, 2023-12-28:
I've installed Zigbee2MQTT v1.34.0 aae7312)

Alas, no joy 🙁. The bulb still comes back as "unrecognised".

```
info 2023-12-28 10:57:42Starting network scan (includeRoutes 'false')
info 2023-12-28 10:57:44Network scan finished
info 2023-12-28 10:57:44MQTT publish: topic 'zigbee2mqtt/bridge/response/networkmap', payload '{"data":{"routes":false,"type":"raw","value":{"links":[{"depth":15,"linkquality":68,"lqi":68,"relationship":2,"routes":[],"source":{"ieeeAddr":"0xb43522fffea8706e","networkAddress":15240},"sourceIeeeAddr":"0xb43522fffea8706e","sourceNwkAddr":15240,"target":{"ieeeAddr":"0xb43522fffe015f1d","networkAddress":0},"targetIeeeAddr":"0xb43522fffe015f1d"},{"depth":0,"linkquality":44,"lqi":44,"relationship":0,"routes":[],"source":{"ieeeAddr":"0xb43522fffe015f1d","networkAddress":0},"sourceIeeeAddr":"0xb43522fffe015f1d","sourceNwkAddr":0,"target":{"ieeeAddr":"0xb43522fffea8706e","networkAddress":15240},"targetIeeeAddr":"0xb43522fffea8706e"}],"nodes":[{"definition":null,"failed":[],"friendlyName":"Coordinator","ieeeAddr":"0xb43522fffe015f1d","lastSeen":1703775455916,"networkAddress":0,"type":"Coordinator"},{"definition":null,"failed":[],"friendlyName":"0xb43522fffea8706e","ieeeAddr":"0xb43522fffea8706e","lastSeen":1703775462582,"networkAddress":15240,"type":"Unknown"}]}},"status":"ok","transaction":"gbacy-3"}'
info 2023-12-28 10:57:51Successfully interviewed '0xb43522fffea8706e', device has successfully been paired
warning 2023-12-28 10:57:51Device '0xb43522fffea8706e' with Zigbee model 'TRADFRI bulb E26 WW globe 800lm' and manufacturer name 'IKEA of Sweden' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info 2023-12-28 10:57:51MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xb43522fffea8706e","ieee_address":"0xb43522fffea8706e","status":"successful","supported":false},"type":"device_interview"}'
```

heviiguy, 2023-12-28:
Just tried it with v1.34.0 ffa4391. Same result.

heviiguy, 2023-12-28:
Yeah, I also saw (bleatedly)  "Successfully configured '0xb43522fffea8706e'".

So, in spite of the zigbee2mqtt GUI showing me that the bulb is still unsupported (and doesn't present the proper graphic), I can finally toggle the bulb!!! There is, however, no option to dim the bulb. In any case, many, thanks, Koen.

[image]

Koenkk, 2023-12-29:
I did not merge it yet, will do it today

Koenkk, 2024-01-13:
What zhc version shows up in frontend -> settings -> about? If it's below 18.11.0 you are not running the latest and need to update.

heviiguy, 2024-01-13:
Thanks, Koen.

I'm on Zigbee-herdsman-converters version. 18.0.2 and z2m 1.34.0-dev commit: ffa4391
I've just done a docker pull on koenkk/zigbee2mqtt:latest-dev but it resulted in the same zhc version.

I'm unsure of how to update zhc alone. Can you please let me know how? 🙁

Koenkk, 2024-01-13:
That's definitely not the latest, it should say 1.35.1-dev at least.

heviiguy, 2024-01-13:
I assume the the issue is still on my end, right?  If so, why do you think that I'm not able to pull 1.35.1-dev successfully?

Koenkk, 2024-01-13:
Yes, maybe try removing the image before pulling?

heviiguy, 2024-01-13:
Yip, that did it!. I've got full control of my bulbs now. Thanks, Koen!

- wee minor issues: 
   -  the proper image of the bulb didn't appear; the Zigbee logo is in its place
   - the "map" view is a bit spastic now; the indivual entities as well as the dongle aren't visible in this view any longer. The only thing which appears now is the centre star which is moving about in spasms:
   
[image]

heviiguy, 2024-01-13:
Well this was strange (just as the docker pull not pulling the lastest image was "strange"); the icons are displaying correctly now and, the map has also calmed down and is showing the items correctly.

All is good!  😁

fynnstr, 2024-04-10:
Another version of this bulb is around -> **TRADFRI bulb E26 WW globe 806lm**

Could support be added @Koenkk?

[image]

fynnstr, 2024-04-10:
Made it work with this external converter

```
const {identify, light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TRADFRI bulb E26 WW globe 806lm'],
    model: 'SUPL',
    vendor: 'IKEA',
    description: 'Automatically generated definition',
    extend: [identify(), light({"brightness":{"range":[0,254]}})],
    meta: {},
};

module.exports = definition;
```

Koenkk, 2024-04-10:
Added!