Koenkk/zigbee2mqtt #20405: Latest-dev Docker Syntax Error

thk-socal, 2023-12-27:
### What happened?

Pulled the latest-dev docker images and have the following error:

`Using '/app/data' as data directory
/app/dist/util/settings.js:1
import data from './data';
^^^^^^
SyntaxError: Cannot use import statement outside a module
    at internalCompileFunction (node:internal/vm:73:18)
    at wrapSafe (node:internal/modules/cjs/loader:1178:20)
    at Module._compile (node:internal/modules/cjs/loader:1220:27)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1310:10)
    at Module.load (node:internal/modules/cjs/loader:1119:32)
    at Function.Module._load (node:internal/modules/cjs/loader:960:12)
    at Module.require (node:internal/modules/cjs/loader:1143:19)
    at require (node:internal/modules/cjs/helpers:119:18)
    at start (/app/index.js:90:22)`

This happens on startup.

### What did you expect to happen?

Docker image to start without error

### How to reproduce it (minimal and precise)

Pull latest-dev docker image

### Zigbee2MQTT version

Latest-dev 27-Dec-2023

### Adapter firmware version

7.3.2.0

### Adapter

Sonoff USB-E

### Debug log

'Using '/app/data' as data directory
/app/dist/util/settings.js:1
import data from './data';
^^^^^^
SyntaxError: Cannot use import statement outside a module
    at internalCompileFunction (node:internal/vm:73:18)
    at wrapSafe (node:internal/modules/cjs/loader:1178:20)
    at Module._compile (node:internal/modules/cjs/loader:1220:27)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1310:10)
    at Module.load (node:internal/modules/cjs/loader:1119:32)
    at Function.Module._load (node:internal/modules/cjs/loader:960:12)
    at Module.require (node:internal/modules/cjs/loader:1143:19)
    at require (node:internal/modules/cjs/helpers:119:18)
    at start (/app/index.js:90:22)'

001Tycoon, 2023-12-27:
same problem here

[21:24:57] INFO: Starting Zigbee2MQTT...
/app/dist/util/settings.js:1
import data from './data';
^^^^^^

SyntaxError: Cannot use import statement outside a module
    at internalCompileFunction (node:internal/vm:73:18)
    at wrapSafe (node:internal/modules/cjs/loader:1178:20)
    at Module._compile (node:internal/modules/cjs/loader:1220:27)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1310:10)
    at Module.load (node:internal/modules/cjs/loader:1119:32)
    at Function.Module._load (node:internal/modules/cjs/loader:960:12)
    at Module.require (node:internal/modules/cjs/loader:1143:19)
    at require (node:internal/modules/cjs/helpers:119:18)
    at start (/app/index.js:90:22)

Koenkk, 2023-12-27:
Should be fixed now!

closed 2023-12-27 by Koenkk

001Tycoon, 2023-12-27:
it works again, thanks for all the work

epicurean70, 2023-12-31:
Its not working for me. same error still