Koenkk/zigbee2mqtt #20407: Loginovo ZG-101ZL button - behaviour does not match docs

uncle-fed, 2023-12-27:
### What happened?

The [button's](https://www.zigbee2mqtt.io/devices/ZG-101ZL.html) actual behaviour differs from docs plus there is a likely actual bug in implemented behaviour:

1. The docs state that its possible to change between the `event`  and `command`  modes by sending a message to `zigbee2mqtt/FRIENDLY_NAME/set` with payload `{"operation_mode": NEW_VALUE}`. This does not work.  Sending the message (bioth in CLI as well as via the web front-end) produces the following error:

`Publish 'set' 'operation_mode' to '0x...' failed: 'Error: Write 0x..../1 genOnOff({"tuyaOperationMode":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"
writeUndiv":false}) failed (Timeout - 60044 - 1 - 186 - 6 - 4 after 10000ms)'`. 

1(a)   ... However. it _is_ possible to change between the `event` and `command` modes by doing a triple-click. I personally am quite puzzled by this behaviour. It is quite easy sometimes to produce 3 clicks when trying to do a double-click, which is a valid action for this button. When the mode changes, all `action` values become totally different. The `single/double` become `on/off`  in the `command`  mode. And this can become very confusing for somebody that expects a consistent behaviour from their button. I really don't follow the reasons of this implementation, but I understand that this is most likely the hardware "feature". Still, I wonder if there was a better way of making sure that such a major setting wouldn't be easily changed by a slightly trembling finger.

2. The docs state that the `action` can also be one of the following: `brightness_move_to_level`, `color_temperature_move`, `brightness_step_up`, `brightness_step_down`. How do I get to see those values? I cannot generate them no matter what I try with the  button. 

The behaviour I have currently is the following. In the `event`  mode the button produces `single`, `double` and `hold`. In the `command` mode button produces `on` (when clicked once), `off` (when clicked twice) and _no action_ when it is being held, however, when the button is long pressed in the `command`  mode, it does generate messages every few hundred ms, but those messages only contain `last_seen` and `linkquality`. As if the action is there but is missing in the actual message.

Just in case I will mention that I have `device_options:   legacy: false` set in my config. Is this long press in the `command`  mode where the `brightness_step_up/down` should be visible? Or do I have to set something extra in my config to make this working properly? In this case I believe the docs should be updated with more detailed information.

### Zigbee2MQTT version

1.34.0 commit: 56589dcc

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus (CC2652)

kahatie, 2024-02-11:
i have same problem
i just add by default `operation_mode` is set to `null` but act like is on `event` mode  
`payload '{"action":"single","battery":100,"linkquality":140,"operation_mode":null,"voltage":3000}'`

i tried to made a custom external_converters but failed it always load `Loginovo | ZG-101ZL` instead of my file

goldiee36, 2024-02-14:
The triple-click behavior is indeed very annoying. If setting the "operation_mode" would be working, at least we could automatically change back to "event" mode, every time someone accidentally triple-clicks.

numminorih-git, 2024-04-28:
i have same problem

Raganal, 2024-05-20:
Not working button (after searching issues and configs nothing help):

Zigbee Model TS004F
Zigbee Manufacturer _TZ3000_ja5osu5g

Firmware version 0122052017
Manufacturer [Loginovo](https://www.zigbee2mqtt.io/supported-devices/#v=Loginovo)
Model [ZG-101ZL](https://www.zigbee2mqtt.io/devices/ZG-101ZL.html#loginovo-zg-101zl)

Action Triggered action (e.g. a button click) Empty string("")

Failed to configure '0xa4c138f4bf3bcdb2', attempt 4 (Error: ZCL command 0xa4c138f4bf3bcdb2/1 genBasic.read([4,0,1,5,7,65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:315:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:770:26) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:401:29) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/moes.ts:473:13) at Configure.configure (/app/lib/extension/configure.ts:121:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

tlogik, 2025-01-04:
This issue is not stale!
I can report i have the same issues as OP has defined

kahatie, 2025-01-06:
I continue to follow and wait to see if anyone finds a solution (^__^)

XAV-01, 2025-03-06:
see https://github.com/Koenkk/zigbee2mqtt/issues/23598

closed 2025-05-13 by github-actions[bot]