Koenkk/zigbee2mqtt #20418: [New device support]:

Herbertholzkopf, 2023-12-28:
### Link

https://www.innr.com/de/produkt/innr-smart-plug-zigbee-3-0/

### Database entry

{"id":138,"type":"Router","ieeeAddr":"0xa4c138e34a01dd56","nwkAddr":19372,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"SP 240","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,8,4096,1794,2820,57345],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"SP 240","manufacturerName":"innr","powerSource":1,"zclVersion":8,"appVersion":20,"stackVersion":54,"hwVersion":1,"dateCode":"20231019","swBuildId":"1.4.3"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":20,"stackVersion":54,"hwVersion":1,"dateCode":"20231019","swBuildId":"1.4.3","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1703776927279,"defaultSendRequestWhen":"immediate"}

### Comments

I bought the plug on Amazon and tried to pair it to ZHA. That worked so I tried it in Zigbee2MQTT but a error Popup came that said that the device is not supported.
I'm not really in to that programming so I hope you can help me get the device working...
And sorry for my bad English :)

I run Zigbee2MQTT as a addon in Homeassistant.
Both are on the newest version.

Koenkk, 2023-12-29:
`SP 240` will be supported in the 1 January z2m release.

closed 2023-12-29 by Koenkk

Laurentwb, 2024-05-07:
> `SP 240` will be supported in the 1 January z2m release.

Hello,
Is it suported now ?
I'm new to all this and have little chance with 2 SP240 while everything is fine with the other devices I own.
Thank you inadvance.
Laurent