Koenkk/zigbee2mqtt #20424: Inovelli Converter Missing Parameters and Wrongly Named Parameters

efaden, 2023-12-28:
### What happened?

For the VZM31 parameter 26 is missing.  

see: https://help.inovelli.com/en/articles/8189241-blue-series-2-1-switch-parameters#h_d74d1ef609

For the VZM35 a bunch of the parameters names/values are incorrect since they differ from the VZM31, but they aren't different in the converter.

see: https://help.inovelli.com/en/articles/8231399-blue-series-fan-switch-parameters

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20230507

### Adapter

Tubes POE

Koenkk, 2023-12-29:
@InovelliUSA could you check this?

efaden, 2023-12-29:
If I have time later I can run through them and submit a PR

efaden, 2023-12-29:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/6786

closed 2024-07-27 by github-actions[bot]