Koenkk/zigbee2mqtt #20428: [New device support]: Namron Thermostat - Tuya TS0601-  _TZE204_p3lqqy2r

christian1986, 2023-12-29:
### Link

https://www.namron.com/products/namron-zigbee-touch-termostat-16a-hvit-155/

### Database entry

{"id":114,"type":"Router","ieeeAddr":"0x3425b4fffe327a1b","nwkAddr":10818,"manufId":4098,"manufName":"_TZE204_p3lqqy2r","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"qO!-iqO!-irO!-irO!-irO!-isO!-isO!-i","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_p3lqqy2r","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}},"hvacThermostat":{"attributes":{}},"genOta":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1703839852491,"defaultSendRequestWhen":"immediate"}

### Comments

Got stuck, lack of knowledge on how to add a Tuya device.

adamchengtkc, 2023-12-30:
https://www.facebook.com/groups/680252689011262/permalink/2045892439113940/

christian1986, 2023-12-30:
Nice, a good start👌🏻

christian1986, 2024-01-02:
Some details, still not able to generate a working solution. Perhaps others can assist?

{"1":"Switch",
"2":"Mode",
"16":"Set temperature",
"24":"Current temperature",
"28":"Temperature correction",
"30":"Child lock",
"101":"Floor  temperature",
"102":"Sensor",
"103":"Swing",
"104":"LOAD",
"105":"Program",
"106":"Open window",
"107":"High temperature protection",
"108":"设备类型",
"109":"Reg模式周期",
"110":"Duty值",
"120":"负载电流", - Ampere
"121":"负载电压", - Volt
"122":"负载功率", - Watt
"123":"当前电量", - kWh
"199":"SN"}

closed 2024-01-04 by christian1986

Jellyfrog, 2024-01-17:
Does the date/clock feature work for you guys? I'm not seeing the date in the top left corner.
Seeing this in the logs: `No converter available for '4512752/4512753' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":142}'`

christian1986, 2024-01-21:
I have the same in my logs.

I don't think it support showing the clock?

Jellyfrog, 2024-01-21:
Not sure but it shows it in the offical documentation
https://www.elektroimportoren.no/docs/lib/4512752-Brukerveiledning-5.pdf