Koenkk/zigbee2mqtt #20431: [Wrong device]: Woolley Zigbee Smart Plug BSD59 being recognized as BSD29

LeoColman, 2023-12-29:
### Link

https://www.cntooya.com/pd/10008

### Model

BSD59 // SA-032-20A-ZB

### Description

Zigbee 3.0 smart plug 20A (BR version) (BSD59)

### Vendor

Woolley

### Picture (link)

https://source.cntooya.com/all/pro/CT-BU-BSD59/details/1.jpg

### Database entry

{"id":9,"type":"Router","ieeeAddr":"0x7cb94c76d9d70000","nwkAddr":54253,"manufId":4742,"manufName":"eWeLink","powerSource":"Mains (single phase)","modelId":"CK-BL702-SWP-01(7020)","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,3,4,5,6,4096,64599,64529],"outClusterList":[25,2820,10],"clusters":{"64529":{"attributes":{"28676":0,"28677":108227,"28678":0}},"genBasic":{"attributes":{"modelId":"CK-BL702-SWP-01(7020)","manufacturerName":"eWeLink","powerSource":0,"zclVersion":8,"appVersion":1,"stackVersion":0,"hwVersion":0,"dateCode":"20210610-BL","swBuildId":"1.0.1"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeb31b2d","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":0,"hwVersion":0,"dateCode":"20210610-BL","swBuildId":"1.0.1","zclVersion":8,"interviewCompleted":true,"meta":{"configured":910400241},"lastSeen":1703857322132,"defaultSendRequestWhen":"immediate"}]

### Notes

Bought in 

https://shopee.com.br/Woolley-Zigbee-Smart-Plug-16A-20A-Padr%C3%A3o-Brasil-Tuya-Wifi-Adapter-Power-Socket-Monitor-Timer-Sa%C3%ADda-Sem-Fio-Vida-%C3%9Atil-Da-De-Energia-Voice-Remote-Works-Para-Google-Home-Alexa-Echo-i.989102607.18894950080

All controls seem to be fine. I'm guessing it's working just like the BSD29, but with different configurations. In the `cntooya` link I've added, it seems to be a big producer for this product as a white label.

Koenkk, 2023-12-30:
Both use the same modelId so we cannot distinguish them, I've changed the model to `BSD29/BSD59`

closed 2023-12-30 by Koenkk

LeoColman, 2023-12-30:
I think the description is still incorrect, as the model can be the 20A Brazilian version. Maybe only "ZigBee 3.0 Smart Plug"?

Koenkk, 2023-12-30:
done!

Bosk22, 2024-01-02:
I don't know what exactly has changed in this version to cause the issue but as of this release (Zigbee2MQTT 1.35.0-1 ) the Woolley Smart Plug can no longer be turned on / off remotely from the GUI or from HomeAssistant. However all parameters are still detected (Watts / Current / Voltage / Link Quality and current ON/OFF state) at each value change. Prior to this release on/off remote control was working.

LeoColman, 2024-01-02:
I don't think this could be related to this change specifically, as these changes are only name and description

Bosk22, 2024-01-02:
[image]

Above is the details of the plug that seems to have  stopped working as of this release.

Apologies if I am posting this to the wrong thread / inappropriate thread. (Is there a better suited one ?)

Based on the following release note I take it this release has a quiet a major change to how device functionality is handled / detected and am I guessing that might be the cause.

Quote - This release contains a quite fundamental change on how Zigbee2MQTT handles unsupported devices.
Zigbee2MQTT will now attempt to automatically detect features supported by the device and discover those to e.g. the frontend and Home Assistant.
This means that if you, for example pair an unsupported light, there is no need to wait for official support. Basic features will work out-of-the-box.
The way to officially add support for devices has also been simplified due to this ([docs](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition)). These changes have been made in https://github.com/Koenkk/zigbee-herdsman-converters/issues/6692 and https://github.com/Koenkk/zigbee2mqtt/pull/20358.

hajabu0, 2024-02-02:
I have also a smart socket that was delevered as BSD33 (SA-029-20A-BUP)  but was discovered as BSD29 and now as DSB29/BSD59. Only the picture is not the British plug but a Europian. But it works fine.
[image]