Koenkk/zigbee2mqtt #20434: Xiaomi ws-euk03 and ws-euk02 control Relay not working

cddu33, 2023-12-29:
### What happened?

When i clicks on decoupled control Relay  
[image]

in previous versions it worked

### What did you expect to happen?

Change configuration

### How to reproduce it (minimal and precise)

Click on decoupled or control Relay 

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20210708

### Adapter

Conbee 2 and sonoff zStack3x0

### Debug log

2023-12-29 19:04:03Received MQTT message on 'zigbee2mqtt/Lumière CP/set' with data '{"operation_mode_left":"control_relay"}'
Debug 2023-12-29 19:04:03Publishing 'set' 'operation_mode' to 'Lumière CP'
Error 2023-12-29 19:04:03Publish 'set' 'operation_mode' to 'Lumière CP' failed: 'Error: undefined is not a object, got string ("control_relay")'
Debug 2023-12-29 19:04:03Error: undefined is not a object, got string ("control_relay") at Object.assertObject (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:511:15) at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:2488:19) at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:248:52) at EventEmitter.emit (node:events:526:35) at EventBus.emitMQTTMessage (/opt/zigbee2mqtt/lib/eventBus.ts:109:22) at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.ts:141:27) at WebSocket.<anonymous> (/opt/zigbee2mqtt/lib/extension/frontend.ts:130:27) at WebSocket.emit (node:events:514:28) at Receiver.receiverOnMessage (/opt/zigbee2mqtt/node_modules/ws/lib/websocket.js:1192:20) at Receiver.emit (node:events:514:28)
Debug 2023-12-29 19:04:12Received Zigbee message from 'Capteurs salon', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2110}' from endpoint 1 with groupID 0

cddu33, 2023-12-30:
Work on 1.33.0

Koenkk, 2023-12-30:
Will be fixed in the 1 January Z2M release.

closed 2023-12-30 by Koenkk

cddu33, 2023-12-30:
Thanks have a happy holiday