Koenkk/zigbee2mqtt #20435: Bad units for TS0601_smart_air_house_keeper

McGiverGim, 2023-12-29:
### What happened?

The device shows very high values for HCHO and VOC.
To me it seems the same problem commented https://github.com/Koenkk/zigbee2mqtt/issues/11033

This are the values that I receive:
```
2023-12-29 19:54:45MQTT publish: topic 'zigbee2mqtt/calidad_aire_comedor', payload '{"co2":718,"formaldehyd":74,"humidity":54.1,"last_seen":"2023-12-29T19:54:56+01:00","linkquality":145,"pm25":3,"temperature":21.1,"voc":412}'
``` 
The HCHO and VOC value are cleanly wrong: in theory according to the manual:
- The value of fomaldehyde must be between 0 and 10, and I have values of more than 100 and the "good" value must be about 0.02 or less.
- The value of VOC must be between 0 and 99.0, and I have values of 500 and the "good" value must be about 0.1 or less.

The device is the `_TZE200_dwcarsat` and I don't know if with something like this commit will be enough: https://github.com/Koenkk/zigbee-herdsman-converters/commit/044ee8db878b33096796fcf448b1a8ca4da25001

the values will remain high but they will seem to be at least between the limits.

### What did you expect to happen?

The values to be correct

### How to reproduce it (minimal and precise)

Simply look at the values published by z2m for this device.

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20230507

### Adapter

Egony E72

### Debug log

```
2023-12-29 19:54:45MQTT publish: topic 'zigbee2mqtt/calidad_aire_comedor', payload '{"co2":718,"formaldehyd":74,"humidity":54.1,"last_seen":"2023-12-29T19:54:56+01:00","linkquality":145,"pm25":3,"temperature":21.1,"voc":412}'
```

McGiverGim, 2023-12-29:
If it helps, it seems to me that the device is "calibrating" (some values are reducing with time), I will post the stable values later, now they are here:
```
2023-12-29 20:15:44MQTT publish: topic 'zigbee2mqtt/calidad_aire_comedor', payload '{"co2":482,"formaldehyd":25,"humidity":54.7,"last_seen":"2023-12-29T20:15:55+01:00","linkquality":145,"pm25":2,"temperature":21,"voc":140}'
```

McGiverGim, 2023-12-29:
More info: The PM 2.5 shows, sometimes, values like 43522 and 43521:
```
2023-12-29 20:38:56MQTT publish: topic 'zigbee2mqtt/calidad_aire_comedor', payload '{"co2":353,"formaldehyd":2,"humidity":54.4,"last_seen":"2023-12-29T20:39:07+01:00","linkquality":149,"pm25":43521,"temperature":21,"voc":6}'

```
I think is another exception to add here:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/15f6a570fa14c1492e062c00bf62228dcb2576de/src/lib/legacy.ts#L4016

McGiverGim, 2023-12-30:
After 24 h, here is the actual value:
```
info  2023-12-30 10:03:02: MQTT publish: topic 'zigbee2mqtt/calidad_aire_comedor', payload '{"co2":353,"formaldehyd":2,"humidity":52.8,"last_seen":"2023-12-30T10:03:02+01:00","linkquality":149,"pm25":7,"temperature":21.6,"voc":5}'
```
The VOC and Formaldehyd, in the current code there is another device that divides by 10 and 100, but I'm not too sure that this is the correct value for this device.
I've found this capture of the tuya application, to see how they look, but without the same values in z2m at the same time I suppose it does not help too much:
[image]

A strange thing is that the VOC value appears in the capture as mg/m3, but according to the documentation of the product it must be ppm, and just the inverse for the Formaldehyde. So or the doc is wrong or the app is wrong (I doubt they do some kind of conversion). Maybe the app shows a device from another brand that is slightly different but I doubt it. We will need someone with the device and the Tuya gateway at the same time to be sure of that.

I've another device, a JQ300, that shows this values for the same place (caution, the units are different than Tuya device and both are cheap devices, so they are not accurate):
- Formaldehyde: 0.002 mg/m3 -> The Tuya shows 2 ug/m3 (seems correct with the conversion)
- TVOC: 0.467 mg/m3 -> The Tuya shows 5 ppm (I don't know how they show the conversion in the app)
- eCO2: 382 ppm -> The Tuya shows 353 ppm (seems ok)
- PM 2.5: 33 ug/m3 -> The Tuya shows 7 ug/m3 (I doubt the JQ300 shows it right, so I don't know who has the "best" value)

Conclusions:
- The first values send by the device were too big acoording to the max and min values from the documentation of the device. This seems to imply that we need to divide them. But the values, when I connect it first, were to high. Dividing them will not totally make them fit in the range, this is the first medition of the device:
```
info  2023-12-29 18:55:33: MQTT publish: topic 'zigbee2mqtt/calidad_aire_comedor', payload '{"co2":1002,"formaldehyd":182,"humidity":53.4,"last_seen":"2023-12-29T18:55:33+01:00","pm25":3,"temperature":21.2,"voc":1014}'
```
- The current values, after some hours, seems more or less ok if I compare them with the JQ300, so it seems that we don't need to divide them.
- The VOC, according to the doc, must be between 0 and 99.9, so it seems it must have a decimal, another reason to divide them.
- The screen capture have very low values for some of them, but without having the real Tuya and gateway to do a real compare it does say nothing, but maybe we need to divide them.
- From time to time, the HCHO shows values over the 0-10 of range, so another reason to divide them.
- The PM 2.5 shows from time to time spikes with values like 43522, 43521, 9254, 9253, etc. very out of range, it seems that values over 1000 (the max value in the documentation of the device) need to be filtered.

So to me, it seems that if we ignore the values of the JQ300, we need to divide and filter them. But if we look at the JQ300, they are ok and maybe we simply need to reduce it to the correct "range". I will keep an eye on the values in the next hours but I doubt I can get to a conclusion without the real app and the real gateway.

McGiverGim, 2023-12-30:
@Burki24, @1stsetup I've seen that you added the commits to support the `_TZE200_dwcarsat` so I suppose you own this device.
Do you want that we talk about it?

I think at least the value of VOC is wrong. The range that appears in the documentation is 0 to 99.9 so it seems to have at least one decimal and actually it shows integer values only. If the documentation of the product is right, the units are ppm and looking into internet this are the "good" values (in ppb, not ppm, so divide by 1000):
[image]
so we need at least one decimal.
Any of you have the "real" tuya gateway to compare against it?

Burki24, 2023-12-31:
> @Burki24, @1stsetup I've seen that you added the commits to support the `_TZE200_dwcarsat` so I suppose you own this device. Do you want that we talk about it?
>
> I think at least the value of VOC is [...]

Yes, I will check it with the original tuya-gateway. Comming back soon.

Greetz
Burkhard

McGiverGim, 2023-12-31:
Thanks! A capture of z2m values against Tuya app with more or less the same conditions will be great. Maybe the easiest one is with window open for some minutes if possible. But any test will answer several questions.

ruttromp, 2024-01-12:
Same issue here. I found this thread https://github.com/zigpy/zha-device-handlers/issues/1563 with some interesting information

McGiverGim, 2024-01-13:
@ruttromp at the end of the post, there are some conclusions comparing with the Tuya app.
@Burki24 maybe you can compare that conclusions with your own observance, to see if they are right.

McGiverGim, 2024-07-19:
The issue remains.

10der, 2024-12-23:
1) issues still here
2) humidity :) 
[image]

McGiverGim, 2025-02-22:
It remains...

closed 2025-05-03 by github-actions[bot]