Koenkk/zigbee2mqtt #20438: Phillips hue bulb unable to display brightness under about 15%

labourrd, 2023-12-30:
### What happened?

Change brightness anywhere below 15-20% and the bulb dies not change. Also going from 0% to 1% results in a large jump in brightness

### What did you expect to happen?

A gradual increase of brightness from 0-1 and above

### How to reproduce it (minimal and precise)

Pair hue bulb with z2m
Attempt changing brightness from 0 to anything between 1 and 15%

### Zigbee2MQTT version

1.34.0-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

kristianrl, 2023-12-30:
In case it was previously paired with the Hue bridge, did you experience the same with the bulb in the Hue app?

I have some LED spots (non-Hue), that are not visible between 0% and 8%, but take a big jump from 8% - 9%.

lux73, 2023-12-30:
i can not second this - all of my Hue Bulbs works flawless from 0-100%

All my Hue's are older ones without BT

So it would be interesting which particular Bulbs are you have

labourrd, 2023-12-30:
> In case it was previously paired with the Hue bridge, did you experience the same with the bulb in the Hue app?
> 
> I have some LED spots (non-Hue), that are not visible between 0% and 8%, but take a big jump from 8% - 9%.

These bulbs were connected to a hue bridge before and they worked fine, had a gradual increase from 0-100.

> i can not second this - all of my Hue Bulbs works flawless from 0-100%
> 
> All my Hue's are older ones without BT
> 
> So it would be interesting which particular Bulbs are you have

My bulbs are a19(w/Bluetooth)

closed 2024-06-28 by thestigjr