Koenkk/zigbee2mqtt #20445: Zigbee2mqtt crashes randomly

derkorte, 2023-12-30:
### Is your feature request related to a problem? Please describe

Zigbee2mqtt ''creates'' issues. Sometimes i have to pair devises about 10 times, sometimes they just connect in a sek (no matter where located).  Also this is a random message since a few weeks: publish 'set' 'state' to failed: 'error: command genonoff.off({}, {"sendwhen":"immediate","timeout":10000,"disableresponse":false,"disablerecovery":false,"disabledefaultresponse":false,"direction":0,"srcendpoint":null,"reservedbits":0,"manufacturercode":null,"transactionsequencenumber":null,"writeundiv":false}) failed (no response received (86))'

Also when i'm pairing a Device in Logs comes:  Device XXXX joined -> starting interview -> Zigbee2Mqtt startet!

### Describe the solution you'd like

perhaps that everything works like id did last month....

### Describe alternatives you've considered

everything started before Christmas, there i was running Home Assistant OS on a PI 3b with SD and an Sonoff Dongle E with latest Firmware.
So i gave myself a little present and bought a pi5 8gb, installed HASSOS on an SSD with an official Case and ..... same issue. So tried a Conbee II (with latest firmware) and same issue, so tried my old cc2531 and same issue. 

### Additional context

Here my Logs:

Zigbee2MQTT:debug 2023-12-30 16:33:21: Received MQTT message on 'homeassistant/sensor/0xa4c1380f62782395/power_on_behavior/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Steckdose 4/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c1380f62782395"],"manufacturer":"TuYa","model":"In-wall outlet (TS011F_wall_outlet)","name":"Steckdose 4"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:power-settings","json_attributes_topic":"zigbee2mqtt/Steckdose 4","name":"Power-on behavior","object_id":"steckdose_4_power_on_behavior","origin":{"name":"Zigbee2MQTT","sw":"1.34.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Steckdose 4","unique_id":"0xa4c1380f62782395_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
Zigbee2MQTT:debug 2023-12-30 16:33:21: Received MQTT message on 'homeassistant/select/0xa4c1380f62782395/power_on_behavior/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Steckdose 4/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/Steckdose 4/set/power_on_behavior","device":{"identifiers":["zigbee2mqtt_0xa4c1380f62782395"],"manufacturer":"TuYa","model":"In-wall outlet (TS011F_wall_outlet)","name":"Steckdose 4"},"entity_category":"config","icon":"mdi:power-settings","json_attributes_topic":"zigbee2mqtt/Steckdose 4","name":"Power-on behavior","object_id":"steckdose_4_power_on_behavior","options":["off","previous","on"],"origin":{"name":"Zigbee2MQTT","sw":"1.34.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Steckdose 4","unique_id":"0xa4c1380f62782395_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
Zigbee2MQTT:debug 2023-12-30 16:33:21: Received MQTT message on 'homeassistant/sensor/0xa4c1380f62782395/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Steckdose 4/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c1380f62782395"],"manufacturer":"TuYa","model":"In-wall outlet (TS011F_wall_outlet)","name":"Steckdose 4"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/Steckdose 4","name":"Linkquality","object_id":"steckdose_4_linkquality","origin":{"name":"Zigbee2MQTT","sw":"1.34.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Steckdose 4","unique_id":"0xa4c1380f62782395_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:debug 2023-12-30 16:33:21: Received MQTT message on 'homeassistant/sensor/0xa4c1380f62782395/last_seen/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Steckdose 4/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c1380f62782395"],"manufacturer":"TuYa","model":"In-wall outlet (TS011F_wall_outlet)","name":"Steckdose 4"},"device_class":"timestamp","enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:clock","json_attributes_topic":"zigbee2mqtt/Steckdose 4","name":"Last seen","object_id":"steckdose_4_last_seen","origin":{"name":"Zigbee2MQTT","sw":"1.34.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Steckdose 4","unique_id":"0xa4c1380f62782395_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'
Zigbee2MQTT:debug 2023-12-30 16:33:21: Received MQTT message on 'homeassistant/light/0x04cd15fffe5cd838/light/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Strahler Pool /availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","brightness":true,"brightness_scale":254,"color_mode":true,"command_topic":"zigbee2mqtt/Strahler Pool /set","device":{"identifiers":["zigbee2mqtt_0x04cd15fffe5cd838"],"manufacturer":"Lidl","model":"Livarno Home outdoor spotlight (HG08010)","name":"Strahler Pool "},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"json_attributes_topic":"zigbee2mqtt/Strahler Pool ","max_mireds":500,"min_mireds":153,"name":null,"object_id":"strahler_pool_","origin":{"name":"Zigbee2MQTT","sw":"1.34.0","url":"https://www.zigbee2mqtt.io"},"schema":"json","state_topic":"zigbee2mqtt/Strahler Pool ","supported_color_modes":["xy","color_temp"],"unique_id":"0x04cd15fffe5cd838_light_zigbee2mqtt"}'
Zigbee2MQTT:debug 2023-12-30 16:33:21: Received MQTT message on 'homeassistant/select/0x04cd15fffe5cd838/effect/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Strahler Pool /availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/Strahler Pool /set/effect","device":{"identifiers":["zigbee2mqtt_0x04cd15fffe5cd838"],"manufacturer":"Lidl","model":"Livarno Home outdoor spotlight (HG08010)","name":"Strahler Pool "},"enabled_by_default":false,"icon":"mdi:palette","json_attributes_topic":"zigbee2mqtt/Strahler Pool ","name":"Effect","object_id":"strahler_pool__effect","options":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"origin":{"name":"Zigbee2MQTT","sw":"1.34.0","url":"https://www.zigbee2mqtt.io"},"unique_id":"0x04cd15fffe5cd838_effect_zigbee2mqtt"}'
Zigbee2MQTT:debug 2023-12-30 16:33:21: Received MQTT message on 'homeassistant/switch/0x04cd15fffe5cd838/switch_do_not_disturb/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Strahler Pool /availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/Strahler Pool /set/do_not_disturb","device":{"identifiers":["zigbee2mqtt_0x04cd15fffe5cd838"],"manufacturer":"Lidl","model":"Livarno Home outdoor spotlight (HG08010)","name":"Strahler Pool "},"json_attributes_topic":"zigbee2mqtt/Strahler Pool ","name":"Do not disturb","object_id":"strahler_pool__do_not_disturb","origin":{"name":"Zigbee2MQTT","sw":"1.34.0","url":"https://www.zigbee2mqtt.io"},"payload_off":"false","payload_on":"true","state_topic":"zigbee2mqtt/Strahler Pool ","unique_id":"0x04cd15fffe5cd838_switch_do_not_disturb_zigbee2mqtt","value_template":"{% if value_json.do_not_disturb %} true {% else %} false {% endif %}"}'
Zigbee2MQTT:debug 2023-12-30 16:33:21: Received MQTT message on 'homeassistant/sensor/0x04cd15fffe5cd838/color_power_on_behavior/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Strahler Pool /availability","value_template":"{{ va [... cut ...]
Zigbee2MQTT:debug 2023-12-30 16:33:50: Successfully pinged 'Leuchte Schlafzimmer ' (attempt 1/1)
Zigbee2MQTT:debug 2023-12-30 16:33:51: Retrieving state of 'Schalter Flur ' after reconnect
TypeError: Cannot read properties of undefined (reading 'find')
    at /app/lib/extension/availability.ts:201:84
    at Array.find (<anonymous>)
    at Object.<anonymous> (/app/lib/extension/availability.ts:201:66)
    at Timeout.later [as _onTimeout] (/app/node_modules/debounce/index.js:32:24)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)

1nfinity84, 2023-12-31:
Yeah, I'm having similar issues since about the same time. I thought I'm getting some sort of interference, but my zigbee mesh is massive...

ggatward, 2024-01-02:
+1 - randomly find my Zigbee devices not working. The plugin shows no errors, but is unable to display any devices. Stopping the service and then starting it SOMETIMES fixes the problem for a day or two, then dead again.  Started around the same timeframe prior to Christmas.   This was on 1.34.0 - Have just updated to 1.35.0, will see if the problem continues and get some debug logs

ottoketney, 2024-01-12:
I have the same behaviour

Zigbee2MQTT version : 1.35.1
Coordinator type : EZSP v9
Coordinator revision : 7.1.1.0 build 273

derkorte, 2024-01-12:
I've fixed it in my case. But the thing is that there was many steps I've read about and changed.

First try to use a normal USB port. Sometimes a USB 3.0 port creates issues. Configure new port in z2m and test if it's stable.

closed 2024-01-27 by derkorte